CharlesGillanders/homeassistant-alphaESS #1: cannot find the integration home-assistant

fragglefrid, 2021-08-07:
This might be a stupid question, but how can I find the integration in homeassistant? Tried Configuration -> Integrations -> Add -> Alpha ESS but it is not there.

johnjoemorgan, 2021-08-15:
@CharlesGillanders Will you be progrssing with this.  I have a client that will be installing a Alpha ESS system later this year and would be willing to work with you on this once the install is running.

charlj22, 2021-08-17:
I am also interested in this. Do you need any help testing? I am trying to get details about the API, but no luck yet.

CharlesGillanders, 2021-08-17:
I haven't been able to find time to work on this for quite some time.  I too was unable to find any documentation on the API, however I was able to monitor the API calls made by a web browser accessing the Alpha ESS site and at least partially reverse engineer what the AlphaESS API does.

I've uploaded the work in progress which I have on the reverse engineered API.

There are methods to connect to AlphaESS, obtain a list of units registered to your credentials, and retrieve statistics from their web app for your unit(s)

My test script works enough at a command line to read those statistics for my Alpha unit but I obviously haven't been able to try this with anyone else.  Nor have I yet figured out how to make this integrate correctly with HomeAssistant so there's a considerable amount of work needed to be done to make this ready for use.

charlj22, 2021-08-17:
Thanks. It works. Only Load not getting a value, but I will check what the problem is. Im thinking of just making this a mqtt sensor to get the info into home assistant

CharlesGillanders, 2021-08-17:
I fixed the load problem, if you want to grab the latest version of the test harness.

johnjoemorgan, 2021-08-18:
@charlj22 MQTT works - then we can create a sensor for the energy dashboard

charlj22, 2021-08-24:
@johnjoemorgan I have forked this to add a MQTT component. It works and I can see the energy in Home Assistant. There is quite a few problems though. HA wants energy, not power, so I had to create a energy sensor for HA do get the data. HA also only do Home, Grid and Solar. No battery yet. The python program also stops and dont restart. Im not sure how to fix that. Maybe add logging to find the reason.

Njibhuvgy, 2021-09-12:
[AlphaESS Web API V2.120210512004203.pdf](https://github.com/CharlesGillanders/homeassistant-alphaESS/files/7150227/AlphaESS.Web.API.V2.120210512004203.pdf)

I found this documentation if its useful. I too would be interested in using this integration to track solar/battery data in home assistant.

Ralphy140, 2021-09-17:
Home Assistant has now updated to support the battery component, I am having a smile 5 in the next few weeks and keen to get this working to complete with the solar integration i have done already

CharlesGillanders, 2021-09-24:
If anyone is interested in trying this out in home assistant I have updated all of the code so it should now work as a custom component.  This is alpha quality code - I am running it myself in my own home assistant but I've only been able to test with my own system.

Ralphy140, 2021-09-25:
My system is booked to be installed on the 13th I will report back once it's connected and online

CharlesGillanders, 2021-09-25:
> My system is booked to be installed on the 13th I will report back once it's connected and online 

Excellent - thanks for this.

johnjoemorgan, 2021-09-25:
MY system won't be installed until the New Year.  Sorry can't be much help till then

dry, 2021-09-25:
Thank you very much for this excellent work. I have installed the component and it seems to be working well. The only slight oddity I've noticed so far is that Battery to Load (data["Ebat"]) doesn't seem to actually report the total battery discharge - unless I have misunderstood what this value is supposed to represent. I have the Smile B3 and it might be an issue with this system's reporting. As you can see, I have had some grid import overnight between 0100 and 0400 and the rest of the night was on stored energy. This is what is shown:

[image]

[image]

[image]

I have a request, would you please add the battery state of charge (data["Soc"]) as an additional sensor so that we can run automations based on Soc and the weather/solar forecast?

CharlesGillanders, 2021-09-25:
@dry Thank you so much for providing such useful feedback.  I will look to add the state of charge - the figure is in the API call that I am using but I took a shortcut with the current sensors by only gathering those that reported kWh so to gather SOC I'll need to fix that shortcut.  It will be better for the longer term to do this so best just to go ahead now!

As for the Ebat figure - that's what I thought that figure was meant to be but as we don't have any actual documentation for the Alpha ESS API it's a bit of a guess - If it's not reporting what I thought then I'll have to look elsewhere for that figure.

dry, 2021-09-25:
It may be that different system types report their settings differently. Do you have a Smile B3? I just realised that I can look back at the history of individual sensors and this graphic is the Battery to Load isolated. It was reporting a value when I first installed the component but, oddly, it was a decreasing value. I'm very new to HA so it may be that I don't have this setup correctly.

[image]

dry, 2021-09-25:
And I've just noticed this log entry:

[image]

CharlesGillanders, 2021-09-25:
I've made some further updates - it now tracks three additional battery measurements - state of charge, charge, and discharge.
I've removed the previous battery to load sensor - it should be replaced with the new discharge sensor.
I am still not sure what is happening with solar to load - more investigation needed there.

Updating should be as simple as stopping home assistant, copying in the updated code and re-starting home assistant. Once started you will need to remove the old Battery to Load sensor from the entities page.

dry, 2021-09-25:
Thank you. I've updated and will report back when it's run for a few hours.

dmsignup, 2021-09-26:
Hi.
I would like to say this integration is very useful to me so thank you for developing it.
Reviewing my setup with this integration I have noticed that the Solar to load appears to be the charge of the battery with a huge exception. 
It can be seen that the value increases while the battery is being charged and decreases when the battery is used to supply power to the load.
However once you hit a new day it drops down to 0 and even when the battery starts to charge the value does not go up until it has charged a few kWh first.
This kind of looks like when a new day is hit the number becomes a negative of what ever is remaining and it shows anything as a negative as a 0.
I do not know enough about the programming to check if this is correct but that’s what the numbers look like.

dry, 2021-09-27:
I've run the latest version (0.0.2) for a full 24 hours now and it works really well in terms of the HA Energy tab. No issues that I can see. And thank you adding the Battery SoC, that works properly and should enable some great automations.

Ralphy140, 2021-10-05:
My Battery install has now been pushed out to early November due to stock issues. I will report back once i have it installed. 

I did get the custom integration up and running(though not connected to anything) but noticed that it uses the cloud service. Any way to make this communicate directly communicate to the battery locally? Don't get me wrong i am super happy that you have developed this so feel free to tell me to go away!

CharlesGillanders, 2021-10-11:
> My Battery install has now been pushed out to early November due to stock issues. I will report back once i have it installed.
>
> I did get the custom integration up and running(though not connected [...]

@Ralphy140 - Sorry not at this time.  I did see that someone had figured out how to setup a fake local server which tricked the Alpha unit into uploading the figures to a local system instead of the cloud but that also disabled all updates from Alpha and of course breaks the vendor app etcetera.  I wanted a solution which allowed home assistant to run alongside the existing app rather than replace it so I focused on using the cloud API rather than the more problematic local server approach.

tasranson, 2021-12-30:
@CharlesGillanders, thank you for this great contribution.  Just to let you know, it seems to work fine for my Alpha ESS Smile 13.3 (at least over the last 24 hours since I got it running).   I will monitor it for a while and compare the readings to the UI and App to see if there are any issues.   I do note in the HA UI, that the Battery SOC does not show up in the default Lovelace config in the same “Location” as the rest of the energy sensors, but that isn’t a significant issue.   In looking to fix this, I found that if I open any of the sensors on the integration page the UI reports something like:

This entity ('sensor.al8001620080041_state_of_charge') does not have a unique ID, therefore its settings cannot be managed from the UI. See the documentation for more detail.

Not sure if that is of concern or not.

Again, thank you so much for this :-)

stevebaele, 2022-01-06:
sorry for pikcing up this old topic. I´m new to Home Assistant and want to connect the information of the Alpha Ess. I have made the folder in de custom_components. copied all the files into it. But at configuration - integrations - add integration I can´t find anything. What am I doing wrong ?

stevebaele, 2022-01-06:
> This might be a stupid question, but how can I find the integration in homeassistant? Tried Configuration -> Integrations -> Add -> Alpha ESS but it is not there. I tried to find it in hacs as well, no luck.

did you got this working ?

chiversc, 2022-01-06:
you should have the files in a directory inside custom_components. If you then restart Homeassistant you should find alpha ess in configuration - >integrations.

stevebaele, 2022-01-06:
> you should have the files in a directory inside custom_components. If you then restart Homeassistant you should find alpha ess in configuration - >integrations. it worked for me.

I have created a folder alphaess in /root/config/custom_components
after I have copied all the folders and files into this directory
restarted Home Assistant

but when I go to configuration - devices|&services - integrations - add integration  I can´t find it

chiversc, 2022-01-06:
Sorry I am not more help then. Here's the files I have loaded on my homeassistant, and this was sufficient (after restart) to make the integration available. Sorry if it is an abvious suggestion, but you might want to check you have all these? Only other suggestion I have is to look at homeassistant log to see if an error is visible.

[image]

stevebaele, 2022-01-06:
> Sorry I am not more help then. Here's the files I have loaded on my homeassistant, and this was sufficient (after restart) to make the integration available. Sorry if it is an abvious suggestion, [...]

Ok, I´m missing some folders, they are not in the zip file

CharlesGillanders, 2022-01-06:
The __pycache__ folder is created automatically by HomeAssistant when the integration is loaded and the .git folder is only there if you have pulled the integration through git rather than using the zip download.  You do need the translations folder though and that should be there when you unzip the master.zip file from github.

stevebaele, 2022-01-06:
> The **pycache** folder is created automatically by HomeAssistant when the integration is loaded and the .git folder is only there if you have pulled the integration through git rather than using the [...]

thanks, then I have all teh correct files, but after restart i can´t find the integrations.  sorry for my noob questions. just started Ha yesterday, so a lot to figure out

chiversc, 2022-01-06:
anything appearing in the error log after you restart ?

stevebaele, 2022-01-06:
just a warning
2022-01-06 13:19:33 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration alphaess which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant

CharlesGillanders, 2022-01-07:
> I figured it out. the integration is now there and working :) coming from a knx this is something different but nice to figure out
> 
> thanks for all your help
> 
> now just find out to get all the information :)

Just in case it ends up helping someone else is there anything useful you can share about what you did to get it working?

Karl-Johns, 2022-02-25:
HI All - Same problem as above.  Cant see to find the integration.  All the folders are correct and in the right place.  Appreciate the assist.  I do get a waring in the logfile:
2022-02-25 15:40:55 WARNING (SyncWorker_1) [homeassistant.loader] We found a custom integration alphaess which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant

closed 2022-07-01 by CharlesGillanders

Bodsworld, 2022-07-04:
I also have the same error as Karl-Johns above

2022-07-04 22:26:36 WARNING (SyncWorker_2) [homeassistant.loader] We found a custom integration alphaess which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant

All folders are in the correct place, but cant find the integration :(