CharlesGillanders/homeassistant-alphaESS #5: Total load calculation error

Njibhuvgy, 2021-09-26:
This integration is really useful, thanks for all your work on it.

The Total Load entity seems to be a sum of Solar to Load, Grid to Load and **Battery Charge**. I think this should be Solar to Load, Grid to Load and **Battery Discharge**.

I hadn't noticed this before but having checked the Alpha ESS monitoring site/ app it seems to be a problem at their end. Has anyone else got this issue?

[image]

Njibhuvgy, 2021-09-26:
Solar to Battery also seems to be getting the wrong number in Home Assitant, it shows 7.4 kWh in my example above which is the total charge. Whereas it should be 5.1 kWh. This is correct on the Alpha monitoring site.

CharlesGillanders, 2021-09-26:
I released a second version yesterday which removed the solar to battery sensor and replaced it with specific sensors for battery charge and discharge (as well as battery state of charge).

I'd be interested in seeing if those sensors appear to be tracking better to what your expected numbers are?

The figures are pulled from a different API in the Alpha website.

Njibhuvgy, 2021-09-26:
I installed the integration this afternoon so assume I have the latest version. I can see the SOC which seems to be working well.

CharlesGillanders, 2021-10-11:
Did the change I made with the battery charge/discharge figures in the 2nd release fix the total load calculations?  In my energy dashboard I think I'm seeing the correct figures now but I'd love additional perspective.

CharlesGillanders, 2021-11-01:
I have updated sensor.py to take advantage of the recently updated API from Alpha ESS - Solar to Battery is now correctly tracking just the solar contribution to battery charge and there is a separate metric for grid to battery - both together add up to the total battery charge.

closed 2021-11-01 by CharlesGillanders