CharlesGillanders/homeassistant-alphaESS #6: Enable grid charging and update the charge level?

dry, 2021-09-27:
It would be fantastic if it were possible to integrate a service to toggle the grid charging and optionally update the charge level Soc. Do you think that might be something you could also enable?

CharlesGillanders, 2021-10-11:
So I found time to look at the AlphaEss site and it does *seem* that the web site allows you to configure the grid charging setup.  There's an API post call to https://www.alphaess.com/api/Account/CustomUseESSSetting with a custom json payload that seems to make some initial sense on first review.

So no promises but it looks like it could be done and what I have to figure out is what needs to go into that json payload to make it work.

Easiest to implement looks like it might be just turning off or on the grid charge setting (without adjusting any of the time settings)

Ralphy140, 2021-10-19:
Hey All, 

Got the battery installed and i am using a real time power company (amber.com.au if you are in Australia)that also integrates into HA. I would be super keen to have a switch that enables ether grid charging or discharge. I have found the section in the Aplha website that allows 2 pre-defined time for both charging and discharging but would love to be able to build an automation that links that to real time power cost that would be the holy grail

Ralphy

Beritknight, 2021-11-30:
I'd love this for two scenarios.

I've noticed that when it's set to charge from the grid in any given period, it won't discharge to load *even if* the state of charge is above the percentage you told it to charge to. Each morning I'd like to check SOC at 6am and if it's below 20%, switch on charge from grid with a max of 20%. That means that if I've drained my battery overnight then from 7-9am I'm using the charge I put in at 6-7am at off peak rates, rather than buying from the grid at peak rates.

The Alpha system is also not day of week capable. On the weekends I want to charge from the grid midday to 2pm both Saturday and Sunday. Right now I'm manually changing to that setting Saturday morning and then changing back to the 6am charge to 20% setting on Sunday afternoon. Would be great to automate that with home assistant.

mikeew, 2022-01-30:
One other scenario in which this would be useful is for charging an EV during off peak and NOT using the battery to charge the EV. 
At the moment I've set up the Alpha to not discharge to load between 4am and 7am and charge the EV between those hours. 
It would be very useful to tell the Alpha integration to disable discharge to load when the EV starts charging and then enable it again after it finishes.
Is this possible?

DarkSlice1, 2022-02-22:
confirmed i was able to programatically update the battery %.
unfortunately the data types you receive in the GET are not the same you send in the POST. (plus some missing)
https://github.com/DarkSlice1/AlphaEss_Monitor/commit/9a31d2bdc9ed122bf8901ce6bf2a389f66dedf89

eviltabouleh, 2022-02-23:
> I'd love this for two scenarios.
>
> I've noticed that when it's set to charge from the grid in any given period, it won't discharge to load _even if_ the state of charge is above the percentage you [...]

This is similar to what I'd love to do , different weekend rules to day of the week, I am currently checking SOC and input during the day and if after 10am (when my power cost drops) I charge for an hour or a lot more if it is raining. That way I am at 100% come peak power cost time

I always have my charge set 12am - 7am as I nearly always use all that plus my production.

bmannz, 2022-10-10:
I see in your alphaess project it's been updated to accommodate setting the charging/discharging details.
Is there any plan to make those available inside Home Assistant via this integration?
Like Eviltabouleh I would really like to be able to change these based on Solar forecast and SOC.

Pollert, 2022-10-31:
I will really appriciate the posibility to change any charging/discharging parameters via HA. This will really change the electricity operation inmy house.
Thank you

matt-oneill, 2022-11-04:
the functionality is in https://github.com/CharlesGillanders/alphaess to charge/discharge a system, the plan is to implement this in HA, but haven't had much time, and with alphaess changing the api frequently I haven't made much progress on this yet

CharlesGillanders, 2022-11-07:
@matt-oneill has implemented this functionality so it is now possible to set battery charge and discharge settings from HomeAssistant.  Release 0.3.0 has just been published with this new feature.

closed 2022-11-07 by CharlesGillanders