CharlesGillanders/homeassistant-alphaESS #7: Instantaneous values

cfgroenewald, 2021-10-09:
Hi
Is it possible to access the Real Time Power Graph info from Alpha ESS, the instantaneous values?

230delphi, 2021-10-11:
> The API exists for these on the AlphaESS website so yes in principal. My
> first thought is that right now Home Assistant polls AlphaESS every 5
> minutes - that seems reasonably frequent without storing too much
> information in the Home Assistant DB.
>
> Would 5 minute snapshots of the instantaneous values be any use to you in
> Home Assistant or would you need more frequent information?

For what it's worth, AlphaESS actually emits the data every 10 seconds.
For any that are interested, I've taken the alternative route of
implementing a proxy to trap this data before sending it to the AlphaESS
cloud servers.

dry, 2021-10-11:
> 
> For what it's worth, AlphaESS actually emits the data every 10 seconds. For any that are interested, I've taken the alternative route of implementing a proxy to trap this data before sending it to the AlphaESS cloud servers.
> […](#)

@230delphi Would you be able to share some details of how you've done this?

fmontenbruck, 2022-05-08:
+1 for instantaneous values, I too want to control something based on excess power

petosak, 2022-05-08:
I heard from my electrician that Alpha has been promising Modbus over TCP for a long time. That would be best. 2 days ago my stats didn't work all day, even on the web. So in HA I had all day the values from the previous night when it stopped working.

dry, 2022-05-08:
@petosak I recently received this document from Alpha when I enquired about Modbus. I have no idea how to proceed with it but apparently it is available.

[Alpha Modbus Protocol_V1.2320220415005254-1.pdf](https://github.com/CharlesGillanders/homeassistant-alphaESS/files/8646709/Alpha.Modbus.Protocol_V1.2320220415005254-1.pdf)

petosak, 2022-05-08:
@dry I have seen this document and tried to connect with my SMILE T-10 HV but was unsuccessful. 
Modbus RTU (RS485) in chapter 1 works, but I have that busy for communication with the Wallbox. 
Modbus TCP in chapter 2 says it is under development and it is possible that it is not yet enabled for my model.....

dry, 2022-05-08:
@petosak Ah okay. As I said I've not tested it because I don't know how to. The Alpha response did not mention any restrictions or issues but I have a B3 so maybe it's enabled for that model.

Spookster, 2022-07-08:
Currently Im pulling the instantaneous values I need via Alpha ESS to MQTT (load & solar gen)
Im about to put a shelly em to handle the load.
Whats the best and most up to date way currently to grab the solar generation?

CharlesGillanders, 2022-07-08:
I’m not sure there’s only one answer;  I suspect it depends on your needs?  I’m pulling in the metrics I want to track and display in home assistant.  I may eventually want to do some automations with this but for now that’s not my need.  So for me 5 minute snapshots work perfectly well and the public AlphaESS API is fine for that frequency of data and it has the metrics I need fairly easily accessible.

Spookster, 2022-07-08:
Forgive me if Im wrong, but is there a current load sensor in your work?

CharlesGillanders, 2022-07-08:
Yes there is.  In home assistant I expose it as total_load

Spookster, 2022-07-08:
Its an increasing figure? or do I need to do some math to get current value?

[image]

CharlesGillanders, 2022-07-08:
It’s an increasing figure per day with alpha resetting to zero every night at midnight.

Spookster, 2022-07-09:
How can we extract the value of each increase? eg actual load at that point in time?

CharlesGillanders, 2022-07-09:
@xenomorpheus proposed a change some months ago that would allow the python package to pull the second by second instantaneous results which is what I believe you want.  I have not tested this change myself but you are certainly free to try it yourself and see if it does what you need.

https://github.com/CharlesGillanders/alphaess/pull/1

CharlesGillanders, 2022-08-31:
I have good news - thanks to @2000game I have added eight instantaneous values to the integration.

If you are using HACS the update is available now. If you have deployed the integration manually you'll need to download the latest release for yourself.

closed 2022-08-31 by CharlesGillanders

Spookster, 2022-08-31:
Nice work, its all going as expected.
Now I just need to convince you to add PPV1 and PPV2 values ;)

Coordinator.py
```
                inverterdata.update({"Instantaneous Load": invertor["powerdata"]["ppv1"] + invertor["powerdata"]["ppv2"] + invertor["powerdata"]["pmeter_dc"] + invertor["powerdata"]["pbat"] + invertor["powerdata"]["pmeter_l1"] + invertor["powerdata"]["pmeter_l2"] + invertor["powerdata"]["pmeter_l3"] })
                inverterdata.update({"Instantaneous PPV1": invertor["powerdata"]["ppv1"]})
                inverterdata.update({"Instantaneous PPV2": invertor["powerdata"]["ppv2"]})                
                self.data.update({invertor["sys_sn"]: inverterdata})
```
                
enums.py
```    GridIOTotal = "Instantaneous Grid I/O Total"
    Load = "Instantaneous Load"
    PPV1 = "Instantaneous PPV1"
    PPV2 = "Instantaneous PPV2"
```
    

 sensor.py
```        AlphaESSSensorDescription(
        key=AlphaESSNames.Load,
        name="Instantaneous Load",
        native_unit_of_measurement=POWER_WATT,
        device_class=SensorDeviceClass.POWER,
        state_class=SensorStateClass.MEASUREMENT,
    ),
        AlphaESSSensorDescription(
        key=AlphaESSNames.PPV1,
        name="Instantaneous PPV1",
        native_unit_of_measurement=POWER_WATT,
        device_class=SensorDeviceClass.POWER,
        state_class=SensorStateClass.MEASUREMENT,
    ),
        AlphaESSSensorDescription(
        key=AlphaESSNames.PPV2,
        name="Instantaneous PPV2",
        native_unit_of_measurement=POWER_WATT,
        device_class=SensorDeviceClass.POWER,
        state_class=SensorStateClass.MEASUREMENT,
    ),
]
```