CharlesGillanders/homeassistant-alphaESS #8: Alpha-ess update broke charge and discharge sensors

Ralphy140, 2021-10-30:
Hi Charles, 

I am sure you are aware that Aplha-ess was down last night for maintenance and it seems that they have changed something that is related the the charge and discharge sensors. While all other sensors continue to work as expected these two seem to be reporting 0 values today

Ralph's

CharlesGillanders, 2021-10-31:
I wasn't aware - thanks for letting me know.  I will try to see what changes need to be made.

dry, 2021-10-31:
It does seem as if quite a lot has changed/been broken  - the new product information menu item for example, and my predictive power has been nulled out. I had reported an issue where phantom PV generation was appearing overnight and Alpha responded that they found they had been calculating some values incorrectly and that they were working on a fix. AFAIK that has not been resolved yet, or at least the ticket hasn't been closed. In addition, I think I see new keys in the SticsByDay - can't be certain because my notes on this aren't great but I don't remember all of the ones currently being returned.

It would be really useful if we were able to compare our systems to see if the data returned is in the same format for all types. I have a Smile B3.

CharlesGillanders, 2021-10-31:
Right now I'm not able to logon to the alphaess.com site at all - I keep getting a 404 error.  It looks like the official mobile app is also not reporting real time statistics either.  I'm going to leave it a few more hours to see if Alpha will finish up making their changes.  I deleted the downtime message from them - does anyone still have it?  Can you see what time they thought they would be done at?

dry, 2021-10-31:
Yes, the message was:

To provide you with better service, the system of Alpha ESS Monitoring with be maintained from 13:00 to 18:00 (UTC+8, Beijing) on October 30, 2021.

Ralphy140, 2021-10-31:
They were supposed to be done at 6pm(+8gmt)on Saturday, I suspect they have fudged the update on there side, I am still having intermittent issues with the mobile app as well, sometime 500 sometime "no system" errors

dry, 2021-10-31:
It does look as if the whole system has gone down. My Smile has lost connection to the cloud.

Njibhuvgy, 2021-10-31:
I also got this message when attempting to login this morning:

To provide you with better service, the system of Alpha ESS Monitoring will be maintained from 17:00 to 19:00 (UTC+8, Beijing) on October 31, 2021. We are notifying you that you will not be able to login to the Alpha ESS Monitoring website (www.alphaess.com) during the system maintenance. We appreciate your understanding.

Njibhuvgy, 2021-10-31:
My battery failed to charge last night during the Octopus Go period. Hoping they fix it before power hour at 12:00-13:00 today.

CharlesGillanders, 2021-10-31:
Site appears to be back up - the statistics by period API (https://www.alphaess.com/api/Power/SticsByPeriod) which I use for most data is working as expected - with at least one new improvement where the charge from PV and grid is now correctly broken out.  

Unfortunately the charge/discharge figures from the system statistics API (https://www.alphaess.com/api/Statistic/SystemStatistic) appear to still be broken for the current day - all previous days are showing correctly but today is returning zero values.  That's the only API call I've been able to find that correctly returns the right figures for discharge.  It may be that there's just a delay in that coming back up after all of their changes.    More monitoring needed on this to see if it gets fixed or if I need a workaround.

Oh and yes the grid charging setup is broken on my Alpha also - I keep trying to enable it and it keeps getting reset back to disabled.

Ralphy140, 2021-10-31:
Seems like charge/discharge are working again this morning, i wonder if they gave up and rolled back or fixed it but i am seeing stats again. Not tried the charge/discharge function but i did see that it was not working yesterday is i tried to charge when prices were very low but it would just keep reverting to off. 

I am guessing the release team did not have a very good weekend...

closed 2022-07-01 by CharlesGillanders