CharlesGillanders/homeassistant-alphaESS #9: data collection starts at 11am daily

Cushion1, 2021-12-17:
Hi all - a small band of Alpha ESS and Home Assistant users out there so thanks to everyone for getting to this integration, Charles especially. I am new to HA and Solar (battery is commissioned and on, but the final inspection of the solar install is still due so I have started playing with the AlphaESS monitoring to understand it before it all leaps into action).
I decided to try a few charge cycles overnight to get a measure of how long the battery lasts under normal use. I can watch this well in the app, but via the integration the data always starts at 11am... I suspect this is a timezone issue but beyond checking that's set correctly in HA (yes, I'm in Melbourne, Aus and so is the timezone) I have no idea how to fix. 
Any ideas?
[image]
Thanks all! Ian

CharlesGillanders, 2021-12-18:
I'm not passing any kind of time zone details into Alpha so I wonder if there's something that needs to be set on their backend for your account.

I do see that on their web portal under https://www.alphaess.com/BasicSettings/userInfoSettings there are fields for time zone, geographic location etcetera - can you check there and make sure what's listed is correct for you?

Cushion1, 2021-12-24:
Well this seems to have steadied itself - a couple of days starting at 8am followed by complete days. I suspect this is down to unreliability in network connection - I have started with an old laptop and VM over wifi - seems like the worst possible way of running HA. Have bought myself an RPi4 for Christmas and will sit it next to the router... then we'll see how we go!

closed 2021-12-24 by Cushion1