CharlesGillanders/homeassistant-alphaESS #11: Values are Incorrect

Buttercream91, 2022-01-02:
Hey Charles,

Im a day one noob and I am trying to configure Home Assistant to turn off my Aircon when solar production drops below a certain value but the values displayed on the energy tab don't align with my actual generation, consumption or grid feed in.
[image]

[image]

And this is how I have set it up in Home Assistant.

[image]

My battery is the Smile BAT 10.1P/10.3P Integrated EMS V03.
Any help would be appreciated, sorry for the formatting.

Buttercream91, 2022-01-02:
From my logs:
[image]
[image]

Buttercream91, 2022-01-02:
It has occurred to me that these values aren't the instantaneous values, but instead the overall generation, so all working as intended I guess.

Spookster, 2022-01-02:
Im a noob to Charles work too, and Im impressed so far
One other git to look at, is :
https://github.com/DasLetzteEinhorn/AlphaESS_Monitor_Hass
The data is a little more current via MQTT (every minute) and accesses a few different things such as ppv1/2
You could easily run both to get the most current / useful data

Spookster, 2022-01-02:
This is some of the basic data it gets.
Some of the values are unused (and not identified yet)
But as far as installing and update speed, works well.
It doesnt integrate into the energy tab like Charles work though, which is why its complementary to have perhaps both integrations

`{
    "_id": "xxxxxxxxxxxxx",
    "createtime": "2022-01-01 22:01:39",
    "uploadtime": "2022-01-02 11:01:14",
    "sn": "ALXXXXXXXXXXX",
    "ppv1": 3045,
    "ppv2": 1194,
    "ppv3": 0,
    "ppv4": 0,
    "preal_l1": 4071,
    "preal_l2": 0,
    "preal_l3": 0,
    "pmeter_l1": -3439,
    "pmeter_l2": 0,
    "pmeter_l3": 0,
    "pmeter_dc": 0,
    "soc": 100,
    "factory_flag": 0,
    "pbat": 0,
    "sva": 4060,
    "varac": -379,
    "vardc": 0,
    "ev1_power": 0,
    "ev1_chgenergy_real": 0,
    "ev1_mode": 0,
    "ev2_power": 0,
    "ev2_chgenergy_real": 0,
    "ev2_mode": 0,
    "ev3_power": 0,
    "ev3_chgenergy_real": 0,
    "ev3_mode": 0,
    "ev4_power": 0,
    "ev4_chgenergy_real": 0,
    "ev4_mode": 0,
    "poc_meter_l1": 0,
    "poc_meter_l2": 0,
    "poc_meter_l3": 0
}`

[image]

Memorist23, 2022-02-21:
Hey Charles
Would be great if you gave us a guide on the HA setup side.
I’m with the other guys, I seem to be getting g the wrong data as well.

CharlesGillanders, 2022-08-31:
Today I added additional sensors that read the instantaneous values as well as the cumulative figures for the day.  I believe that should give you the data that you are looking for.

closed 2022-08-31 by CharlesGillanders