CharlesGillanders/homeassistant-alphaESS #12: Wallbox readings

Tupsi, 2022-04-04:
I was wondering if it is possible to have a new entity for what is going into the wallbox if one is installed.
The AlphaESS Monitoring shows something there, so I was hoping it exists on the api as well you pull. 

[image]

CharlesGillanders, 2022-04-04:
Just had to lookup what a wallbox is - that's an EV charger is that right?  I don't have one of those so I'd need support from you or someone else who does have one to see how it shows up in the API.  If someone can grab a capture (using whatever proxy software you like) of the web service call between your browser and AlphaESS that would likely be enough.

Tupsi, 2022-04-04:
sorry for the word confusion (thought that is the world wide used name for the thing on the wall which loads your electric car :-) ).
Anyway, if you get me some pointers on how to capture what you need I am glad to give it a go.

Tupsi, 2022-04-04:
[image]

I am still trying to figure this out, but could it be the "EChargingPile"? At least the number fits.

CharlesGillanders, 2022-04-04:
Thank you!  the API result is perfect and I think you are right about what data I need to grab.  Home Assistant doesn't let automations interact with services directly themselves, everything needs to go through a separate library.  So I first need to update that library and then I can update the home assistant integration to actually display the sensor.  I'll try to get time this week to do that.

Tupsi, 2022-04-22:
May I ask where this library is located in an HA installation? I can see you have another repo here on github with it, but I am puzzled on how this alphaess library repo make its way into my HA installation as I can not find any reference in the installation or anywhere on my computer where I installed HA.

CharlesGillanders, 2022-04-22:
It's part of how home assistant works.  If you look in manifest.json for every integration (including this one for AlphaESS) you'll see a requirements entry - that tells home assistant which PyPI modules it should grab to support the integration.  In my case there's only one module that's needed.

HomeAssistant code standards requires all interaction with external services to be done through pypi modules and not directly in code within HA integrations.

Tupsi, 2022-04-22:
found that line thanks, but how does HA knows where to get the library from? Does it get fetched automatically from your repo? Then it should be some sort of config, or do you have to request that your lib be included in HA at some point? Also I went over your lib and I do not see anything which has to be changed there; you grap the whole data field from the json, so the charger entry should already be in there, when this addon requests it, no?

Tupsi, 2022-04-22:
but something is missing, just tried it out again and finally found something more descriptive in the logs
```
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 382, in async_add_entities
    await asyncio.gather(*tasks)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 614, in _async_add_entity
    await entity.add_to_platform_finish()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 799, in add_to_platform_finish
    self.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 532, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 570, in _async_write_ha_state
    state = self._stringify_state(available)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 538, in _stringify_state
    if (state := self.state) is None:
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 410, in state
    value = self.native_value
  File "/config/custom_components/alphaess/sensor.py", line 139, in native_value
    return  invertor["statistics"]["EChargingPile"][index]
TypeError: 'float' object is not subscriptable
```

Tupsi, 2022-04-22:
hah, reading helps! Found the error, it seems to work now.

Tupsi, 2022-04-22:
2nd try, this works. I made a pull request for you to check out.
https://github.com/CharlesGillanders/homeassistant-alphaESS/pull/15

CharlesGillanders, 2022-04-22:
PyPi is an existing large 3rd party repository of Python modules – HA expects to pull any required module from PyPi.

So to get a helper module accessible to HA you need to get it added to PyPi.  Thankfully that’s all relatively straightforward but it means the functionality doesn’t come from my Repo it comes from the code that I previously uploaded to PyPi, any time I need to make a change to that library I need to submit the revised version to PyPi.

Charles
Charles Gillanders | Personal Account
***@***.******@***.***>
Please consider the environment before printing this e-mail

closed 2022-07-01 by CharlesGillanders