CharlesGillanders/homeassistant-alphaESS #13: Using the API to change AlphaESS settings

mikeew, 2022-04-04:
I'm wondering if it is possible to alter AlphaESS settings, specifically Discharge Time Control and Discharging Cutoff SOC (%)?
At the moment I'm using your integration to read sensors and that is working great.

The scenario that I'd like to automate is:

1. EV starts charging
2. The AlphaESS stops discharging to the load
3. EV stops charging
4. AlphaESS enables discharging to the load

I don't want the AlphaESS to charge the EV.
A possible automation would be to 

[image]

CharlesGillanders, 2022-04-04:
Enabling switches and control of AlphaESS settings is on my list of things to do.  

I'm really far behind on finishing the necessary edits for getting an accepted pull request for this as an official Home Assistant integration.  Unfortunately for the first pull of an HA integration the dev team want only a single Home assistant domain which basically means sensors first and then add switches later.

It can certainly be done and anyone who wants to is very welcome to take this code and fork it to add switches and I would happily pull that in later once I'm done with getting onto the official integrations list.

Spookster, 2022-06-07:
Has anyone managed to achieve this, or provide some hints on how it can be done?

eviltabouleh, 2022-06-22:
I'm keen for this , wish I could help out some way other than providing support and testing 
I'd be happy with some sort of seperate script if the HA dev team are slowing this down 
I've been trying to learn python but I'm more of a hardware guy

> Has anyone managed to achieve this, or provide some hints on how it can be done?

DavidqStokes, 2022-08-23:
I've recently installed an AlphaESS home battery and was delighted to see an integration already available. Well done and thank you Charles.

I also have the Forecast.Solar integration and each evening I now have some HA automations that run, which look at the AlphaESS SOC and the forecast solar PV generation the next day. Currently they send me a notification to either (a) charge the battery overnight, when SOC and solar PV forecast are both low (we have cheap power between 00.30 and 04.30); (b) allow the battery to continue discharging overnight when the SOC and solar PV forecast are both good or (c) neither charge or discharge (i.e. allow the house to use cheap grid power overnight, and save some power in the battery for the next day).

(This also depends on whether or not we need to charge the EV, which I could also determine based on the SOC of the vehicle - also available to HA).

It would be ideal to be able to control the AlphaESS from HA (either automatically if the car charge point is plugged in - MiEnergy Zappi status is also available in HA, or via actionable notification).

So basically, a thumbs up from me for extending the API to allow this, and hopefully further encouragement for Charles for all the great work he's done and is doing on this.

FrankC-BEL, 2022-09-16:
#All of the above!  Great implementation and indeed being able to change the parameters (battery timing & SOC, grid loading, ...) would make it even better.

CharlesGillanders, 2022-11-07:
@matt-oneill has implemented this functionality so it is now possible to set battery charge and discharge settings from HomeAssistant.  Release 0.3.0 has just been published with this new feature.

closed 2022-11-07 by CharlesGillanders