CharlesGillanders/homeassistant-alphaESS #14: Time til SOC = 0

Spookster, 2022-04-05:
Apologies if this is the wrong place to post, but I trying here first as there are at least a couple of 'alpha' experts here that may have done this
Feel free to delete otherwise.
Has anyone considered the math to get the battery run time remaining? ie, Assuming load is higher than generation, how long will it last.
We have all of the inputs via Charles code, but assuming thats all we need, what would the equation be to get a rolling 'time till SOC = x%'
It will obviously be longer or shorter less depending on current load. Thoughts?

CharlesGillanders, 2022-04-05:
I love the idea but I'm not sure we have the data we would need to implement it.  I'm no expert but I believe that batteries discharge on a curve rather than a straight line.  There's some information I found with a quick search [https://www.mpoweruk.com/performance.htm](https://www.mpoweruk.com/performance.htm) 

I guess we could try to derive a crude estimate for the portion of the curve that is more linear but once the discharge hits that more rapid drop off I don't know what math would be needed to calculate that.

Spookster, 2022-04-05:
I guess at its simplest form without putting it through the complicator, would be the known battery values.

eg
0 minutes: 100%
1 hour: 90%
2 hours: 80%

Based on battery value battery (x) hr - battery (now) hour and a load comparison, you could predict a basic lineal value (although we know its not)

Spookster, 2022-04-07:
Using 2 sensors, Ive ended up here so far as a proof of concept.
Now I just have to work on fine tuning and making them useable :)

`- platform: statistics
    name: "House Load Av thirty mins"
    entity_id: sensor.solar_load_total
    state_characteristic: mean
    max_age:
      minutes: 30

  - platform: statistics
    name: "House Battery i/o thirty mins"
    entity_id: sensor.solar_battery_i_o
    state_characteristic: mean
    max_age:
      minutes: 30`

`{% set state = states('sensor.house_battery_i_o_thirty_mins') %}
{% set load = states('sensor.house_load_av_thirty_mins') %}
{% set battery = states('sensor.solar_battery_soc') %}
{% if float(state) == 0 or float(battery) == 100 %}
Battery is full. Current average House load {{load}}
{% elif float(state) < 0  %}
Battery is charging. Current average House load {{load}}
{% elif float(state) > 0  %}
Battery is discharging. Current average House load {{load}}
{% endif %}
Battery run time at current load: {{ 9 * (float(battery)/100)/(float(load) /1000)  }}`

Spookster, 2022-04-13:
If anyone is interested, I ended up doing it as below (Developer > template)
Im new to jinja, so not entirely sure its correct. If anyone has any comments on format, flaws or logic, let me know

Otherwise. Carry on :)

```
{## Set Battery KW ##}
{% set battery_kw = 9 %}
{## Set UPS reserve (percent) ##}
{% set battery_ups = 15 %}
{## Battery - UPS ##}
{% set battery_capacity = float(battery_kw - (battery_ups / battery_kw)) | round(2) %}
{% set battery_soc = float(states('sensor.solar_battery_soc')) -%}

    {% set load = int(states('sensor.solar_load_total')) -%}
    {% set solar = states('sensor.solar_generation')| int(default=0) -%}
    {% set battery_i_o = states('sensor.solar_battery_i_o')| int(default=0) -%}

{%- if (load-solar) < 0 or battery_i_o < 0 -%}
    {## Battery Charging or at max  - show current capacity ##}
    {% set runtime=(battery_capacity*(battery_soc/100)) %}
{%- else -%}
    {## load > solar, reduce runtime ##}
    {%- set runtime=(battery_capacity*(battery_soc/100)/((load-solar)/1000)) -%}
    {%- if (runtime) > battery_capacity  -%}
      {## Only show max battery ##}
      {% set runtime=(battery_capacity*(battery_soc/100)) %}
  {%- endif -%} 
{%- endif -%} 

{% set hours = runtime | float -%}
{% set minutes = ((hours % 1) * 60) | int %}
{% set hours = (hours - (hours % 1)) | int %}

{%- if (battery_i_o) < 0 -%}
      {## battery charging ##}
    {% set hours_to_full = ((9-(9*(battery_soc/100))) /(battery_i_o/1000)*-1) | float -%}
    {% set minutes_to_full = ((hours_to_full % 1) * 60) | int %}
    {% set hours_to_full = (hours_to_full - (hours_to_full % 1)) | int %}
    {% set battery_to_full = '%02i:%02i hrs'%(hours_to_full, minutes_to_full) %}
{%- else -%}
  {## battery full / discharging ##}
  {%- set battery_to_full = 'n/a' %} 
{%- endif -%}

==============
Load: {{ load }}w
==============
Solar: {{ solar }}w
==============
{% if battery_i_o == 0 and (load-solar) < 0 -%}
Exporting to Grid: {{ ((load-solar)) * -1 }}w
{%- else -%}
Not exporting to grid
{%- endif %} 
Load-Solar = {{(load-solar)}}
{% if battery_i_o == 0 -%}
Battery is full 
{% elif battery_i_o < 0  -%}
Battery is charging
{% elif battery_i_o > 0 -%}
Battery is discharging
{% endif -%}

==============
Battery kwh: {{ battery_capacity }}
Battery SOC: {{ battery_soc }} %
Battery i/o: {{ battery_i_o }}
Battery to full: {{ ((battery_capacity*100/battery_capacity) - battery_soc) | round(2)}} %
Time to fill battery: {{ battery_to_full }}
Battery kwh to full: {{ (battery_capacity - (battery_capacity*(battery_soc / 100)))|round(2) }}
Battery run time at current load: {{ '%02i:%02i'%(hours, minutes) }} hrs
Battery empty at: {{ ((runtime | int *3600) + (now().timestamp() | int)) | timestamp_custom("%H:%M") }}
==============

```
My sensor code in /config/configuration.yaml is :

```
template:
  - sensor:
      - name: "Solar Battery Runtime"
        state: >
          {% set battery_kw = 7.33 %}
          {% set load = states('sensor.house_load_av_15_mins')|int(default=0) %}
          {% set solar = states('sensor.house_solar_gen_15_mins')| int(default=0) %}
          {% set battery_i_o = states('sensor.house_battery_i_o_15_mins')| int(default=0) %}
          {% set battery_soc = float(states('sensor.solar_battery_soc')) %}
          {% if (load-solar) < 0 or battery_i_o < 0 %}
          {% set runtime = (battery_kw*(battery_soc/100)) %}
          {% else %}
          {% set runtime=(battery_kw*(battery_soc/100)/((load-solar)/1000)) %}
          {% if runtime > battery_kw %}
          {% set runtime=(battery_kw*(battery_soc/100)) %}
          {% endif %} 
          {% endif %}
          {% set hours = runtime|float %}
          {% set minutes = ((hours % 1)*60)| int %}
          {% set hours = (hours-(hours % 1))| int %}
          {{ '%02i:%02i'%(hours, minutes) }}
```

230delphi, 2022-04-13:
nice work! I thought my charging logic was overly complex.. turns out this stuff is hard!

likewise, nothing to see here. carry on :)

```
  process_battery_charge_rules:
    alias: Process Battery charging rules
    sequence:
      - variables:
          # daily_load: 15
          daily_name: "sensor.{{now().strftime('%A')}}_load_5weekstat"
          daily_load: "{{states(daily_name)|float(-1)}}"
          battery_min_cap: 10
          battery_total_kwh: 11.4
          battery_hourly_charge_rate: 4.0
          battery_starting_SOC: "{{states('sensor.alphaess1_state_of_charge')}}|float(0)"
          # TODO adjust based on hourly usage rate and start of charging time? 
          # eg.. in winter, power from early am - later sun rise; 
          # maybe 10 + x * hours to sun up {{ ((as_timestamp(state_attr('sun.sun','next_rising')) - as_timestamp(now())) /60/60) }}
          # in summer charge from same early am to earlier sun rise?
          battery_minimum_kwh: "{{battery_total_kwh|int | multiply(10) / 100}}"
          forecast_capacity_gap: >
            {{min(daily_load|int 
              - states('input_number.solar_was_forecasted_for_today')|int,battery_total_kwh)}}
          battery_target_percent: >
            {{ max((max(forecast_capacity_gap|float,0) 
              / battery_total_kwh|float) * 100, battery_min_cap|float) | round(1, ceil)}}
          # need to allow extra % to allow start from 0 and second battery to catch up at half rate. 
          battery_target_charge_minutes: >
            {{(((battery_target_percent|round(0) 
              - states('sensor.alphaess1_state_of_charge')|float) * battery_total_kwh|float / 100) 
                / battery_hourly_charge_rate|float * 60 * 1.11)|int}}
      - choose:
        - alias: "If Battery needs to be charged"
          conditions: "{{ battery_target_percent|float > states('sensor.alphaess1_state_of_charge')|float }}"
          sequence:
            - service: script.enable_battery_charge_custom
              data_template:
                minimum_duration: "{{battery_target_charge_minutes|int}}"
                battery_high_cap: "{{battery_target_percent|int}}"
                start_hour: -1
            - wait_template: "{{ battery_target_percent|float <= states('sensor.alphaess1_state_of_charge')|float(0) }}"
              timeout: 
                minutes: "{{battery_target_charge_minutes|int}}" 
              continue_on_timeout: true
            - service: script.disable_battery_charge_now
            - delay: 11 # Ensure update from system on timeout
            - service: input_number.set_value
              target:
                entity_id: "input_number.daily_grid_kwh_to_battery"
              data:
                value: "{{((states('sensor.alphaess1_state_of_charge')|float(0) - battery_starting_SOC|float(0)) * battery_total_kwh|float(0) / 100)|round(1)}}"
            - alias: "Test charge reached target"
              choose:
                - alias: "Charge < expected"
                  conditions: "{{ states('sensor.alphaess1_state_of_charge')|float(0) <= (battery_target_percent|float - 0.5) }}" # add 0.5% tolerance
                  sequence:
                    - service: script.turn_on
                      entity_id: script.send_debug_message
                      data:
                        variables:
                          message: "Charging Failed: LOADT:{{daily_load}} SOC- WAS: {{battery_starting_SOC}} :NOW: {{states('sensor.alphaess1_state_of_charge')}} Target: {{battery_target_percent}} Time- Est: {{battery_target_charge_minutes}} :Taken: {{ battery_target_charge_minutes|int + 5 - (wait.remaining|int/60) | round(0, ceil)}}"
              # ELSE As expected
              default:
                - service: script.turn_on
                  entity_id: script.send_debug_message
                  data:
                    variables:
                      message: "Charging Complete: LOADT:{{daily_load}} SOC- WAS: {{battery_starting_SOC}} :NOW: {{states('sensor.alphaess1_state_of_charge')}} Target: {{battery_target_percent}} Time- Est: {{battery_target_charge_minutes}} :Taken: {{ battery_target_charge_minutes|int + 5 - (wait.remaining|int/60) | round(0, ceil)}}"
        # ELSE (ensure it is disabled)
        default:
          - service: script.disable_battery_charge_now
          - service: notify.pushover_notify
            data_template:
              message: "Not Charging: {{states('input_number.solar_was_forecasted_for_today')}} : SOC: {{states('sensor.alphaess1_state_of_charge')}} target: {{battery_target_percent}}"

```

closed 2022-06-07 by Spookster