CharlesGillanders/homeassistant-alphaESS #16: manifest is not using most recent alphaess libarary

Tupsi, 2022-04-23:
After you explained the whole PyPi thing and that this addon has to rely on a lib to get the data I was looking around a bit and found your other repo which has a version of 0.0.3 since February.

Yet the manifest of this addon states 

 "requirements": ["alphaess==0.0.2"],

checking the container I can see that there realy is 0.0.2 (instead of 0.0.3) installed. Is there a reason for this are should I change the value in the manifest to 0.0.3?

CharlesGillanders, 2022-04-23:
0.03 is the version I’m working on now (very slowly) to try to get this all submitted into the mainline home assistant.  There’s new code that I’m writing for that which needs the changes to the library.  The code in this custom component isn’t using that newer library so it should stay at 0.02.

Charles Gillanders | Personal Account
***@***.******@***.***>
Please consider the environment before printing this e-mail

Tupsi, 2022-04-23:
ah thanks for explaining!

closed 2022-04-23 by Tupsi