CharlesGillanders/homeassistant-alphaESS #17: Intermittent data

KortgatGH, 2022-05-17:
[image]

Anybody else experiencing intermittent Alpha data recently? Mine started playing up around the time that the Alpha web interface was updated, but it was also around the time that HA core-2022.5 was released.

Spookster, 2022-05-17:
No issues noted here.
A question though - Batt limit? hard coded  or extracted from data elsewhere?

KortgatGH, 2022-05-17:
Batt Limit is just a local constant so that I can draw a line on the graph.

KortgatGH, 2022-05-27:
Could my problem be related to this error?

"This error originated from a custom integration.

Logger: custom_components.alphaess
Source: helpers/update_coordinator.py:223
Integration: Alpha ESS (documentation)
First occurred: May 25, 2022, 18:11:26 (31 occurrences)
Last logged: 05:33:20

Error fetching alphaess data:"

closed 2022-08-31 by CharlesGillanders