CharlesGillanders/homeassistant-alphaESS #19: Request for info

bavala3010, 2022-06-30:
Hi, this is not an issue but a request for info.

I am considering buying an Alpha ESS SMILE-T10-HV but only if it can be integrated in Home Assistant.
More info on the device: https://alphaess.com/smile-t10-hv-10kw-residential-energy-storage-system

2 questions:

- I imagine that with the code written by @CharlesGillanders you can easily READ all parameters/values of the sensors.
Question 1: Can you also CHANGE settings eg. battery discharge time control ?

- According to the information in below link, (some) Alpha ESS devices have a port with a ModBus protocol: 
https://www.alpha-ess.de/images/downloads/handbuecher/AlphaESS-Handbuch_SMILET30_ModBus_RTU_V17-DE.pdf
HomeAssistant can also communicate with ModBus. Below a small part of the modbus definition for my car charger.
Question 2: Did anyone already try to communicate via ModBus to the Alpha ESS from Home Assistant?

```
modbus:
  - name: EV charger
    type: tcp
    host: 192.168.1.4
    port: 502
    sensors:
      - name: charger_name
        slave: 200
        address: 100
        count: 17
        data_type: string
```

petosak, 2022-06-30:
Modbus RTU is different than Modbus TCP.....

bavala3010, 2022-07-01:
At which frequency do the sensors from this integration update in Home Assistant? Every minute, every 5 minutes or ... ?
I will need to have values every minute to see how much solar panels are producing and adjusting the car charger in function of that.

CharlesGillanders, 2022-07-01:
Right now it is set to every 5 minutes.  

There is a line in  __init__.py which sets up the refresh frequency

SCAN_INTERVAL = timedelta(minutes=5)

You could easily change that if you needed to.

Tupsi, 2022-07-12:
>   Question 2: Did anyone already try to communicate via ModBus to the Alpha ESS from Home Assistant?

I already have a T10 since march, but was unable to find anyone providing a working modbusTCP implementation so far. There is one doing it for RTU (which does not help our case) or people like Charles here polling the web services (which results in null data when they are down again like atm :-) ). As my programming skills are limited I have to be patient until the lucky day a programmer decides to buy a T10 and does it for all of us. I am not even sure that the modbus TCP part is actually already working as it is still labeled "in development" in the document. I already asked the alphaess support about that, but got no response.

petrkrauscz, 2022-07-13:
> edit: I just realised you are referring to an old version of that
> document, I have a v2 here which has TCP in it from last year. But I am
> struggeling with getting a setup out of it for testing it in ha.

Hello,
I have T10-HV and Modbus TCP working for me.
Hele is part of my configuration:

- name: "Storion"
    type: tcp
    host: XXX.XXX.XXX.XXX
    port: 502
    sensors:
      - name: Storion String1 W
        unit_of_measurement: W
        slave: 85
        address: 1055
        data_type: int32
      - name: Storion String2 W
        unit_of_measurement: W
        slave: 85
        address: 1059
        data_type: int32
      - name: Storion Battery
        unit_of_measurement: '%'
        slave: 85
        address: 258
        data_type: int16
        scale: 0.1

Petr

petosak, 2022-07-13:
> Hello, I have T10-HV and Modbus TCP working for me. Hele is part of my configuration: - name: "Storion" type: tcp host: XXX.XXX.XXX.XXX port: 502 sensors: - name: Storion String1 W [...]

WOW. You're right!!!! It works now!!!!
Thanks

Tupsi, 2022-07-13:
yep, experimenting with it as we type. It is a PITA to find the right values for what you want, but in the end it does the job. Still trying to figure out some of the correct units of measurements and scale factors, but will get there in the end. So thanks @petrkrauscz for the snippet! The slave: 85 was the kicker here. Did not understand I need to set that.

petosak, 2022-07-13:
> I received this documentation from alpha ess support. Important is "translate" values from hex to integer: https://www.easycalculation.com/hex-to-integer.php In documentation is modbus address 0x55 = 85. From documentation is possible to identify address, units and scale also. I add only some of sensors and 90% of them working for me. Petr st 13. 7. 2022 v 10:39 odesílatel Tupsi ***@***.***> napsal:
> […](#)

I use standart MS Calc for conversion:
[image]

petosak, 2022-07-13:
My current setup (still in development):
```
- name: Storion
  type: tcp
  host: x.x.x.x
  port: 502
  sensors:
    # --- GRID METER ---
    - name: Storion Grid Meter - Total Energy Feed to Grid
      unit_of_measurement: kWh
      slave: 85
      address: 16
      data_type: uint32
      scale: 0.01
      precision: 1
      scan_interval: 60
    - name: Storion Grid Meter - Total Energy Consume From Grid
      unit_of_measurement: kWh
      slave: 85
      address: 18
      data_type: uint32
      scale: 0.01
      precision: 1
      scan_interval: 60
    - name: Storion Grid Meter - L1 Power
      unit_of_measurement: W
      slave: 85
      address: 27
      data_type: int32
      scan_interval: 5
    - name: Storion Grid Meter - L2 Power
      unit_of_measurement: W
      slave: 85
      address: 29
      data_type: int32
      scan_interval: 5
    - name: Storion Grid Meter - L3 Power
      unit_of_measurement: W
      slave: 85
      address: 31
      data_type: int32
      scan_interval: 5
    - name: Storion Grid Meter - Total Power
      unit_of_measurement: W
      slave: 85
      address: 33
      data_type: int32
      scan_interval: 5

    # --- INVERTER ---
    - name: Storion Inverter - L1 Power
      unit_of_measurement: W
      slave: 85
      address: 1030
      data_type: int32
      scan_interval: 5
    - name: Storion Inverter - L2 Power
      unit_of_measurement: W
      slave: 85
      address: 1032
      data_type: int32
      scan_interval: 5
    - name: Storion Inverter - L3 Power
      unit_of_measurement: W
      slave: 85
      address: 1034
      data_type: int32
      scan_interval: 5
    - name: Storion Inverter - Total Power
      unit_of_measurement: W
      slave: 85
      address: 1036
      data_type: int32
      scan_interval: 5
    - name: Storion Inverter - Backup Total Power
      unit_of_measurement: W
      slave: 85
      address: 1050
      data_type: uint32
      scan_interval: 5
    - name: Storion Inverter - String 1 Power
      unit_of_measurement: W
      slave: 85
      address: 1055
      data_type: int32
      scan_interval: 10
    - name: Storion Inverter - String 2 Power
      unit_of_measurement: W
      slave: 85
      address: 1059
      data_type: int32
      scan_interval: 10
    - name: Storion Inverter - Temperature
      unit_of_measurement: "℃"
      slave: 85
      address: 1077
      data_type: uint16
      scale: 0.1
      precision: 1
      scan_interval: 60
    - name: Storion Inverter - Total PV Energy
      unit_of_measurement: kWh
      slave: 85
      address: 1086
      data_type: uint32
      scale: 0.1
      precision: 1
      scan_interval: 60

    # --- BATTERY ---
    - name: Storion Battery - SoC
      unit_of_measurement: "%"
      slave: 85
      address: 258
      data_type: uint16
      scale: 0.1
      precision: 1
      scan_interval: 60
    - name: Storion Battery - Max Cell Temperature
      unit_of_measurement: "℃"
      slave: 85
      address: 272
      data_type: uint16
      scale: 0.1
      precision: 1
      scan_interval: 60
    - name: Storion Battery - Capacity
      unit_of_measurement: kWh
      slave: 85
      address: 281
      data_type: uint16
      scale: 0.001
      precision: 2
      scan_interval: 60
    - name: Storion Battery - Power
      unit_of_measurement: W
      slave: 85
      address: 294
      data_type: int16
      #      scale: 0.1
      #      precision: 1
      scan_interval: 5
    - name: Storion Battery - Charge Energy
      unit_of_measurement: kWh
      slave: 85
      address: 288
      data_type: uint32
      scale: 0.1
      precision: 1
      scan_interval: 60
    - name: Storion Battery - Discharge Energy
      unit_of_measurement: kWh
      slave: 85
      address: 290
      data_type: uint32
      scale: 0.1
      precision: 1
      scan_interval: 60
    - name: Storion Battery - Remaining Time
      unit_of_measurement: m
      slave: 85
      address: 295
      data_type: uint16
      #      scale: 0.1
      #      precision: 1
      scan_interval: 60
```

Tupsi, 2022-07-13:
Thanks for that! This is really helpfull! Do you have the Alpha EV Charger as well by chance? I am trying to find the variables for this in the documentation but so far, no luck. It should be somewhere there, as it gets pushed to the web gui, so a value for it should be in the table. Just do not see it.

bavala3010, 2022-07-13:
Many thanks. I am really happy to see these positive results to integrate the Alpha ESS into HA.

As far as I can see from images, the Alpha ESS SMILE-T10-HV has different RJ45 ports. One of them is supposed to be called "LAN". One of the other RJ45 ports is used for Modus RTU.
Is that right?

If so, simply plug in a RJ45 network cable into the Alpha ESS in the port called "LAN", connect it to your router, it will get an ip address to use as "host" in the configuration.yaml.

Tupsi, 2022-07-13:
> Many thanks. I am really happy to see these positive results to integrate the Alpha ESS into HA.
>
> As far as I can see from images, the Alpha ESS SMILE-T10-HV has different RJ45 ports. One of them is [...]

Yes, that is true. I have to meters connected somehow with modbus to my T-10 and they show up in the "Grid Meter" and "PV Meter" variables from the documentation. The ev charger is somehow talking to the battery over RTU as well while the LAN port I use to get it connected to my LAN an access now with the ha modbus options @petrkrauscz and @petosak where so nice to explain here in lengthy detail. Which brings me to the owner of this repo: @CharlesGillanders SORRY for hijacking your repo for this as this is totally not the right place, so I would understand if you are annoyed by what we do here as your projects is all about grabbing the data from the cloud of alpha.

petosak, 2022-07-13:
> Thanks for that! This is really helpfull! Do you have the Alpha EV Charger as well by chance? I am trying to find the variables for this in the documentation but so far, no luck. It should be somewhere there, as it gets pushed to the web gui, so a value for it should be in the table. Just do not see it.
> 
> Also I am trying to figure out which variable has the string for the current software version. There a lots of it in the table, but most of them return zero and the ones which do give something back do not look like a version number to me.

Unfortunately, I have a different brand of wallbox. I didn't even see any modbus entries for the wallbox in the modbus table...

CharlesGillanders, 2022-07-18:
> > Many thanks. I am really happy to see these positive results to integrate the Alpha ESS into HA.
> 
> > 
> 
> > As far as I can see from images, the Alpha ESS SMILE-T10-HV has different RJ45 ports. One of them is supposed to be called "LAN". One of the other RJ45 ports is used for Modus RTU. Is that right?
> 
> > 
> 
> > If so, simply plug in a RJ45 network cable into the Alpha ESS in the port called "LAN", connect it to your router, it will get an ip address to use as "host" in the configuration.yaml.
> 
> 
> 
> Yes, that is true. I have two meters and my ev charger connected with modbus RTU to my T-10 and they show up in the "Grid Meter" and "PV Meter" variables from the documentation. The ev charger should be somewhere as well, but that one I have not found yet. The LAN port I use to get it connected to my LAN an access now with the ha modbus options @petrkrauscz and @petosak where so nice to explain here in lengthy detail. Which brings me to the owner of this repo: @CharlesGillanders SORRY for hijacking your repo for this as this is totally not the right place, so I would understand if you are annoyed by what we do here as your projects is all about grabbing the data from the cloud of alpha.

I'm fascinated by what you are managing to do - I had no idea some of the Alpha ESS systems had such a relatively easy mechanism for local query!

Spookster, 2022-10-21:
> Yes, that is true. I have two meters and my ev charger connected with modbus RTU to my T-10 and they show up in the "Grid Meter" and "PV Meter" variables from the documentation. The ev charger [...]

Are you able to expand on if and how you are getting local variables from the system?
It would be good to have a definitive way of doing that if Alpha are determined to make web queries impossible

Tupsi, 2022-10-31:
I asked AlphaEss in a ticket to include the ev charger. Currently it is missing from the modbusTCP implementations. All they told me so far was that they have forwareded to their R&D department. That was in july after I got this running with the help from @petosak here. I requested an updated on the issue last week, but so far no dice.

closed 2023-11-14 by CharlesGillanders