CharlesGillanders/homeassistant-alphaESS #28: Stopped Updating

KortgatGH, 2022-08-27:
Since this afternoon (27 Aug) my Alpha ESS sensors stopped updating. I tried to uninstall and reinstall the integration, and now I get the following after entering the correct login credentials:
[image]
Anybody else experiencing problems?

DavidqStokes, 2022-08-27:
My AlphaESS is also showing as Unavailable. The AlphaESS app is working fine.

[image]<https://user-images.githubusercontent.com/49630727/187034521-205fc87e-e0d8-4532-9f35-9fc310740def.png>

KortgatGH, 2022-08-27:
If only we could interface directly to our inverters and not rely on the Alpha server. I wonder if Alpha made a small change that broke the Home Assistant integration without affecting their own mobile and web apps.

CharlesGillanders, 2022-08-27:
I’m awY this weekend, I’ll take a look at whatever changed as soon as I can.

DavidqStokes, 2022-08-27:
HA logs show the following:

2022-08-27 15:33:58.273 WARNING (MainThread) [homeassistant.config_entries] Config entry 'name.name@domain.co.uk' for alphaess integration not ready yet; Retrying in background

(Email address changed)

Spookster, 2022-08-27:
Same issue here, however the values are still being returned by DasLetzteEinhorn's add in:
https://github.com/DasLetzteEinhorn/AlphaESS_Monitor_Hass

So something odd has happened with Charles values and not DasLetzteEinhorn

tbgoose, 2022-08-27:
@Spookster same for me, Alpha ESS to MQTT add-on by MarioCake still returning values (DasLetzteEinhorn addon).

Thanks for your work @CharlesGillanders, in your own time :)

Spookster, 2022-08-28:
I use a bit of a failsafe in using both integrations
eg:
`    - name: "Const Solar Battery SOC"
      unit_of_measurement: "%"
      device_class: battery
      state: >
        {%if states('sensor.alxxxxxxxxxxxxxxxx_battery_soc') == 'unavailable' %}
        {{ states('sensor.solar_battery_soc') |int   }}
        {%else%}
        {{ states('sensor.alxxxxxxxxxxxxxxxx_battery_soc') |int   }}
        {%endif%}
`
If both are unavailable, Im screwed :p

matt-oneill, 2022-08-28:
looks like an issue with the date format
[image]

CharlesGillanders, 2022-08-28:
thanks to @matt-oneill I have found the error.  Date format change on the part of AlphaEss.  I'll get a fix built today.

KortgatGH, 2022-08-28:
Thanks @CharlesGillanders, what would us non-programmer types do without you gurus!?

DavidqStokes, 2022-08-28:
> Thanks @CharlesGillanders, what would us non-programmer types do without you gurus!?

Seconded!

CharlesGillanders, 2022-08-28:
I have updated the AlphaEss library to fix the date format issue and I've tried to code it to catch future similar changes.  Time will tell.....

If you are using the HACS method of installing the integration then you will need to go into HACS and update the integration (if HACS hasn't updated the version yet you can force it to be updated using the hamburger menu on the integration in HACS)

If you installed manually you'll need to download the updated integration code from github and restart home assistant,

DavidqStokes, 2022-08-28:
I have installed the update via HACS (it had appeared) and everything is now working fine. Excellent work Charles and thank you very much for your quick attention to this.

closed 2022-08-28 by CharlesGillanders

KortgatGH, 2022-10-09:
Just me or are more people getting "unavailable" responses? Looks like it started around 09:00 UTC today

Tupsi, 2022-10-10:
hmm, I just "updated" via HACS and now have a broken version because of the changed date. Shouldnt the latest update fix it in any case? I will try to completly delete and reinstall with HACS, maybe that fixes it. Just wanted to let you know that the update process isn't that smooth as hoped.

CharlesGillanders, 2022-10-10:
@Tupsi - updating to the latest version in HACS will fix the integration but I need to actually write the fix first.  I'll update the open GitHub issue when I have done so.

Tupsi, 2022-10-10:
and now I remember that you do some stuff with that external lib stuff and that needs an update as well, so I will run and get that by hand installed in the hope it fixes it.

Tupsi, 2022-10-10:
ok, can someone please clarify if it is currently possible to install a working version or not, because that is still confusing for me to read here, as DavidqStokes above wrote he reinstalled via HACS the latest version and it now works. I have tried that and are now stuck getting my credentials back in (because I removed it completly before the fresh install). If it needs some manuall poking I can do that as I have to issues with going into touch with the command line, just need some pointers where to do/replace what please.

Buttercream91, 2022-10-10:
@Tupsi - Charles is still working on the Fix, he has not yet released the working update.

Tupsi, 2022-10-10:
> @Tupsi - Charles is still working on the Fix, he has not yet released the working update.

thanks for the clarification, then I will wait patiently. Is that maybe another/new problem with the dates? Because I remember we already had that a few weeks ago (and fixed). Because as I said, it was running fine here until me stupid pressed update this morning :-)

Buttercream91, 2022-10-10:
I am not sure, when people started reporting the issue my instance was working fine until I updated to home assistant Core 2022.10.2, and then I got the same error as everyone else.

Tupsi, 2022-10-10:
yeah, the v10 broke a few things here as well. now it is whining about me having portainer installed and refuses to update. Silly.

CharlesGillanders, 2022-10-10:
I've just pushed the latest fix which should hopefully prevent future date related API issues....

KortgatGH, 2022-10-10:
Thanks again for the quick response @CharlesGillanders, you're a star!