CharlesGillanders/homeassistant-alphaESS #30: Integration uses deprecated entity_type

dan-s-github, 2022-08-29:
I discovered that the source of these log entries must come from this integration as the disappeared when integration was broken over the weekend

```
Integration uses deprecated entity_type

Logger: homeassistant.helpers.frame
Source: helpers/frame.py:74 
First occurred: 7:10:28 PM (11 occurrences) 
Last logged: 7:10:28 PM

Detected code that uses str for device registry entry_type. This is deprecated and will stop working in Home Assistant 2022.3, it should be updated to use DeviceEntryType instead. Please report this issue.
```

JohnnyZeta, 2022-08-29:
Same here, I was wondering why the integration was broken

CharlesGillanders, 2022-08-31:
I have published a new version today which switches to using DeviceEntryType if you upgrade to the v0.2.1 release then those warnings should stop showing in your logs.

JohnnyZeta, 2022-08-31:
I have just updated via HACS, and I found an error about the configuration flow ("Failed to load configuration flow: undefined") when I try to add the integration. This is the log that comes from /config/logs:

```
Logger: aiohttp.server
Source: custom_components/alphaess/config_flow.py:69
Integration: Alpha ESS (documentation, issues)
First occurred: 17:10:48 (1 occurrences)
Last logged: 17:10:48

Error handling request
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/aiohttp/web_protocol.py", line 435, in _handle_request
    resp = await request_handler(request)
  File "/usr/local/lib/python3.10/site-packages/aiohttp/web_app.py", line 504, in _handle
    resp = await handler(request)
  File "/usr/local/lib/python3.10/site-packages/aiohttp/web_middlewares.py", line 117, in impl
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 60, in security_filter_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 94, in forwarded_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 28, in request_context_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 82, in ban_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 236, in auth_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/view.py", line 136, in handle
    result = await result
  File "/usr/src/homeassistant/homeassistant/components/config/config_entries.py", line 145, in post
    return await super().post(request)
  File "/usr/src/homeassistant/homeassistant/components/http/data_validator.py", line 73, in wrapper
    result = await method(view, request, data, *args, **kwargs)
  File "/usr/src/homeassistant/homeassistant/helpers/data_entry_flow.py", line 70, in post
    result = await self._flow_mgr.async_init(
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 222, in async_init
    flow, result = await task
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 249, in _async_init
    result = await self._async_handle_step(flow, flow.init_step, data, init_done)
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 359, in _async_handle_step
    result: FlowResult = await getattr(flow, method)(user_input)
  File "/config/custom_components/alphaess/config_flow.py", line 69, in async_step_user
    step_id="user", data_schema=STEP_USER_DATA_SCHEMA, errors=errors
UnboundLocalError: local variable 'errors' referenced before assignment
```

CharlesGillanders, 2022-08-31:
> I have just updated via HACS, and I found an error about the configuration flow ("Failed to load configuration flow: undefined") when I try to add the integration. This is the log that comes from [...]

I pushed another update minutes after this one with the fix. My apologies for missing the mistake the first time.

dan-s-github, 2022-09-01:
updated tonight and the anoying log entries are gone. thanks for all the hard work you put into this integration

closed 2022-10-10 by CharlesGillanders