CharlesGillanders/homeassistant-alphaESS #31: add AlphaEss integration to HA is not possible

Z0472, 2022-08-31:
System info:
Home Assistant 2022.8.7
Supervisor 2022.08.5
Operating System 8.5
Frontend 20220802.0 - latest

I follow the instructions exactly, but the AlphaEss option does not appear in the integrations.
I copied all the v0.2.2 files into the custom_components\alphaess folder. I restarted HA. However, it is not possible to add AlphaEss integration. Not in the list. Where can the problem be?
Zdenek

CharlesGillanders, 2022-08-31:
You need to enable "advanced mode" in your user profile in home assistant to be able to add custom integrations.  If you want an easy way to add integrations such as Alpha ESS you might want to check out HACS.

https://hacs.xyz/

Z0472, 2022-08-31:
Thank you very much for quick reply. 
The advanced mode was on. After log out the user an log in again the integration appeard. 
But is there other error : 
This error originated from a custom integration.

Logger: aiohttp.server
Source: custom_components/alphaess/config_flow.py:69
Integration: Alpha ESS (documentation, issues)
First occurred: 7:53:26 PM (2 occurrences)
Last logged: 7:54:20 PM

Error handling request
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/aiohttp/web_protocol.py", line 435, in _handle_request
    resp = await request_handler(request)
  File "/usr/local/lib/python3.10/site-packages/aiohttp/web_app.py", line 504, in _handle
    resp = await handler(request)
  File "/usr/local/lib/python3.10/site-packages/aiohttp/web_middlewares.py", line 117, in impl
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 60, in security_filter_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 100, in forwarded_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 28, in request_context_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 82, in ban_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 236, in auth_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/view.py", line 136, in handle
    result = await result
  File "/usr/src/homeassistant/homeassistant/components/config/config_entries.py", line 145, in post
    return await super().post(request)
  File "/usr/src/homeassistant/homeassistant/components/http/data_validator.py", line 73, in wrapper
    result = await method(view, request, data, *args, **kwargs)
  File "/usr/src/homeassistant/homeassistant/helpers/data_entry_flow.py", line 70, in post
    result = await self._flow_mgr.async_init(
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 222, in async_init
    flow, result = await task
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 249, in _async_init
    result = await self._async_handle_step(flow, flow.init_step, data, init_done)
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 359, in _async_handle_step
    result: FlowResult = await getattr(flow, method)(user_input)
  File "/config/custom_components/alphaess/config_flow.py", line 69, in async_step_user
    step_id="user", data_schema=STEP_USER_DATA_SCHEMA, errors=errors
UnboundLocalError: local variable 'errors' referenced before assignment

CharlesGillanders, 2022-08-31:
Thanks for the error report - I've just pushed a fix to github v0.2.3, please let me know if that fixes the issue.

Z0472, 2022-08-31:
Great! It works. 
Thank you very much, excellent work. 
BR Zdenek

closed 2022-08-31 by CharlesGillanders