CharlesGillanders/homeassistant-alphaESS #32: Nice work, its all going as expected.

Spookster, 2022-08-31:
Nice work, its all going as expected.
Now I just need to convince you to add PPV1 and PPV2 values ;)

Coordinator.py
```
                inverterdata.update({"Instantaneous Load": invertor["powerdata"]["ppv1"] + invertor["powerdata"]["ppv2"] + invertor["powerdata"]["pmeter_dc"] + invertor["powerdata"]["pbat"] + invertor["powerdata"]["pmeter_l1"] + invertor["powerdata"]["pmeter_l2"] + invertor["powerdata"]["pmeter_l3"] })
                inverterdata.update({"Instantaneous PPV1": invertor["powerdata"]["ppv1"]})
                inverterdata.update({"Instantaneous PPV2": invertor["powerdata"]["ppv2"]})                
                self.data.update({invertor["sys_sn"]: inverterdata})
```
                
enums.py
```    GridIOTotal = "Instantaneous Grid I/O Total"
    Load = "Instantaneous Load"
    PPV1 = "Instantaneous PPV1"
    PPV2 = "Instantaneous PPV2"
```
    

 sensor.py
```        AlphaESSSensorDescription(
        key=AlphaESSNames.Load,
        name="Instantaneous Load",
        native_unit_of_measurement=POWER_WATT,
        device_class=SensorDeviceClass.POWER,
        state_class=SensorStateClass.MEASUREMENT,
    ),
        AlphaESSSensorDescription(
        key=AlphaESSNames.PPV1,
        name="Instantaneous PPV1",
        native_unit_of_measurement=POWER_WATT,
        device_class=SensorDeviceClass.POWER,
        state_class=SensorStateClass.MEASUREMENT,
    ),
        AlphaESSSensorDescription(
        key=AlphaESSNames.PPV2,
        name="Instantaneous PPV2",
        native_unit_of_measurement=POWER_WATT,
        device_class=SensorDeviceClass.POWER,
        state_class=SensorStateClass.MEASUREMENT,
    ),
]
```

_Originally posted by @Spookster in https://github.com/CharlesGillanders/homeassistant-alphaESS/issues/7#issuecomment-1233424049_

CharlesGillanders, 2022-08-31:
Yes those should be simple to add.

KortgatGH, 2022-09-13:
Obviously useful for monitoring strings facing different directions, but another possible use could be to detect dirty panels for same-facing strings. I'm looking forward to this! 
Are more of the parameters that are visible on the inverter's display (like voltages and currents) available to be retrieved this way?

Spookster, 2022-09-13:
These are a list of values that DasLetzEinhorn lists as returns from his add on. I would assume that they would be exposed in the same way.
What are the useful ones and why?

    "ppv1": 1151,
    "ppv2": 1302,
    "ppv3": null,
    "preal_l1": 362,
    "preal_l2": 0,
    "preal_l3": 0,
    "pmeter_l1": -247,
    "pmeter_l2": -84,
    "pmeter_l3": 324,
    "pmeter_dc": 926,
    "soc": 86,
    "factory_flag": 0,
    "pbat": -1908.2,
    "sva": 524,
    "varac": 0,
    "vardc": 0,
    "ev1_power": 0,
    "ev1_chgenergy_real": 0,
    "ev1_mode": 0,
    "ev2_power": 0,
    "ev2_chgenergy_real": 0,
    "ev2_mode": 0,
    "ev3_power": 0,
    "ev3_chgenergy_real": 0,
    "ev3_mode": 0,
    "ev4_power": 0,
    "ev4_chgenergy_real": 0,
    "ev4_mode": 0,
    "poc_meter_l1": 0,
    "poc_meter_l2": 0,
    "poc_meter_l3": 0

KortgatGH, 2022-09-13:
I am no expert but this may shed some light:
https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=&ved=2ahUKEwj8_7PlrZL6AhUMJ8AKHdCFASEQFnoECAoQAQ&url=https%3A%2F%2Fcommunity.openenergymonitor.org%2Fuploads%2Fshort-url%2F2qysTQ2mAaEEctdEIHXuaaofYIQ.pdf&usg=AOvVaw3MQPnmteJ_orHO6ncd3C6Z

Not sure why there would be a ppv3 if we have only two MPPT controllers. I've noticed that the two string voltages go active quite some time before the MPPTs start making power at sunrise. Maybe handy for fault finding?

Something that I really miss (and I don't see it in the document linked above) is a simple flag to indicate when the Alpha is in UPS mode (it is indicated on the LCD display on my Smile-5). Neither the mobile apps nor the web app provide this. As most of my loads are on the essential load outputs, I don't notice when the grid fails, and it would be handy if I could get that info on Home Assistant. Is it perhaps captured in the InvWorkMode parameter?

CharlesGillanders, 2022-10-10:
@Spookster I've accepted your pull request and the latest version contains your code to show PPV1 and PPV2

closed 2022-10-10 by CharlesGillanders