CharlesGillanders/homeassistant-alphaESS #35: Alpha issue?

KortgatGH, 2022-10-09:
Just me or are more people getting "unavailable" responses? Looks like it started around 09:00 UTC today.
Their own web interface and app seems to work. Node Red Alpha ESS node also seems not to be working.

KortgatGH, 2022-10-09:
Log indicates "Unexpected error fetching alphaess data: time data '2022/10/9 下午12:15:55' does not match format '%Y/%m/%d %H:%M:%S'"

jt1971, 2022-10-09:
Same for me also.

achie188, 2022-10-09:
Same for me!

KortgatGH, 2022-10-09:
For those that use Node-Red: There was an update for the Alpha ESS node that fixed it. I prefer Charles's integration though.

AeroMicky, 2022-10-09:
Oh no, same here. 

Retrying setup: time data '2022/10/9 下午3:31:53' does not match format '%Y/%m/%d %H:%M:%S'

jetscrubber, 2022-10-09:
Same issue.

petosak, 2022-10-09:
Same issue.

CharlesGillanders, 2022-10-09:
Alpha changing their date reporting format again, it's happened before.  I should be able to fix it quickly tomorrow.

qbushome, 2022-10-09:
Hi Charles, thanks for picking up this so quickly !
I was talking to my battery installer earlier this week and he told me that they have an in-house written software that is able to control the AlphaESS battery. They can command it to start charging from the grid (for peak-shaving purposes). They don't want to make their software public :-(
I assume they have access to the battery via an API (or see below) ?
Do you know of the existence of such an API ? I tried contacting AlphaESS, and they ignore me (both in China and Germany). The representative for Belgium gave me a document with information for reading (not writing) information from the battery 'Alpha Modbus Protocol_V1.23.pdf' (don't know if it's a good idea to publish it here). But in that case you need a Modbus to Ethernet converter to read the date (but you won't have these annoying errors anymore).

AeroMicky, 2022-10-10:
Hi bmannz , where did you make this change, i'd like to try it myself. thanks

> As a quick fix I did the following to get it working but I haven't done a pull request as you may have a better way to fix it. tmpstrdate = TokenCreateTime.replace(u'\u4e0b\u5348',"PM") tmpstrdate = tmpstrdate.replace(u'\u4e0a\u5348',"AM") self.tokencreatetime = datetime.strptime(tmpstrdate,"%Y/%m/%d %p%H:%M:%S")
> 
> Hope that may be of some help.

CharlesGillanders, 2022-10-10:
I have pushed a new version of both the API library and the Home Assistant component which should solve this issue, seeing as the library was always beginning with a new authentication call to the API I have chosen to remove the code that checked for  authentication token expiry - it should avoid future issues every time AlphaESS change the formatting on how they return dates.

jetscrubber, 2022-10-10:
It works!! Many thanks Charles!

CharlesGillanders, 2022-10-10:
@matt-oneill spotted an issue that needs to be fixed.  I'll push another update shortly to remove that bug.

CharlesGillanders, 2022-10-10:
Apologies for the repeated releases today.  v0.2.5 is the latest release, it fixes the bug that @matt-oneill identified, please update to that if you have not already done so.

closed 2022-10-10 by CharlesGillanders