CharlesGillanders/homeassistant-alphaESS #36: Instantaneous generation, load, ... not available

pawlikch, 2022-10-12:
Thank you for the last update. Unfortunately, since then all "instantaneous" entities show only 0. 

Thank you for a short update

CharlesGillanders, 2022-10-12:
That's unexpected - all other measurements are ok? 

If you restart home assistant does it fix the issue?  If so that might suggest it's a problem with the API library re-authenticating to Alpha ESS cloud.

pawlikch, 2022-10-12:
Yes - all other measurements are fine. restart does not fix the issue. I will try to re-authenticate! Thank you!!

pawlikch, 2022-10-14:
Re-authentication has not solved the problem - strange

CharlesGillanders, 2022-10-17:
A new version of the API library was published this morning with some alternative code for handling authentication refreshes, I'd be interested to know if the update fixes your issue.

pawlikch, 2022-10-17:
Installed, solved the other problem but not my issue. I saw that, also on AlphaESS online Dashboard, Real-Time Power Graph is not available - maybe this is the reason
Thank you for your help!!

CharlesGillanders, 2022-10-17:
Aha,

Ok if you are not seeing the real time statistics from the official AlphaESS dashboard then that’s a different issue.  My code basically grabs the data that the official dashboard uses so if there’s nothing there to pull I won’t be able to do anything either.

Charles Gillanders | Personal Account
***@***.******@***.***>
Please consider the environment before printing this e-mail

closed 2022-10-17 by CharlesGillanders