CharlesGillanders/homeassistant-alphaESS #37: new version

petrzz1, 2022-10-16:
AlphaCloud V5.0.0.2 
2022-10-16
Add
 Signature verification and anti-crawl data functionality will be added in all APIs of AlphaCloud.

gryg427, 2022-10-16:
It was probably added maybe 2 hours ago in my case. Access is dismissed by the HACS addon even NodeRed reader :-(. Is there any hope for needed appropriate development?

gryg427, 2022-10-16:
Repeated log by NodeRed: 

16. 10. 2022 13:52:22[node: Alpha ESS monitoring]
msg : string[163]
"There was an error during login operation into Alpha ESS monitoring portal: SyntaxError: Unexpected end of JSON input↵↵We got the following unprocessable body:↵"

matt-oneill, 2022-10-16:
Alpha screwing us over again.
Looks like the request now needs to add the following headers
authtimestamp: {epoch unix timestamp}
authsignature: {}
Don't know what the signature is, but usually something like an api key and secret base64 encoded.
I managed to get a response using fiddler to get values for these headers, but when the timestamp and signature don't match this response is returned from AlphaESS
`{
  "code": 9006,
  "info": "Network exception, please try again later!",
  "exMessageInfo": "If you need to use Api, please contact Alpha Sales to purchase authorization"
}`

jetscrubber, 2022-10-16:
Same issue.....another time! Damn!!

KortgatGH, 2022-10-16:
The Node-Red Alpha ESS node was updated to attempt to fix this. Maybe there is a clue here in their workaround:
https://github.com/dehsgr/node-red-contrib-alphaess

CharlesGillanders, 2022-10-16:
Thanks all - I'll be able to look at this tomorrow.

CharlesGillanders, 2022-10-17:
@matt-oneill proposed changes that fix this for now, thanks Matt.  I've updated the PyPi library and I'll be pushing a version of the homeassistant custom component shortly.  If AlphaESS are determined to prevent 3rd party use of their API this may be a longer term problem.

closed 2022-10-17 by petrzz1

vantech14, 2022-10-17:
> @matt-oneill proposed changes that fix this for now, thanks Matt. I've updated the PyPi library and I'll be pushing a version of the homeassistant custom component shortly. If AlphaESS are determined to prevent 3rd party use of their API this may be a longer term problem.

Is there any way there could be a subscription to access their API? Has anyone reached out the Alpha?

matt-oneill, 2022-10-17:
> Is there any way there could be a subscription to access their API? Has anyone reached out the Alpha?

Yes, here's the response so far
[image]

And here's the api response with an invalid signature
[image]

vantech14, 2022-10-17:
> Yes, here's the response so far
>
> [image]
>
>
>
> And here's the api response with an invalid signature
>
> [image]

Ah Alex is a bundle of joy. I've dealt with him before. I've also reached out. 

Is there perhaps a way to scrape this data from their webpage, rather than using an API that they are obviously keen to block?

matt-oneill, 2022-10-17:
> Ah Alex is a bundle of joy. I've dealt with him before. I've also reached out.
>
> Is there perhaps a way to scrape this data from their webpage, rather than using an API that they are obviously keen [...]

Hopefully Alex can get us an API key!
I don't think its just the api calls they want to block as the notes mention anti-crawl functionality too.

HA scraper is basic and would need an id or css selector to identify the values, and configuration for each. Scraping values from the website doesn't guarantee any more stability than mimicking the api calls as that can change too.
https://www.home-assistant.io/integrations/scrape/