CharlesGillanders/homeassistant-alphaESS #40: Token is Expire error

bmannz, 2022-10-21:
Like the other person that logged it I am getting the following error 1-2 times a day and when it occurs it stops reporting data until I reload the integration.
It logs the below.
Any idea what is causing it?

Logger: alphaess.alphaess
Source: custom_components/alphaess/coordinator.py:30
Integration: Alpha ESS (documentation, issues)
First occurred: 1:46:51 PM (51 occurrences)
Last logged: 2:36:51 PM
200, message='Token is Expire', url=URL('https://cloud.alphaess.com/api/Account/RefreshToken')

Gerrett84, 2022-10-21:
I have the same problem.

homeconnexde, 2022-10-21:
Same error with me.

Tupsi, 2022-10-21:
it has to do with the latest changes on the alphaess website. They made it harder to pull the data.

jasonwins, 2022-10-21:
Same error here using v0.2.6 (latest). The component runs successfully for a while and then starts failing. Reloading the component sorts it out again for a while.

Tupsi, 2022-10-21:
ah my bad. Then I haven't run into the "expire" you described as I just updated and the errors went away. Seems Charles needs to built in some sort of restart/renew into the module each x minutes and then it should work with doing what the module currently only does at the start.

matt-oneill, 2022-10-23:
Sorry probably my bad not handling the refresh tokens properly. Mines just stopped working today, don't know how it didn't break before. I'll have time to look at a fix tomorrow evening but reloading the integration will fix for now.
Tricky one to test properly without waiting 10 hours for the token to expire

CharlesGillanders, 2022-10-24:
Thanks to @matt-oneill again for the quick fix.  An updated release is now published.

Spookster, 2022-10-25:
Logger: homeassistant.setup
Source: setup.py:184
First occurred: 15:58:08 (1 occurrences)
Last logged: 15:58:08

Setup failed for custom integration alphaess: Requirements for alphaess not found: ['alphaess==0.0.13'].

Spookster, 2022-10-25:
Logger: homeassistant.util.package
Source: util/package.py:98
First occurred: 15:57:51 (3 occurrences)
Last logged: 15:58:08

Unable to install package alphaess==0.0.13: ERROR: Could not find a version that satisfies the requirement alphaess==0.0.13 (from versions: 0.0.1, 0.0.2, 0.0.3, 0.0.4, 0.0.5, 0.0.6, 0.0.7, 0.0.8, 0.0.9, 0.0.10, 0.0.11, 0.0.12) ERROR: No matching distribution found for alphaess==0.0.13

CharlesGillanders, 2022-10-25:
Sorry folks, that's my mistake.  I can fix it today.

CharlesGillanders, 2022-10-25:
> integration alphaess: Requirements for alphaess not found: ['alphaess==0.0.13'].

I've published the missing version of the api library, a simple restart of home assistant should fix this problem now.

matt-oneill, 2022-10-26:
Got the error again last night, new pr sent to check for token expire message when refreshing the token

bmannz, 2022-10-26:
Yeah I also received the expiration again today.

bmannz, 2022-10-27:
I see Alpha Cloud has announced a new release on the 29th October.
"Platform security has been improved further, and multiple anti-crawling mechanisms have been added. After this update, please clear the browser cache and log into the system again if there is message that "Network anomaly, please try again later "

I assume this is going to cause havoc again :-(

Spookster, 2022-10-27:
I only got to see the message in chinese, so I assumed the worst

Tupsi, 2022-10-27:
Maybe it is time that we as customers explain to alphaess to stop with this bs OR provide a fully working modprobeTCP implementation so that we do not need this plugin any longer. Prohibiting us to get our own data just feels wrong. I can understand that they do not want us to hammer there web service, but we would not have to if everything would be provided locally as a lot of others in the solar market do. 

For instance I already asked them to feature complete modbusTCP in comparison to what we can get from the website but only got a polite "we will forward it to the Devs" which we all know can mean anything or nothing.

CharlesGillanders, 2022-10-27:
> Maybe it is time that we as customers explain to alphaess to stop with this bs OR provide a fully working modprobeTCP implementation so that we do not need this plugin any longer. Prohibiting us to [...]

It seems very unlikely that a handful of customers scraping their API could actually be causing any website performance issues.  I don't have any insight into why AlphaESS are concerned about 3rd party use of their API or any other mechanism to gather local data but my cynicism is inclined to wonder if there's something else planned.  Maybe there is a plan to monetise the App and Website by adding adverts or something?

Spookster, 2022-10-27:
Its concerning as my whole home automation journey started with and is based around being able to obtain power and battery data. It pretty much renders the main parts of my HA system useless. I hope someone is smart enough or persistent enough to get a more permanent solution :/

bmannz, 2022-10-27:
Good point @CharlesGillanders . There must be some ulterior motive to their actions.

And I agree @Spookster, Everyone wants to optimise their solar consumption and HA is perfect for that but it needs to have the valid data from the solar system.

While I never got it to work for me, with them being such a pain with the API maybe we need to look at using the route of capturing the data being sent from the Inverter instead.
https://github.com/230delphi/hassio-addons/tree/main/alphaess-proxy-addon

dry, 2022-10-27:
I was in direct contact with UK Alpha technical support today and even though I asked, I wasn't able to get a straight answer about Alpha's supported and preferred way for endusers to integrate their systems into home automations systems. On the subject of other potential solutions, I recently found this: https://github.com/dxoverdy/Alpha2MQTT.

Spookster, 2022-10-29:
Looks like the update is happening.... good luck.....

closed 2023-11-14 by CharlesGillanders