CharlesGillanders/homeassistant-alphaESS #43: Update via HACS failed (v0.2.8)

thospe, 2022-10-28:
After updating the alphaess module via HACS I got the following error:

Unable to prepare setup for platform alphaess.sensor: Platform not found (No module named 'custom_components.alphaess.entity').

I restarted twice, updated HASS as well, error still remains. Checking the files via Terminal 
showed the files which are in the repository.

I also realized that I'm not getting valid data since Oct.9th 

Any Idea how to fix it ? Where is my mistake ?

My installation:  
- Home Assistant 2022.10.5
- Supervisor 2022.10.0
- Operating System 9.3 
- Frontend 20221010.0 - latest  
- AlphaESS [v0.2.8]

CharlesGillanders, 2022-10-28:
That's very strange that it cannot find the module entirely.  I'd like to suggest perhaps removing it entirely and then restarting and then re-installing from the beginning via HACS?  But this is a bit of a poke it and see solution as what you are seeing is very unexpected.

thospe, 2022-10-28:
... removing it entirely
I assume all collected data will be deleted, is there an easy way to export the existing collected PV data ?

CharlesGillanders, 2022-10-28:
Home assistant should not delete any historical data - that’s kept entirely separately from the active integration.

Charles Gillanders | Personal Account
***@***.******@***.***>
Please consider the environment before printing this e-mail

thospe, 2022-11-05:
I did not delete the whole app but updated to the newest version at that time and restarted. The this issue was gone and I ended in issue #45. Interesting to read about the hen and egg situation where AlphaEss tries to keep their users out of their portal and data. 

Has anyone tried to get in touch with them about opening and documenting the API officially ?

thospe, 2022-11-05:
After installing 2 further updates (OS and AlphaESS) my Energy Widget is working again. Thanks for continously checking and updating the intrgration.

 Home Assistant 2022.11.1 Supervisor 2022.10.2 Operating System 9.3 Frontend 20221102.1 - latest  + AlphaESS 0.2.10

closed 2022-11-05 by thospe