CharlesGillanders/homeassistant-alphaESS #44: Alphacloud V5003

petrzz1, 2022-10-29:
loging doesn't work

DavidqStokes, 2022-10-29:
My integration is showing:

Retrying setup: 405, message='Method Not Allowed', url=URL('https://cloud.alphaess.com/api/Power/SticsByPeriod')

It's the same, even after a reload

strickersuk, 2022-10-29:
Same error for me.

CharlesGillanders, 2022-10-29:
New breaking changes to the AlphaESS cloud api today.  I've just pushed a new release - if you update in HACS it should be ok.

strickersuk, 2022-10-29:
Thanks again Charles, appreciate the quick fix.

DavidqStokes, 2022-10-29:
Big thanks to Matt and Charles - great work!

homeconnexde, 2022-10-29:
I get: Retrying setup: 401, message='Incorrect User Name or Password!', url=URL('https://cloud.alphaess.com/api/Account/Login')

For Info: Unfortunately there is a space in my user name. Maybe there is a change in submitting a space.

petrzz1, 2022-10-29:
Big thanks you for your work and speed.

closed 2022-10-29 by petrzz1