CharlesGillanders/homeassistant-alphaESS #45: Network Exception

DaveBinM, 2022-10-31:
Around 1200 AEDT, I started seeing this error with my integration. The official app and webpage are still working, so is this perhaps part of their anti-scraping work?

```
Retrying setup: 200, message='Network exception, please try again later!', url=URL('https://cloud.alphaess.com/api/Account/Login')
```

bmannz, 2022-10-31:
I am also getting the same:  
```
  200, message='未知错误', url=URL('https://cloud.alphaess.com/api/Power/SticsByPeriod?beginDay=2022-10-30&endDay=2022-10-30&tDay=2022-10-30&isOEM=0&SN=AL2002221060226&userID=&noLoading=true')
    Cannot connect to host cloud.alphaess.com:443 ssl:default [Try again]
    200, message='Network exception, please try again later!', url=URL('https://cloud.alphaess.com/api/Account/GetCustomMenuESSList')
```

DaveBinM, 2022-10-31:
`未知错误` translates as "unknown error", which isn't super helpful

DaveBinM, 2022-10-31:
...and it's back now. Perhaps a little premature on the bug filing 😅

bmannz, 2022-10-31:
> ...and it's back now. Perhaps a little premature on the bug filing 😅

You're right.  Great that it back working.
Damn Alpha keeping us on the edge due to their antics.

DaveBinM, 2022-10-31:
Okay, I'm seeing this error again, so not sure if this is something worth looking into

bmannz, 2022-10-31:
Same for me.  Has been occurring for the last 1.5hrs.

strickersuk, 2022-10-31:
Same for me:

Retrying setup: 200, message='Network exception, please try again later!', url=URL('https://cloud.alphaess.com/api/Account/Login')

KortgatGH, 2022-10-31:
Same here. Very frustrating. I'm at the point where I have to stop recommending Alpha to people that are interested in home automation.

matt-oneill, 2022-10-31:
here's the full error message. 
``` json
{
    "code": 9006,
    "info": "Network exception, please try again later!",
    "exMessageInfo": "Please use web browser to view the data"
}
```

CharlesGillanders, 2022-10-31:
I'm about to push a new version that will fix this for now.

The Node Red integration https://github.com/dehsgr/node-red-contrib-alphaess has code which automatically extracts the latest AlphaESS signature key from the Alpha javascript files.  That's very clever and I will see if I can implement the same functionality in Python.

CharlesGillanders, 2022-10-31:
New version pushed to HACS.

KortgatGH, 2022-10-31:
Thanks @CharlesGillanders - would have been a lost case without you!

Dejv311, 2022-11-26:
Hi,
we still experienced the "Network Exception" bug even with version 3.1 - in the end we tried reconfigurating the plugin, which resulted in us not being able to login due to "Unknown error" in the configuration. Any help is appreciated.

CharlesGillanders, 2022-11-26:
Are you using the HACS integration or setting it up manually?

Dejv311, 2022-11-26:
I am using HACS. I tried v0.3.0 and v0.3.1 so far. Both of them got "Network exception" and I am not able to "add" integration anymore...

Dejv311, 2022-11-26:
By the look of things, even the "add" integration is failing due to "Network exception"... Here is the log error caused by adding the alpha ess integration:

```
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/aiohttp/web_protocol.py", line 435, in _handle_request
    resp = await request_handler(request)
  File "/usr/local/lib/python3.10/site-packages/aiohttp/web_app.py", line 504, in _handle
    resp = await handler(request)
  File "/usr/local/lib/python3.10/site-packages/aiohttp/web_middlewares.py", line 117, in impl
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 60, in security_filter_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 100, in forwarded_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 28, in request_context_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 82, in ban_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 236, in auth_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/view.py", line 136, in handle
    result = await result
  File "/usr/src/homeassistant/homeassistant/components/config/config_entries.py", line 180, in post
    return await super().post(request, flow_id)
  File "/usr/src/homeassistant/homeassistant/components/http/data_validator.py", line 73, in wrapper
    result = await method(view, request, data, *args, **kwargs)
  File "/usr/src/homeassistant/homeassistant/helpers/data_entry_flow.py", line 110, in post
    result = await self._flow_mgr.async_configure(flow_id, data)
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 280, in async_configure
    result = await self._async_handle_step(
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 367, in _async_handle_step
    result: FlowResult = await getattr(flow, method)(user_input)
  File "/config/custom_components/homeassistant-alphaESS/config_flow.py", line 58, in async_step_user
    await validate_input(self.hass, user_input)
  File "/config/custom_components/homeassistant-alphaESS/config_flow.py", line 35, in validate_input
    raise e
  File "/config/custom_components/homeassistant-alphaESS/config_flow.py", line 29, in validate_input
    await client.authenticate(data[CONF_USERNAME], data[CONF_PASSWORD])
  File "/usr/local/lib/python3.10/site-packages/alphaess/alphaess.py", line 64, in authenticate
    raise aiohttp.ClientResponseError(response.request_info, response.history,
aiohttp.client_exceptions.ClientResponseError: 200, message='Network exception, please try again later!', url=URL('https://cloud.alphaess.com/api/Account/Login')
```

CharlesGillanders, 2022-11-27:
@Dejv311 that's a very unexpected error.  It's an error coming back to you from the AlphaESS server API, I presume that everything is working correctly with your Alpha unit when accessing it from the actual AlphaESS web page?

Dejv311, 2022-11-27:
Yes, everything is working through the browser.

closed 2023-11-14 by CharlesGillanders