CharlesGillanders/homeassistant-alphaESS #46: network exception in  after latest version installed

Z0472, 2022-10-31:
I am not sure if the URL is right. In the browser it returns HTTP 500. 

This error originated from a custom integration.
Logger: alphaess.alphaess
Source: custom_components/alphaess/coordinator.py:30
Integration: Alpha ESS (documentation, issues)
First occurred: 9:06:42 AM (6 occurrences)
Last logged: 9:09:17 AM

200, message='Network exception, please try again later!', url=URL('https://cloud.alphaess.com/api/Account/Login')

DaveBinM, 2022-10-31:
Previously reported in https://github.com/CharlesGillanders/homeassistant-alphaESS/issues/45

Z0472, 2022-10-31:
Previously reported in https://github.com/CharlesGillanders/homeassistant-alphaESS/issues/45

Z0472, 2022-10-31:
#46 deleted, thanks for notice

closed 2022-10-31 by Z0472