CharlesGillanders/homeassistant-alphaESS #51: Alpha Ess error

terosteros, 2022-11-20:
hey all,

When trying to start the AlphaEss integration I get Config flow could not be loaded: {"message":"Invalid handler specified"} anyone else get the same issue?

Dejv311, 2022-11-26:
Hi, we also have a problem with config. We get "Unknown error occurred" when trying to setup the integration. We tried to reset the integration after we could not connect to AlphaESS for over a month.

CharlesGillanders, 2022-11-26:
As an initial exercise I would suggest removing the integration entirely via HACS and then trying a re-install after rebooting between removal and re-install.

Dejv311, 2022-11-26:
Hi, I tried reinstalling it (with reboots) as suggested, but it did not help. I still get "unknown error" when trying to "add" the integration. (Ver 0.3.1)

CharlesGillanders, 2022-11-29:
So I think I'm going to need more lower level information to try to help with this.  The good news is that the library that communicates with Alpha cloud is able to be used on a standalone basis outside of Home Assistant.  If you are at all familiar with postman there is a Postman collection available at https://github.com/CharlesGillanders/alphaess/blob/main/AlphaESS.postman_collection.json I'd be very interested in the results of Login
as well as GetCustomUseESSSetting and GetLastPowerDataBySN assuming that Login works correctly for you.

CharlesGillanders, 2022-12-05:
> Hi,
> I unfortunately do not know Postman. I tried running your Python alphaess library on my PC and I was able to login and retreive data. I also tried to delete, restart and install again the HA integration, but no luck. The login still does not work there. I hoped it might have been caused by some forgotten dependency, but I could not test that.

If you can use the library fine my apologies but I’m now entirely confused.  Are you running HA on a different device to your pc?  Are you able to download and try the library on that device?

Charles Gillanders | Personal Account
***@***.******@***.***>
Please consider the environment before printing this e-mail

tasranson, 2023-10-30:
Did this ever get resolved?  I have been unable to use the Alpha-ESS for some time now.  Just tried a fresh addition via HACS and adding the integration and got the same unknown error.  I can use the Alpha ESS iPhone app and the monitoring web site with the same user name and password.

CharlesGillanders, 2023-10-30:
Without more low level diagnostics this issue cannot be addressed.  The library simply calls the same web APIs as the website so there's something different happening when the connection is attempted from an affected HA instance compared to using it from another PC.

I need the low level information when ran on an affected HA system.

CharlesGillanders, 2023-10-30:
I deleted my previous reply - I didn't get this issue fixed but looking at similar problems with several other integrations it seems that maybe there's some reference issue happening.  

https://community.home-assistant.io/t/error-adding-integration/485237

https://github.com/music-assistant/hass-music-assistant/issues/1091

Can you increase your HA logging and send me any errors that are thrown when you try to add the alphaESS component?

closed 2023-11-14 by CharlesGillanders