CharlesGillanders/homeassistant-alphaESS #52: Determine UPS mode

DaveBinM, 2022-11-29:
This isn't so much of an issue as a question. I was wondering if the API returns whether the battery and inverter are in UPS mode or not? 

I know it says UPS mode on the actual device when it's in that mode, but if it's possible to get that in Home Assistant, it would be quite useful for disabling certain automations, and issuing warnings about the battery and it's SOC.

CharlesGillanders, 2022-11-29:
My unit doesn't support UPS mode so I am unable to check this myself.  But the good news is that the underlying library that communicates with AlphaESS has both a standalone test as well as a Postman collection that can be used to interactively gather direct responses from the Alpha cloud.

If you are at all familiar with postman there is a Postman collection avaible at https://github.com/CharlesGillanders/alphaess/blob/main/AlphaESS.postman_collection.json I'd be very interested in the results of GetCustomUseESSSetting and GetLastPowerDataBySN when running in both UPS mode and not running in UPS mode.

DaveBinM, 2022-11-29:
Yeah, I've used postman before, but not in a while. Unfortunately for me though, my internet connection isn't powered by the UPS, so I can't query the API while in UPS mode. Having had a look at the data provided by @cpmueller, I was hopeful that 
```
        "ups1": "1",
        "switch_on1": 0,
        "switch_off1": 0,
```
would be what we were after, but it unfortunately is the same in both the UPS on and UPS off files provided 😞 There was about three minutes between the two requests, which I hope would be enough time for this to be flagged, but maybe no?

closed 2023-11-14 by CharlesGillanders