CharlesGillanders/homeassistant-alphaESS #53: Multiple inverters

hjpfourie, 2022-12-12:
Hi @CharlesGillanders, would the integration work if I have more than one Alpha Inverter? Do I need to add the integration twice? Or will it pick up that I have a second inverter on my account and create the necessary entities?

CharlesGillanders, 2022-12-12:
In theory it should pick up both but I don't have two so I haven't been able to confirm that it works correctly.  If you do have two and can test that would be amazing.....

hjpfourie, 2022-12-30:
@CharlesGillanders so I get information from multiple inverters workout a problem, but setting charge/discharge settings seems to apply the same settings regardless of the serial I pass through?

hjpfourie, 2023-01-11:
@CharlesGillanders, any idea why the songs seem to apply to all inverters regardless of the Serial passed in the API?

hjpfourie, 2023-01-12:
I've tried putting the serial in quotes like some of the other values, still the same issue. It's as if it ignores which inverter I am trying to set, and sets both. 

I have tried an invalid serial too, and that fails. 

So it definitely needs a valid value, it just doesn't seem to set only that serial's values when it does apply settings.

CharlesGillanders, 2023-01-12:
I don't have multiple inverters so I cannot see what's happening here myself. Is there any way you could use a proxy or a traffic capture tool to see exactly what the official AlphaESS website sends when setting just one inverter? Once I know what the official API call is I should be able to replicate it correctly.

hjpfourie, 2023-01-12:
I can do that. Any specific traffic capture tool you can recommend? Not very familiar with how to do this.

hjpfourie, 2023-01-12:
Ok, so I have used Fiddler and this is what I get as the JSON being sent through to the API:

{
    "code": 200,
    "info": "Success",
    "data": {
        "sys_sn": "AL7011xxxxxxxxx", --- removed my actual serial number
        "ems_version": "V0.10.74",
        "charge_workdays": "0000000",
        "grid_charge": 1,
        "time_chaf1a": "13:00",
        "time_chae1a": "14:00",
        "time_chaf2a": "00:00",
        "time_chae2a": "00:00",
        "bakbox_ver": "V0.00",
        "ctr_dis": 1,
        "time_disf1a": "00:00",
        "time_dise1a": "00:00",
        "time_disf2a": "00:00",
        "time_dise2a": "00:00",
        "bat_high_cap": 100.00,
        "bat_use_cap": 25.00,
        "charge_weekend": "0000000",
        "grid_Charge_we": 0,
        "time_cha_fwe1a": "00:00",
        "time_cha_ewe1a": "00:00",
        "time_cha_fwe2a": "00:00",
        "time_cha_ewe2a": "00:00",
        "bat_highcap_we": 0.00,
        "ctr_dis_we": 0,
        "time_dis_fwe1a": "00:00",
        "time_dis_ewe1a": "00:00",
        "time_dis_fwe2a": "00:00",
        "time_dis_ewe2a": "00:00",
        "bat_usecap_we": 0.00,
        "basic_mode_jp": null,
        "peace_mode_jp": null,
        "vpp_mode_jp": null,
        "channel1": "0",
        "control_mode1": null,
        "start_time1a": null,
        "end_time1a": null,
        "start_time1b": null,
        "end_time1b": null,
        "date1": "0000000",
        "charge_soc1": null,
        "ups1": "0",
        "switch_on1": 0,
        "switch_off1": 0,
        "delay1": "0",
        "duration1": "0",
        "pause1": "0",
        "channel2": "0",
        "control_mode2": null,
        "start_time2a": null,
        "end_time2a": null,
        "start_time2b": null,
        "end_time2b": null,
        "date2": "0000000",
        "charge_soc2": null,
        "ups2": "0",
        "switch_on2": 0,
        "switch_off2": 0,
        "delay2": "0",
        "duration2": "0",
        "pause2": "0",
        "l1_priority": 1,
        "l2_priority": 2,
        "l3_priority": 3,
        "l1_soc_limit": 0.00,
        "l2_soc_limit": 0.00,
        "l3_soc_limit": 0.00,
        "charge_mode2": 1,
        "charge_mode1": 1,
        "backupbox": null,
        "minv": "SMILE5-INV",
        "mbat": "SMILE5-BAT",
        "generator": 0,
        "gc_output_mode": 0,
        "generator_mode": 0,
        "gc_soc_start": 10,
        "gc_soc_end": 10,
        "gc_time_start": 0,
        "gc_time_end": 0,
        "gc_charge_power": 0,
        "gc_rated_power": 0,
        "dg_cap": null,
        "dg_frequency": 50,
        "gc_rate_percent": 80,
        "chargingpile": 0,
        "currentsetting": 32.00,
        "chargingmode": 1,
        "charging_pile_list": [],
        "chargingpile_control_open": 1,
        "peak_fill_en": 0,
        "peakvalue": 20.0,
        "fillvalue": 0.0,
        "delta": 1.0,
        "peak_s1a": "00:00",
        "peak_e1a": "00:00",
        "peak_s2a": "00:00",
        "peak_e2a": "00:00",
        "fill_s1a": "00:00",
        "fill_e1a": "00:00",
        "fill_s2a": "00:00",
        "fill_e2a": "00:00",
        "pm_offset": 0.0,
        "pm_max": 0.0,
        "pm_offset_en": 0,
        "pm_offset_s1a": "00:00",
        "pm_offset_e1a": "00:00",
        "pm_offset_s2a": "00:00",
        "pm_offset_e2a": "00:00",
        "stoinv_type": 1,
        "loadcut_soc": 10,
        "loadtied_soc": 11,
        "ac_tied": 0,
        "soc_50_flag": 0,
        "auto_soccalib_en": 0,
        "three_unbalance_en": 0,
        "enable_current_set": 1,
        "enable_obc_set": 1,
        "upsReserve": 0
    }
}

hjpfourie, 2023-01-12:
Slightly different values on some of the fields for my other inverter:

{
    "code": 200,
    "info": "Success",
    "data": {
        "sys_sn": "AL2002xxxxxxxxx",
        "ems_version": "V1.02.68",
        "charge_workdays": "0000000",
        "grid_charge": 1,
        "time_chaf1a": "13:00",
        "time_chae1a": "14:00",
        "time_chaf2a": "00:00",
        "time_chae2a": "00:00",
        "bakbox_ver": "V0.00",
        "ctr_dis": 1,
        "time_disf1a": "00:00",
        "time_dise1a": "00:00",
        "time_disf2a": "00:00",
        "time_dise2a": "00:00",
        "bat_high_cap": 100.00,
        "bat_use_cap": 25.00,
        "charge_weekend": "0000000",
        "grid_Charge_we": 0,
        "time_cha_fwe1a": "00:00",
        "time_cha_ewe1a": "00:00",
        "time_cha_fwe2a": "00:00",
        "time_cha_ewe2a": "00:00",
        "bat_highcap_we": 0.00,
        "ctr_dis_we": 0,
        "time_dis_fwe1a": "00:00",
        "time_dis_ewe1a": "00:00",
        "time_dis_fwe2a": "00:00",
        "time_dis_ewe2a": "00:00",
        "bat_usecap_we": 0.00,
        "basic_mode_jp": null,
        "peace_mode_jp": null,
        "vpp_mode_jp": null,
        "channel1": "0",
        "control_mode1": "1",
        "start_time1a": "",
        "end_time1a": "",
        "start_time1b": "",
        "end_time1b": "",
        "date1": "0000000",
        "charge_soc1": "",
        "ups1": "1",
        "switch_on1": 0,
        "switch_off1": 0,
        "delay1": "0",
        "duration1": "0",
        "pause1": "0",
        "channel2": "0",
        "control_mode2": "1",
        "start_time2a": "",
        "end_time2a": "",
        "start_time2b": "",
        "end_time2b": "",
        "date2": "1111111",
        "charge_soc2": "",
        "ups2": "1",
        "switch_on2": 0,
        "switch_off2": 0,
        "delay2": "0",
        "duration2": "0",
        "pause2": "0",
        "l1_priority": 1,
        "l2_priority": 1,
        "l3_priority": 1,
        "l1_soc_limit": 10.00,
        "l2_soc_limit": 0.00,
        "l3_soc_limit": 0.00,
        "charge_mode2": 1,
        "charge_mode1": 1,
        "backupbox": 0,
        "minv": "SMILE5-INV",
        "mbat": "SMILE5-BAT",
        "generator": 0,
        "gc_output_mode": 0,
        "generator_mode": 0,
        "gc_soc_start": 10,
        "gc_soc_end": 0,
        "gc_time_start": 0,
        "gc_time_end": 0,
        "gc_charge_power": 0,
        "gc_rated_power": 0,
        "dg_cap": 0,
        "dg_frequency": 50,
        "gc_rate_percent": 80,
        "chargingpile": null,
        "currentsetting": 32.00,
        "chargingmode": 1,
        "charging_pile_list": null,
        "chargingpile_control_open": 1,
        "peak_fill_en": 0,
        "peakvalue": 20.0,
        "fillvalue": 0.0,
        "delta": 1.0,
        "peak_s1a": "00:00",
        "peak_e1a": "00:00",
        "peak_s2a": "00:00",
        "peak_e2a": "00:00",
        "fill_s1a": "00:00",
        "fill_e1a": "00:00",
        "fill_s2a": "00:00",
        "fill_e2a": "00:00",
        "pm_offset": 0.0,
        "pm_max": 0.0,
        "pm_offset_en": 0,
        "pm_offset_s1a": "00:00",
        "pm_offset_e1a": "00:00",
        "pm_offset_s2a": "00:00",
        "pm_offset_e2a": "00:00",
        "stoinv_type": 1,
        "loadcut_soc": 10,
        "loadtied_soc": 11,
        "ac_tied": 0,
        "soc_50_flag": 0,
        "auto_soccalib_en": 0,
        "three_unbalance_en": 0,
        "enable_current_set": null,
        "enable_obc_set": null,
        "upsReserve": 0
    }
}

CharlesGillanders, 2023-01-12:
That's close, you have given me the response back from AlphaESS which is helpful but what I think I really need is the payload in the POST request to Alpha - if you can copy the raw values sent to Alpha in fiddler (minus your actual serial) I think that might be perfect.

hjpfourie, 2023-01-12:
Ok, apologies, I see I sent the response, not the request... here is for my one inverter, calling API at https://cloud.alphaess.com/api/Account/CustomUseESSSetting:. From what I can see, there is a system_id field right at the end which is unique per inverter. Not sure where that comes from, but when you add that in, then it sets the specific inverter... omit it, and it sets both. 
{
    "sys_sn": "al2002xxxxxxxxx",    ----- removed my actual System Serial
    "ems_version": "V1.02.68",
    "charge_workdays": "0000000",
    "grid_charge": 1,
    "time_chaf1a": "13:00",
    "time_chae1a": "14:00",
    "time_chaf2a": "00:00",
    "time_chae2a": "00:00",
    "bakbox_ver": "V0.00",
    "ctr_dis": 1,
    "time_disf1a": "00:00",
    "time_dise1a": "00:00",
    "time_disf2a": "00:00",
    "time_dise2a": "00:00",
    "bat_high_cap": "100",
    "bat_use_cap": "25",
    "charge_weekend": "0000000",
    "grid_Charge_we": 0,
    "time_cha_fwe1a": "00:00",
    "time_cha_ewe1a": "00:00",
    "time_cha_fwe2a": "00:00",
    "time_cha_ewe2a": "00:00",
    "bat_highcap_we": 0,
    "ctr_dis_we": 0,
    "time_dis_fwe1a": "00:00",
    "time_dis_ewe1a": "00:00",
    "time_dis_fwe2a": "00:00",
    "time_dis_ewe2a": "00:00",
    "bat_usecap_we": 0,
    "basic_mode_jp": 0,
    "peace_mode_jp": 0,
    "vpp_mode_jp": 0,
    "channel1": "0",
    "control_mode1": "1",
    "start_time1a": "",
    "end_time1a": "",
    "start_time1b": "",
    "end_time1b": "",
    "date1": "0000000",
    "charge_soc1": "",
    "ups1": "1",
    "switch_on1": 0,
    "switch_off1": 0,
    "delay1": 0,
    "duration1": 0,
    "pause1": 0,
    "channel2": "0",
    "control_mode2": "1",
    "start_time2a": "",
    "end_time2a": "",
    "start_time2b": "",
    "end_time2b": "",
    "date2": "1111111",
    "charge_soc2": "",
    "ups2": "1",
    "switch_on2": 0,
    "switch_off2": 0,
    "delay2": 0,
    "duration2": 0,
    "pause2": 0,
    "l1_priority": 1,
    "l2_priority": 1,
    "l3_priority": 1,
    "l1_soc_limit": "10",
    "l2_soc_limit": 0,
    "l3_soc_limit": 0,
    "charge_mode2": 1,
    "charge_mode1": 1,
    "backupbox": 0,
    "minv": "SMILE5-INV",
    "mbat": "SMILE5-BAT",
    "generator": 0,
    "gc_output_mode": 0,
    "generator_mode": 0,
    "gc_soc_start": 10,
    "gc_soc_end": 0,
    "gc_time_start": "00",
    "gc_time_end": "00",
    "gc_charge_power": 0,
    "gc_rated_power": 0,
    "dg_cap": 0,
    "dg_frequency": 50,
    "gc_rate_percent": 80,
    "chargingpile": 0,
    "currentsetting": "32",
    "chargingmode": 1,
    "charging_pile_list": [],
    "chargingpile_control_open": 1,
    "peak_fill_en": 0,
    "peakvalue": "20",
    "fillvalue": 0,
    "delta": "1",
    "peak_s1a": "00:00",
    "peak_e1a": "00:00",
    "peak_s2a": "00:00",
    "peak_e2a": "00:00",
    "fill_s1a": "00:00",
    "fill_e1a": "00:00",
    "fill_s2a": "00:00",
    "fill_e2a": "00:00",
    "pm_offset": 0,
    "pm_max": 0,
    "pm_offset_en": 0,
    "pm_offset_s1a": "00:00",
    "pm_offset_e1a": "00:00",
    "pm_offset_s2a": "00:00",
    "pm_offset_e2a": "00:00",
    "stoinv_type": 1,
    "loadcut_soc": 10,
    "loadtied_soc": 11,
    "ac_tied": 0,
    "soc_50_flag": 0,
    "auto_soccalib_en": 0,
    "three_unbalance_en": 0,
    "enable_current_set": 1,
    "enable_obc_set": 1,
    "upsReserve": 0,
    "sts_en": 1,
    "auto_startdg_en": 0,
    "max_gridcharge": 10,
    "system_id": "q1FNswGWrNf3QE5ck3",  ----- this seems to be unique and if not specified, sets both inverters
    "languageCode": "en-US"
}

hjpfourie, 2023-01-12:
So just checked now... all I need to do is specify a valid Serial for one of my inverters, the trick then is to specify the system_id for the specific inverter I want to set.

hjpfourie, 2023-01-12:
Looks like this API gives you the system_id based on serial: https://cloud.alphaess.com/api/Account/GetCustomUseESSList

hjpfourie, 2023-01-12:
So I guess the quick fix is allow me to specify a system ID in the service call in HA... better solution would be to look up the system_ID based on Serial provided as part of the service call. Hope this helps @CharlesGillanders.

CharlesGillanders, 2023-01-12:
I think that's enough information to fix this properly - all I need now is time to do so.

hjpfourie, 2023-01-12:
Great, happy to test for you when you have found a gap. Appreciated.

hjpfourie, 2023-01-23:
@CharlesGillanders , just checking up on this... have you managed to find some time yet to have a look? Appreciate your efforts and as mentioned, happy to be a beta tester on this one for you.

hjpfourie, 2023-01-31:
@CharlesGillanders  have you managed to find a gap?

CharlesGillanders, 2023-02-01:
I have pushed a new release today that includes updates to fix this issue.  As before I am unable to test myself but please check and confirm that this is now working and if so I will close the issue.

hjpfourie, 2023-02-01:
Works like a charm. Thank you so much @CharlesGillanders

closed 2023-02-01 by hjpfourie