CharlesGillanders/homeassistant-alphaESS #57: request for Voltage per MPPT

joitde, 2023-01-31:
Hi,

@Spookster already implemented the Power of PPV1 and PPV2. I'd like to see the Voltage as well. 
Would it be possible for someone to implement? Or is this something I could achieve myself with configuration?
Little coding experience here...

Best
Hannes

Spookster, 2023-01-31:
Looking quickly at the docs, the values you are looking for are Upv1 and Upv2
I havent tried to access them, but also using node-red-contrib-alphaess and they arent available from that.

joitde, 2023-01-31:
oh that was a quick reply. 
Sorry I don't know node-red-contrib-alphaess, what does it do?

Spookster, 2023-01-31:
I use that component in Node Red alongside Charless code for other reasons:
[image]

I had a quick look at Upv, tried some code, but it seems its not accessible from the standard json code?
Someone else may be able to confirm

closed 2023-11-14 by CharlesGillanders