CharlesGillanders/homeassistant-alphaESS #58: Error while Inverter is offline

ckarrie, 2023-02-01:
Hi, I get following error after initialization of the integration while my Inverter is offline:

```
Traceback (most recent call last):
  File "/usr/lib/python3.10/site-packages/homeassistant/helpers/update_coordinator.py", line 225, in _async_refresh
    self.data = await self._async_update_data()
  File "/config/custom_components/alphaess/coordinator.py", line 46, in _async_update_data
    inverterdata.update({"Instantaneous Grid I/O L1": invertor["powerdata"]["pmeter_l1"]})
TypeError: 'NoneType' object is not subscriptable
```

One solution could be to make the dict more robuts, i.e.

```invertor.get("powerdata", {}).get("pmeter_l1", None)```

instead of 

```invertor["powerdata"]["pmeter_l1"]```

ckarrie, 2023-02-01:
see https://github.com/CharlesGillanders/homeassistant-alphaESS/pull/59 for a fix proposal

CharlesGillanders, 2023-02-01:
I have merged your pull request and pushed a new release that includes your fix.  Thank you!

closed 2023-02-01 by CharlesGillanders