CharlesGillanders/homeassistant-alphaESS #61: Warning in log after updating to HA 2023.2

dan-s-github, 2023-02-03:
After updating Home Assistant to 2023.2 I noticed following warning in the logs

```Detected integration that called async_setup_platforms instead of awaiting async_forward_entry_setups; this will fail in version 2023.3. Please report issue to the custom integration author for alphaess using this method at custom_components/alphaess/__init__.py, line 54: hass.config_entries.async_setup_platforms(entry, PLATFORMS)```

[image]

CharlesGillanders, 2023-02-03:
Thank you for spotting this.

I'm looking at the documentation and I see the new requirements are covered here https://developers.home-assistant.io/docs/config_entries_index#for-platforms

I will need to make adjustments to this integration to comply with the updated standard.

CharlesGillanders, 2023-02-09:
Fixed in release https://github.com/CharlesGillanders/homeassistant-alphaESS/releases/tag/v0.3.3

closed 2023-02-09 by CharlesGillanders

dan-s-github, 2023-02-25:
Sorry for late reply, only had remote access to system and didn't want to do updates when not at home.
I can confirm that the above warning has disappeared.
Thanks for making this integration and keeping it up to date