CharlesGillanders/homeassistant-alphaESS #62: No data fron https://cloud.alphaess.com/

ckarrie, 2023-02-03:
[image]

I get no statistics, only realtime.

I know this is probably not an integration error, but can somebody confirm this or is it just me?

CharlesGillanders, 2023-02-03:
I'm certainly seeing my own statistics correctly.  The image you attached was from the AlphaCloud service.  Is that showing statistics for today or not?  If AlphaCloud doesn't have statistics then this integration will not either - I can only pull what is available on Alpha.

ckarrie, 2023-02-03:
Yes, the screenshot is from today.
Okay thank you, then maybe the time settings of my Alpha are incorrect. 

Yesterday statistics ends at 22:30.

[image]

ckarrie, 2023-02-03:
Yeah, its definately a time offset. Marked as fixed

closed 2023-02-03 by ckarrie