CharlesGillanders/homeassistant-alphaESS #64: Cant choose "Instantaneous" Entity in HA Energy Dashboard

SiL-X, 2023-02-21:
Hello,

for the correct Energy-distribution it needs the Entity
"Instantaneous Grid I/O Total"
so it can show the correct energy flow.

But it is not possible to choose it in the configuration of the Energy Dashboard.

CharlesGillanders, 2023-02-22:
I am using three different sensor readings

Grid To Load
Grid To Battery
Solar To Grid

[image]

The Instantaneous sensors do not show a cumulative figure over time only the specific instant reading when the sensor is collected - I don't think that would work correctly for the energy dashboard as you would miss all of the data between sensor readings?

SiL-X, 2023-02-22:
Hello,

thanks for your fast answer!

Yes, you´re right.
Sorry, I am new to HA. I thought I need to use the instantaneous values for the build-in HA Energy-Dashboard.

My intention was to see the instantaneous values on my "Home-Dashboard".
For this I now added a custom power-flow-card via HACS and used the instantaneous values. Everything works fine now. :)

Maybe your posted screenshot in this topic is something for the manual? That you added 2 sensors for the Grid consumption is also a good advice.

closed 2023-02-22 by SiL-X