CharlesGillanders/homeassistant-alphaESS #66: request for instantanious EV load

FrankC-BEL, 2023-03-03:
It would be great to also have an instant EV load value.

CharlesGillanders, 2023-03-03:
Could you please take a look at this closed issue - I think it should be what you are looking for.

https://github.com/CharlesGillanders/homeassistant-alphaESS/issues/12

FrankC-BEL, 2023-03-03:
Hi,

Sorry to be so ignorant, but yes it seems like what I want to know, but I don’t know how to get it……

Met vriendelijke groeten/ Best Regards/ Viele Grüße,

Frank Caljé
European Business Development Manager – BeNeLux & DACH

M +32 470 80 50 80<tel:+32470805080>

#12<https://eur02.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2FCharlesGillanders%2Fhomeassistant-alphaESS%2Fissues%2F12&data=05%7C01%7Cfcalje%40nt-ware.com%7C8ac725e403374c300f5d08db1c02b484%7Cb487e69e0cf84caaad76e4eb4c043905%7C0%7C0%7C638134570162112007%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=agpqbVpe2PHVEnHcdWXgWnTR8wkRPeKzxxLXK5lPKZA%3D&reserved=0>

closed 2023-11-14 by CharlesGillanders