CharlesGillanders/homeassistant-alphaESS #67: New API & Integration

matt-oneill, 2023-03-05:
@CharlesGillanders have you given any thought to upgrading the library and the HA integration with using the new Alpha public API?
Given that its now possible to register systems that aren't yours with a developer key, I think the HA Integration should take the serial number as a parameter, with the possibility of adding the integration multiple times - rather than how it will behave now retrieving all systems for the account.
i.e. AppId, AppSecret, SysSN for each integration setup.

CharlesGillanders, 2023-03-06:
I haven’t (yet) been able to get the new API working on my own system.  I'm running into roadblocks figuring out my check code.  I can't work on getting the library updated to use the proper public API until that's resolved.

matt-oneill, 2023-03-06:
Here's my postman collection which is working.
Just need to add your AppId and AppSecret to the variables
[AlphaESS Open API.postman_collection.json.zip](https://github.com/CharlesGillanders/homeassistant-alphaESS/files/10901085/AlphaESS.Open.API.postman_collection.json.zip)

I was getting a 6007 error last week and had to generate a new AppSecret for it to work again.

CharlesGillanders, 2023-03-09:
I have just managed to get my own system registered with the developer portal so I can start looking at this now.  Thanks for the Postman collection - that's incredibly helpful!

closed 2023-11-14 by CharlesGillanders