CharlesGillanders/homeassistant-alphaESS #69: Cannot find it on HACS

BigDazza77, 2023-04-01:
Just got my Alpha ESS Battery and Solar system installed and was keen to add it into Home Assistant.  I cannot find it on in HACS.  Has it been removed?

Spookster, 2023-04-01:
How have you tried?
HACS > Integrations > Hamburger Menu > Custom Repositories add https://github.com/CharlesGillanders/homeassistant-alphaESS with category set to integration.

BigDazza77, 2023-04-02:
Perfect.  Thanks.

I wasn't adding it as a custom just searching in HACS.

Thanks for your help and all your good work.

AlbaT6, 2023-04-04:
> Perfect. Thanks. I wasn't adding it as a custom just searching in HACS. Thanks for your help and all your good work.
> […](#)

I just discovered I was doing the same!

closed 2023-11-14 by CharlesGillanders