CharlesGillanders/homeassistant-alphaESS #70: Service to change EV Charger Mode

ckarrie, 2023-04-18:
Hi, I would like to have a new service to change/set one of the four EV Charger Mode (SMILE-EVCT11):

- ECO: Slow charging (German: Langsamladung)
- ECO: Care charging (German: Schonladung)
- ECO: Fast charging (German: Schnellladung)
- Performance Mode

Needed parameters (I guess):
- Serial Number of Wallbox
- Serial Number of AlphaESS (unsure)
- Mode

see https://www.manualslib.de/manual/800017/Alpha-Ess-Smile-Evct11.html?page=11 for a german manual.

roggepaul, 2023-04-30:
Hi @ckarrie. It seems like you found a solution to change charging modes via HTTP request but I can not figure it out how to do it by looking at the java code. Do you have a simple example POST-request including the URL and the json body to update the charging mode? You would help me a lot.

ckarrie, 2023-05-01:
Hi @roggepaul , I havn't tested it out yet.
I'm not really familiar with Java, but if you u can read one code, you can read all codes :-)

Shouldn't be much magic, posting to cloud-settings URL with parameter `chargingmode` (integer) and with token as auth. Maybe someone else could figure it out, before I do.

vdwals, 2023-05-03:
Hi all,

I can provide a more detailed JSON in the evening, but so far: The payload to set the charging mode is the same you receive when requesting the system settings inclusive the wallbox settings. Like Charge- and Discharge time. 
You just change the charging mode received and sent the adjusted JSON back.
@ckarrie has already listed the integer explanation of the different modes you need to set at the 'chargingmode' property.

roggepaul, 2023-05-04:
Hi all,

Thank you for your responses. I tried out your suggestions and used a postman collection (https://github.com/CharlesGillanders/alphaess/blob/main/AlphaESS.postman_collection.json) I found online. 
I used the endpoint https://cloud.alphaess.com/api/Account/CustomUseESSSetting?system_id=xxx to receiver settings and pasted the whole data response into an post request to https://cloud.alphaess.com/api/Account/CustomUseESSSetting. I was able to successfully update the chargingpile_control_open to 0 or 1. I got a HTTP 200 and the "info":"Success" in the response body. The change was applied and visible in the app after a app refresh. But when I try to update the chargingmode value to 1,2,3 or 4 or any value inside the charging_pile_list array I still get a HTTP 200 and the "info":"Success" in the response body but the update doesnt get applied and is not visible in the app. Do you have any suggestions? Has anyone successfully updated the chargingmode via an api?

roggepaul, 2023-05-04:
@vdwals thank you for your response. I get the same structure like you when performing a GET on https://cloud.alphaess.com/api/Account/CustomUseESSSetting?system_id=xxx. I know that the chargingmode is not in the charging_pile_list array and I never had it there. I just tried to change different settings in the array. When I edit the chargingmode and POST it I still get a success message but no update inside the app. The same applies to changes in the charging_pile_list array (not regarding the charging_pile_list array). To my mind it should work and changing the chargingmode and values inside the charging_pile_list array returns a HTTP 200 but the update is not shown in the APP. Can you try it out and tell me if the values get updated and applied by Alphaess and is displayed in the app?

the chargingpile_control_open is the value for "Kontrollrecht offen" that is also changeable in the app inside the "EV-Charger Settings" menu.

Tupsi, 2023-05-23:
Did anyone here get that working? I was wondering if it is possible to get a button in HA for start/stopping charging. From what I figured out in that java code it should be possible, have not found it implemented for HA yet though. From what I can see on his bitbucket @vdwals seems to be using it with HA, but I have not figured out yet, how to use that. Getting the container running is one thing, but how would I use it from within HA? So far I only pulled data from the MQTT broker, never sent anything back.

vdwals, 2023-05-29:
Hi all,

sorry for the late reply. There was a bug in my eclipse and I was unable to debug it to give you the precise messages. So the request is a post request to https://cloud.alphaess.com/api/Account/CustomUseESSSetting
and the body is
```
{
  "ac_tied": 0,
  "auto_soccalib_en": 0,
  "auto_startdg_en": 0,
  "backupbox": 0,
  "bakbox_ver": "V0.00",
  "basic_mode_jp": 0,
  "bat_high_cap": "100.0",
  "bat_highcap_we": 0,
  "bat_use_cap": "4.0",
  "bat_usecap_we": 0,
  "channel1": "0",
  "channel2": "0",
  "charge_mode1": 1,
  "charge_mode2": 1,
  "charge_soc1": null,
  "charge_soc2": "20",
  "charge_weekend": null,
  "charge_workdays": null,
  "charging_pile_list": [
    {
      "chargingpile_hard_ver": null,
      "chargingpile_id": "EV1",
      "chargingpile_phase": 3,
      "chargingpile_sn": "<replaced>",
      "chargingpile_soft_ver": null,
      "chargingpile_startpower": null,
      "chargingpile_switch": 1,
      "chargingpilename": null,
      "chargingpiletype": null,
      "max_current": 16.0,
      "priority": 1,
      "time_charge_1": 0,
      "time_charge_2": 0,
      "time_charge_e1": "15:00",
      "time_charge_e2": "16:00",
      "time_charge_s1": "13:00",
      "time_charge_s2": "07:00"
    }
  ],
  "chargingmode": 2,   <--- Normal
  "chargingpile": 1,
  "chargingpile_id": "EV1",
  "chargingpile_sn": "<replaced>",
  "chargingpile_switch": 1,
  "control_mode1": null,
  "control_mode2": "2",
  "ctr_dis": 0,
  "ctr_dis_we": 0,
  "currentsetting": "32",
  "date1": "0000000",
  "date2": "0000000",
  "delay1": 0,
  "delay2": 0,
  "delta": 1,
  "dg_cap": 0,
  "dg_frequency": 50,
  "duration1": 0,
  "duration2": 0,
  "ems_version": "V0.10.62T",
  "end_time1a": null,
  "end_time1b": null,
  "end_time2a": "00:00",
  "end_time2b": "00:00",
  "fill_e1a": "00:00",
  "fill_e2a": "00:00",
  "fill_s1a": "00:00",
  "fill_s2a": "00:00",
  "fillvalue": 0,
  "gc_charge_power": 0,
  "gc_output_mode": 0,
  "gc_rate_percent": 80,
  "gc_rated_power": 0,
  "gc_soc_end": 10,
  "gc_soc_start": 10,
  "gc_time_end": "0",
  "gc_time_start": "0",
  "generator": 0,
  "generator_mode": 0,
  "grid_charge": 0,
  "grid_Charge_we": 0,
  "l1_priority": 1,
  "l1_soc_limit": "0.00",
  "l2_priority": 2,
  "l2_soc_limit": "0.00",
  "l3_priority": 3,
  "l3_soc_limit": "0.00",
  "languageCode": "de-DE",
  "loadcut_soc": 10,
  "loadtied_soc": 11,
  "max_gridcharge": 10,
  "mbat": "SMILE5-BAT",
  "minv": "SMILE5-INV",
  "pause1": 0,
  "pause2": 0,
  "peace_mode_jp": 0,
  "peak_e1a": "00:00",
  "peak_e2a": "00:00",
  "peak_fill_en": 0,
  "peak_s1a": "00:00",
  "peak_s2a": "00:00",
  "peakvalue": 20,
  "pm_max": 0,
  "pm_offset": 0,
  "pm_offset_e1a": "00:00",
  "pm_offset_e2a": "00:00",
  "pm_offset_en": 0,
  "pm_offset_s1a": "00:00",
  "pm_offset_s2a": "00:00",
  "priority": 1,
  "soc_50_flag": 0,
  "start_time1a": null,
  "start_time1b": null,
  "start_time2a": "00:00",
  "start_time2b": "00:00",
  "stoinv_type": 1,
  "sts_en": 1,
  "switch_off1": 0,
  "switch_off2": 0,
  "switch_on1": 0,
  "switch_on2": 0,
  "sys_sn": "<replaced>",
  "system_id": "<replaced>",
  "time_cha_ewe1a": "00:00",
  "time_cha_ewe2a": "00:00",
  "time_cha_fwe1a": "00:00",
  "time_cha_fwe2a": "00:00",
  "time_chae1a": "00:00",
  "time_chae2a": "00:00",
  "time_chaf1a": "00:00",
  "time_chaf2a": "00:00",
  "time_charge_1": 0,
  "time_charge_2": 0,
  "time_charge_e1": "15:00",
  "time_charge_e2": "16:00",
  "time_charge_s1": "13:00",
  "time_charge_s2": "07:00",
  "time_dis_ewe1a": "00:00",
  "time_dis_ewe2a": "00:00",
  "time_dis_fwe1a": "00:00",
  "time_dis_fwe2a": "00:00",
  "time_dise1a": "00:00",
  "time_dise2a": "00:00",
  "time_disf1a": "00:00",
  "time_disf2a": "00:00",
  "ups1": "0",
  "ups2": "0",
  "upsReserve": 0,
  "vpp_mode_jp": 0
}
```

Which is exactly the response body from GET https://cloud.alphaess.com/api/Account/GetCustomUseESSSetting?system_id=<replaced>, but with adjusted:
* chargingpile_id
* chargingpile_sn
* time_charge_1
* time_charge_2
* time_charge_s1
* time_charge_e1
* time_charge_s2
* time_charge_e2
* chargingpile_switch
* priority
* system_id
* chargingmode

So it is like lifting the affected charger from the charger list into the main structure.
I'm sorry for the previous confusion, I did hide the Response -> Request transformation well in a few steps by transforming one java DTO into another DTO which looked like using the same body.
So this body is from my working debug session.
Let me know if I can help you further.
If you like to have a look, start at https://bitbucket.org/vdwals/alphaesscloud2mqtt/src/master/src/main/java/de/vdw/io/alpha2mqtt/services/alpha/ChargingService.java line 253. This triggers receiving the system settings, creating a setting dto to adjust charging mode and does the final request to the service as well as validates the result.

roggepaul, 2023-05-29:
thanks @vdwals, I am now able to successfully update the chargingmode with your suggested json structure by copying some elements in the root of the json object. Is there any way to start/stop the EV-charging via an api like I do it in the app in the "EV-charger settings" tab (where a car and a wallbox is shown an there are two buttons at the bottom for starting and stopping the charging)?

ckarrie, 2023-08-24:
I'm just to dump to make a successful GET/POST request.
What I've done:
- got system_id from Browser DevTool Network Tab
- and the Bearer Token

Python Code:
```python3

import requests

url = "https://cloud.alphaess.com/api/Account/GetCustomUseESSSetting?system_id=z1aSXXXXXXXXXXX"
token = "eyJhbXXXXX.XXXXXXXXXXXXX...XXX"

class BearerAuth(requests.auth.AuthBase):
    def __init__(self, token):
        self.token = token
    def __call__(self, r):
        r.headers["authorization"] = "Bearer " + self.token
        return r

resp = requests.get(url, auth=BearerAuth(token))
```

Get only a empty response with code 500.

vdwals, 2023-08-25:
> thanks @vdwals, I am now able to successfully update the chargingmode with your suggested json structure by copying some elements in the root of the json object. Is there any way to start/stop the [...]

Hi, yes, Start and Stop are simple calls to the uris:
* https://cloud.alphaess.com/api/ESS/StartCharging
* https://cloud.alphaess.com/api/ESS/StopCharging

with the body:
```
{
    "sys_sn": "AL****",
    "chargingpile_sn": "ALP****"
}
```

ckarrie, 2023-10-07:
Finally I got it to work.

Here is my python3 script:
Usage (python3-requests needed):

      ## set max_current ("Ladestrom A/Phase"):
      python3 alpha.py -u <username> -p <password> -m 10

      ## stop charging:
      python3 alpha.py -u <username> -p <password> --stop 1

      ## start charging
      python3 alpha.py -u <username> -p <password> --start 1

```python3
import requests
from datetime import datetime
import hashlib
import argparse

parser = argparse.ArgumentParser(prog='alpha.py')
parser.add_argument('-u', '--username')
parser.add_argument('-p', '--password')
parser.add_argument('-m', '--max_current', type=int, help="set max current of charging pile (6-16)")
parser.add_argument('--start', help="Start charging")
parser.add_argument('--stop', help="Stop charging")
args = parser.parse_args()

# thanks to https://bitbucket.org/vdwals/alphaesscloud2mqtt/src/master/src/main/java/de/vdw/io/alpha2mqtt/config/Constants.java
AUTH_SIGNATURE_START = "al8e4s"
AUTH_SIGNATURE_HASH = "LS885ZYDA95JVFQKUIUUUV7PQNODZRDZIS4ERREDS0EED8BCWSS"
AUTH_SIGNATURE_END = "ui893ed"

MAX_CURRENT_RANGE = range(6, 16)

def get_auth_headers():
    now = datetime.now()
    auth_ts = str(int(datetime.timestamp(now)))    
    
    hash_str = AUTH_SIGNATURE_HASH + auth_ts
    auth_sig = AUTH_SIGNATURE_START + hashlib.sha512(hash_str.encode()).hexdigest() + AUTH_SIGNATURE_END
    return {
        "Authsignature": auth_sig,
        "Authtimestamp": auth_ts,
    }

def login(username, password):
    url = "https://cloud.alphaess.com/api/Account/Login"
    req = requests.post(url, json={"username": username, "password": password}, headers=get_auth_headers())
    data = req.json().get('data')
    return {
        'AccessToken': data['AccessToken'],
        'ExpiresIn': data['ExpiresIn'],
        'RefreshTokenKey': data['RefreshTokenKey'],                
    }
    

def get_settings(login_data, system_id=None):
    url = "https://cloud.alphaess.com/api/Account/GetCustomUseESSSetting"
    headers = get_auth_headers()
    headers["Authorization"] = "Bearer " + login_data["AccessToken"]
    #print(headers)
    params = {}
    if system_id:
        params = {
            'system_id': system_id
        }
    req = requests.get(url, params=params, headers=headers)
    #print(req)
    return req.json()
    
def post_settings(login_data, settings):
    url = "https://cloud.alphaess.com/api/Account/CustomUseESSSetting"
    headers = get_auth_headers()
    headers["Authorization"] = "Bearer " + login_data["AccessToken"]
    req = requests.post(url, json=settings, headers=headers)
    return req.json()
    

def get_system_list(login_data):
    url = "https://cloud.alphaess.com/api/Account/GetCustomUseESSList"
    headers = get_auth_headers()
    headers["Authorization"] = "Bearer " + login_data["AccessToken"]
    req = requests.get(url, headers=headers)
    return req.json()
    
def get_cp_status(login_data, sys_sn, chargingpile_id="EV1"):
    url = "https://cloud.alphaess.com/api/ESS/GetChargPileStatusByPileSn"
    headers = get_auth_headers()
    headers["Authorization"] = "Bearer " + login_data["AccessToken"]
    req = requests.post(url, json={'sys_sn': sys_sn, 'chargingpile_id': chargingpile_id}, headers=headers)
    j = req.json()
    pile_status_code = int(j['data'])
    
    if pile_status_code == 3:
        return "Lädt"
    if pile_status_code == 6:
        return "Ladevorgang beendet"
    return "Unbekannter Code %s, bitte auf der Webseite nachschauen" % pile_status_code
    
    
def start_charging(login_data, sys_sn, chargingpile_sn):
    url = "https://cloud.alphaess.com/api/ESS/StartCharging"
    headers = get_auth_headers()
    headers["Authorization"] = "Bearer " + login_data["AccessToken"]
    resp = requests.post(url, json={'sys_sn': sys_sn, 'chargingpile_sn': chargingpile_sn}, headers=headers)
    return resp.json()['code'] == 200
    
def stop_charging(login_data, sys_sn, chargingpile_sn):
    url = "https://cloud.alphaess.com/api/ESS/StopCharging"
    headers = get_auth_headers()
    headers["Authorization"] = "Bearer " + login_data["AccessToken"]
    resp = requests.post(url, json={'sys_sn': sys_sn, 'chargingpile_sn': chargingpile_sn}, headers=headers)
    return resp.json()['code'] == 200

login_data = login(username=args.username, password=args.password)
#settings_json = get_settings(login_data=login_data)
system_list_json = get_system_list(login_data=login_data)
first_system_id = system_list_json['data'][0]['system_id']
settings_json = get_settings(login_data=login_data, system_id=first_system_id)

cp_data = settings_json['data']['charging_pile_list'][0]
# Status Wallbox
status_text = get_cp_status(login_data, sys_sn=settings_json['data']['sys_sn'], chargingpile_id=cp_data['chargingpile_id'])
print("Status Wallbox: " + status_text)

# Lademodus, convert to int
max_current = int(cp_data['max_current'])
print("Aktueller Ladestrom %sA" % str(max_current))

if args.max_current:
    if int(args.max_current) in MAX_CURRENT_RANGE:
        max_current = int(args.max_current)
    else:
        print("Ladestrom ausserhalb 6-16A Bereich: %s" % str(args.max_current))
    print("Neuer Ladestrom %sA" % str(max_current))
    cp_data['max_current'] = str(max_current)
    cp_data['system_id'] = first_system_id

    settings_json['data']['charging_pile_list'][0] = cp_data
    settings_json['data'].update(cp_data)

    settings_data_json = settings_json['data']

    #import pprint
    #pprint.pprint(settings_json)

    post_settings(login_data=login_data, settings=settings_data_json)
    
if args.start:
    successful = start_charging(login_data, sys_sn=settings_json['data']['sys_sn'], chargingpile_sn=cp_data['chargingpile_sn'])
    if successful:
        print("Charging starting")
    
if args.stop:
    successful = stop_charging(login_data, sys_sn=settings_json['data']['sys_sn'], chargingpile_sn=cp_data['chargingpile_sn'])  
    if successful:
        print("Charging stopping")    
```

closed 2023-11-14 by CharlesGillanders