CharlesGillanders/homeassistant-alphaESS #72: Use of TCP Modbus?

sebastianmachon, 2023-05-06:
Hi,
Have you tried to use the integrated Modbus TCP?
The used API needs a working Alpha API and internet connectivity.

Modbus TCP works local.

dry, 2023-05-06:
See https://github.com/dxoverdy/Alpha2MQTT

Spookster, 2023-05-26:
Works well :)

Spookster, 2023-05-29:
One idea that I picked up from Daniels work was the use of ApexCharts to display data. It can be applied to Charles work as well to duplicate Alphas charts. I wish Id seen it sooner :)

[image]

aw1604, 2023-07-25:
Hello, I think what sebastian meant was a direct tcp modbus approach since for example the Smile-T10 now supports native modbus/tcp over the network. You can directly pull all data using the HA "modbus tcp" integration (and set too). 
Is there a way to re-write the AlphaESS integration to use HA internal modbus sensor data instead of relying on external mqtt and serial-to-network controllers (as they shouldn't be needed anymore) ?
I am not a coder myself, but willing to sponsor development..
Best Regards, Andreas

closed 2023-11-14 by CharlesGillanders