CharlesGillanders/homeassistant-alphaESS #73: Incorrect Power Usage Readings

DaveBinM, 2023-05-30:
I'm not entirely sure if this is an AlphaESS issue or a HA issue, but I couldn't see anything obvious in their issues related to this.

Sometimes I see my power usage is duplicated at some stage of the day, sometimes at the end of the day, and other times in the middle of the day, where it adds together all the energy usage from the day so far, and puts it as a single hour as shown below. Happy to file with HA if this is a HA issue, but wanted to check here first, in case other AlphaESS users have seen this

[image]

[image]

zaphod0, 2023-06-02:
Hi, I am having the same issue and came across your post. Upon looking into it further, I can see that the solar production entity in HA gets reset to 0 and immediately back to its original value within the same hour where HA Energy shows the spike in production / usage. 

Not sure what might be causing this or of it is an issue with the Integration or HA.  

[image]
[image]

beastien, 2023-06-21:
I didn't see this post, but I believe it's the same problem.
https://github.com/CharlesGillanders/homeassistant-alphaESS/issues/74

closed 2023-11-14 by CharlesGillanders