CharlesGillanders/homeassistant-alphaESS #74: Sensor intermittent zero anomalies

beastien, 2023-06-20:
I'm using the following sensors for custom templates and input into the energy dashboard.

sensor.serial_solar_to_grid
sensor.serial_grid_to_load
sensor.serial_grid_to_battery
sensor.serial_solar_production

The timing doesn't look consistent but every few days to a week there is an intermittent drop to zero on these sensors.
Due to them being state_class total_increasing this causes the sum to add the entire days reading to the total messing up the graphs etc.
Sometimes I can see in the error log that HA was unable to connect to AlphaEss briefly and that coincides with these readings but not every time.

Is it possible to ignore these drops or have the sensor become unavailable or unknown?

[image]

DavidqStokes, 2023-06-21:
Although not as frequent as it was a few months ago, I also see a similar 'drop to zero' in my AlphaESS home battery state of charge (which doesn't impact on the energy dashboard, but which can mess up my templated calculations on whether the battery needs to charge overnight). This is only ever very short lived, and does appear to be because of loss of connectivity to the AlphaESS API. For battery SOC it would be better to retain the last value during loss of communication.

kinektek, 2023-06-21:
Similar for me - have been ignoring it quite happily (solution working well for me, thank you!) but in case more datapoints are helpful in any way, here's a shot of my graph:

[image]

CharlesGillanders, 2023-11-29:
closing as duplicate of #85

Poshy163, 2024-09-02:
This is a dupe of this: https://github.com/CharlesGillanders/homeassistant-alphaESS/issues/127