CharlesGillanders/homeassistant-alphaESS #75: Sensors not zeroing at midnight (zeroing in the wrong time)

JoeArny, 2023-09-04:
Hi, I'm having a weird issue when using this integration in a docker and I could not get to the root cause yet.
Sensors like the "Solar Production" should zero itself every day at mid-night, but it's zeroing near 8am (imgage below).
[image]

I'm using this same integration in a hassio installation (Raspberry Pi) and it all works perfectly fine, with sensors zeroing at midnight.
I checked the time location, but both are set to London. Does anyone have an idea on what I should look for now?
Thank you very much in advance.

PS: The Energy tab goes crazy because of this behaviour, that's why I'm looking to fix this.

SiL-X, 2023-09-25:
Hello!

I am also having this bug. An example here:
[image]

Zeroing not on midnight causes the Energy Board to add it to the same day.

In the picture I added the Energy Board for the 24.09.2023. This is a great example. PV Production goes crazy...

I have no idea how to fix this... :/

SiL-X, 2023-09-25:
I have an idea:

The sensor is set to
total_increasing

At total_increasing: A decreasing value is interpreted as the start of a new meter cycle or the replacement of the meter.

If I see it correctly the Alpha API resets automatically at midnight?

Shouldnt set the sensor to
total
fix this problem?

CharlesGillanders, 2023-09-25:
Using total in the sensor ends up just providing an always increasing value which never resets. That breaks all of the longer term calculations.

I'm not sure this is necessarily an alphaESS issue - it doesn't happen in my system and others don't have the same problem.  

I suspect there's a timezone misalignment somewhere but I'm unable to repeat the problem so it's not easy to figure out.

If someone can figure out where the timezone is getting confused that would be helpful.

SiL-X, 2023-09-25:
Its not always on the same time.
These are my times where it sets to 0:
17.09 - 12:42:05 
18.09 - 13:27:32
19.09 - 20:04:38
20.09 - no error
21.09 - 14:59:56
22.09 - no error
23.09 - no error
24.09 - 22:09:06
25.09 - 10:29:25

HA timezone: GMT +1:00 Berlin
AlphaESS Timezone +1:00 Berlin
Server Timezone +1:00 Berlin

I dont see a pattern...

Long time I also had 0 issues. I have no idea when it came up the first time.
I restartet everything yesterday. Server, Router, HA, everything. Did not help. :/

CharlesGillanders, 2023-09-25:
Yeh that’s far more random than I thought.  Almost certainly not a timezone issue then.

When it goes to zero it looks like the next reading is back right where it should be?  That's not what I'd necessarily expect from a total increasing sensor.  When those actually reset to zero it takes time to ramp back up (as it should do).

I'm wondering if this is an indicator of broken error handling somewhere.  At the times this goes to zero are there any possibly related AlphaESS messages in the home assistant log?

SiL-X, 2023-09-25:
Yes, the next reading is back right where it should be. And Home Assistant thinks it has to add the whole thing again, which in the worst case, doubles the PV production, like my example from 24.09..

No, there are no errors in the log.

So, HA asks every 5min what state the sensor has? Maybe he cant connect in the right time and set a zero?

I enabled debug logging for the integration now. Maybe we get something out of it when the next error occurs.

ckarrie, 2023-09-26:
Can confirm this

JoeArny, 2023-09-26:
> Hello!
> 
> I am also having this bug. An example here: [image]
> 
> Zeroing not on midnight causes the Energy Board to add it to the same day.
> 
> In the picture I added the Energy Board for the 24.09.2023. This is a great example. PV Production goes crazy...
> 
> I have no idea how to fix this... :/

Seems that the issues are different, yours is one that was previously reported that the sensor zero itself sometimes and them comes back. This can happen at random times.
The one I'm experiencing is different because it was always steady, but every night at midnight it was supposed to zero and stay zeroed, but it jumps back to the previous value. It only zero itself properly when the panels start producing again.
Unfortunately my app is not working since the server issues they had about two weeks ago, so nothing is working anymore. :(

SiL-X, 2023-10-15:
Hello,
since I activated the debug logging mode, no bug appeared.
The only bug I have from time to time is:
```
2023-10-15 10:30:46.778 ERROR (MainThread) [alphaess.alphaess] 404, message='Not Found', url=URL('https://cloud.alphaess.com/api/Account/GetCustomUseESSList')
2023-10-15 10:30:46.780 ERROR (MainThread) [alphaess.alphaess] 404, message='Not Found', url=URL('https://cloud.alphaess.com/api/Account/GetCustomUseESSList')
2023-10-15 10:30:46.780 ERROR (MainThread) [alphaess.alphaess] 404, message='Not Found', url=URL('https://cloud.alphaess.com/api/Account/GetCustomUseESSList')
2023-10-15 10:30:46.780 ERROR (MainThread) [custom_components.alphaess] Error fetching alphaess data: 404, message='Not Found', url=URL('https://cloud.alphaess.com/api/Account/GetCustomUseESSList')
```
From time to time i got this message, but with no problems afterwards.

I can see some data points missing (maybe due to the error above) but i never get "an early 0" (and then it resets). The next  value is == or >= the last one, so it does not miscalculate.

So, in conclusion:
ATM everything is fine here, like in the past. Maybe it was a bug from Alpha and they fixed it?

CharlesGillanders, 2023-10-15:
I'm so glad it's working now for you.  I've still no idea what's happening of course....  let me know if it shows any other problems again.

SiL-X, 2023-10-23:
Hello.

I shouldnt have write that my problem is solved. >.< :/
[image]

Happened again on 17.10., 19.10. and 20.10..
3 times on 17.10.

But the timings in the log does not match with the "0 status".
On 17.10. - "0 status" is at:
10:32:54
12:55:47
14:14:41

Here is **every** AlphaESS Error Message from 17.10.:
```
2023-10-17 08:55:46.673 ERROR (MainThread) [alphaess.alphaess] 404, message='Not Found', url=URL('https://cloud.alphaess.com/api/ESS/GetLastPowerDataBySN?noLoading=true&sys_sn=*')
2023-10-17 08:55:46.675 ERROR (MainThread) [alphaess.alphaess] 404, message='Not Found', url=URL('https://cloud.alphaess.com/api/ESS/GetLastPowerDataBySN?noLoading=true&sys_sn=*')
2023-10-17 08:55:46.675 ERROR (MainThread) [custom_components.alphaess] Error fetching alphaess data: 404, message='Not Found', url=URL('https://cloud.alphaess.com/api/ESS/GetLastPowerDataBySN?noLoading=true&sys_sn=*')
```

```
2023-10-17 12:24:56.910 ERROR (MainThread) [alphaess.alphaess] 404, message='Not Found', url=URL('https://cloud.alphaess.com/api/Statistic/SystemStatistic')
2023-10-17 12:24:56.911 ERROR (MainThread) [alphaess.alphaess] 404, message='Not Found', url=URL('https://cloud.alphaess.com/api/Statistic/SystemStatistic')
2023-10-17 12:24:56.911 ERROR (MainThread) [custom_components.alphaess] Error fetching alphaess data: 404, message='Not Found', url=URL('https://cloud.alphaess.com/api/Statistic/SystemStatistic')
```

```
2023-10-17 13:09:05.659 ERROR (MainThread) [alphaess.alphaess] 404, message='Not Found', url=URL('https://cloud.alphaess.com/api/Account/GetCustomUseESSSetting?system_id=*')
2023-10-17 13:09:05.661 ERROR (MainThread) [alphaess.alphaess] 404, message='Not Found', url=URL('https://cloud.alphaess.com/api/Account/GetCustomUseESSSetting?system_id=*')
2023-10-17 13:09:05.661 ERROR (MainThread) [custom_components.alphaess] Error fetching alphaess data: 404, message='Not Found', url=URL('https://cloud.alphaess.com/api/Account/GetCustomUseESSSetting?system_id=*')
```

```
2023-10-17 13:35:07.447 ERROR (MainThread) [alphaess.alphaess] 404, message='Not Found', url=URL('https://cloud.alphaess.com/api/ESS/GetLastPowerDataBySN?noLoading=true&sys_sn=*')
2023-10-17 13:35:07.449 ERROR (MainThread) [alphaess.alphaess] 404, message='Not Found', url=URL('https://cloud.alphaess.com/api/ESS/GetLastPowerDataBySN?noLoading=true&sys_sn=*')
2023-10-17 13:35:07.449 ERROR (MainThread) [custom_components.alphaess] Error fetching alphaess data: 404, message='Not Found', url=URL('https://cloud.alphaess.com/api/ESS/GetLastPowerDataBySN?noLoading=true&sys_sn=*')
```

```
2023-10-17 14:23:13.539 ERROR (MainThread) [alphaess.alphaess] 404, message='Not Found', url=URL('https://cloud.alphaess.com/api/Account/GetCustomUseESSSetting?system_id=*')
2023-10-17 14:23:13.541 ERROR (MainThread) [alphaess.alphaess] 404, message='Not Found', url=URL('https://cloud.alphaess.com/api/Account/GetCustomUseESSSetting?system_id=*')
2023-10-17 14:23:13.541 ERROR (MainThread) [custom_components.alphaess] Error fetching alphaess data: 404, message='Not Found', url=URL('https://cloud.alphaess.com/api/Account/GetCustomUseESSSetting?system_id=*')
```

```
2023-10-17 15:47:08.743 ERROR (MainThread) [alphaess.alphaess] 404, message='Not Found', url=URL('https://cloud.alphaess.com/api/ESS/GetLastPowerDataBySN?noLoading=true&sys_sn=*')
2023-10-17 15:47:08.745 ERROR (MainThread) [alphaess.alphaess] 404, message='Not Found', url=URL('https://cloud.alphaess.com/api/ESS/GetLastPowerDataBySN?noLoading=true&sys_sn=*')
2023-10-17 15:47:08.746 ERROR (MainThread) [custom_components.alphaess] Error fetching alphaess data: 404, message='Not Found', url=URL('https://cloud.alphaess.com/api/ESS/GetLastPowerDataBySN?noLoading=true&sys_sn=*')
```

Stupid idea / question:
I have a template sensor which calculates the pv surplus:
sensor.*instantaneous_generation - sensor.*instantaneous_load

**Everytime** I get the above listed error message, I also get these error message, e.g. at 15.47.08:
```
2023-10-17 15:47:08.749 ERROR (MainThread) [homeassistant.components.template.template_entity] TemplateError('ValueError: Template error: int got invalid input 'unavailable' when rendering template '{% set pv_uberschuss = (states('sensor.*_instantaneous_generation') | int - states('sensor.*_instantaneous_load') | int ) %} {% if pv_uberschuss < 0 %}
  0
{% else %}
  {{ pv_uberschuss }}
{% endif %}' but no default was specified') while processing template 'Template<template=({% set pv_uberschuss = (states('sensor.*_instantaneous_generation') | int - states('sensor.*_instantaneous_load') | int ) %} {% if pv_uberschuss < 0 %}
  0
{% else %}
  {{ pv_uberschuss }}
{% endif %}) renders=9652>' for attribute '_attr_native_value' in entity 'sensor.pv_uberschuss'
```

I dont know how it really works. But is it maybe possible that the template sensor wants to refresh, asks the AlphaESS Server for new data and sometimes the alphaess server refuses the update cause off already too much load or too many request or too many request from my system? And then this bug happens?
Or is it vice versa, the template sensor refreshes after it gets new data from the alpha server so the template error is an outcome of the error?

Well, I am not the professional. It´s all hobby based here. But with my knowledge i have, i would say its not an implementation bug, more likely an AlphaESS Server Problem?

Another idea:
Is it possible to say HomeAssistant that it hould not reset before "GMT 0:00"?

CharlesGillanders, 2023-11-29:
duplicate of #85

closed 2023-11-29 by CharlesGillanders