CharlesGillanders/homeassistant-alphaESS #76: No Data since today's Alpha ESS Update

cpmueller, 2023-11-04:
Hi,
Since the update from Alpha ESS, the data query no longer works.
I get the following error:

```
2023-11-04 09:46:13.375 ERROR (MainThread) [alphaess.alphaess] 'data'
2023-11-04 09:46:13.376 ERROR (MainThread) [custom_components.alphaess] Unexpected error fetching alphaess data: 'data'
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 290, in _async_refresh
self.data = await self._async_update_data()

File "/config/custom_components/alphaess/coordinator.py", line 30, in _async_update_data
jsondata: json = await self.api.getdata()

File "/usr/local/lib/python3.11/site-packages/alphaess/alphaess.py", line 154, in getdata
units = await self.__get_data("Account/GetCustomMenuESSList")

File "/usr/local/lib/python3.11/site-packages/alphaess/alphaess.py", line 176, in __get_data
if not await self.__connection_check():

File "/usr/local/lib/python3.11/site-packages/alphaess/alphaess.py", line 147, in __connection_check
return await self.authenticate(self.username, self.password)

File "/usr/local/lib/python3.11/site-packages/alphaess/alphaess.py", line 67, in authenticate
if "AccessToken" in json_response["data"]:

KeyError: 'data'
2023-11-04 09:46:13.379 DEBUG (MainThread) [custom_components.alphaess] Finished fetching alphaess data in 0.525 seconds (success: False)
```

Thank you for looking into that issue!

MorneSaunders360, 2023-11-04:
It appears that Alpha ESS has updated its API authentication protocol, transitioning from a simple password-based login to a more secure method that involves hashing the password prior to login. Additionally, the login endpoint has been modified; what was previously accessed via https://cloud.alphaess.com/api/Account/Login has now been changed to https://cloud.alphaess.com/api/base/user/login.

MorneSaunders360, 2023-11-04:
Code response able for creating the encrypted password.
Hope this helps.

export function encrypt(password, username) {
  let aeskey = CryptoJS.SHA256(username)
  let iv = CryptoJS.MD5(username)
  let encrypted = CryptoJS.AES.encrypt(password, aeskey, {
    iv: iv,
    mode: CryptoJS.mode.CBC,
    padding: CryptoJS.pad.Pkcs7,
  })
  return encrypted.toString()
}

ckarrie, 2023-11-04:
- password is sent encrypted to https://cloud.alphaess.com/api/base/user/login (see @MorneSaunders360)
- Header "Authorization" is not "Bearer eyJ..." any more, its without Bearer prefix now
- Header "Authsignature" and "Authtimestamp" is gone

weissjuergen, 2023-11-04:
It seems some of you know how to get the integration to work again. I am a bloody user only.
Can anyone advise what I would need to do in my home Assistent or in the integration files to fix the issue, greatly appreciated.

Poshy163, 2023-11-04:
Someone needs to make a pull request with these changes, and hopefully it gets pulled, packaged and updated by the repo owner

strickersuk, 2023-11-04:
If it's doable then Charles is usually pretty quick to fix, just hang tight.

matt-oneill, 2023-11-04:
I’ve posted a quick fix on an issue to the alphaess library, hopefully Charles can publish a new version which will sort it temporarily. 
Alpha are trying to push people to use their new public api but from what I’ve seen is still unstable and doesn’t have all functionality

CharlesGillanders, 2023-11-04:
Thanks Matt!  I can get this pulled and an update out later today.  Much appreciated.

DavidqStokes, 2023-11-04:
Charles and other contributors all do this voluntarily and are usually very good and diagnosing issues, updating libraries, testing and issuing an update to the HA integration.

Everyone needs to be respectful of their time and efforts. If people don't understand that this is the way that open source software works then HA is probably the wrong home automation solution for them.

Charles, Matt et al - thank you so much for your continued work on this integration and for putting up with the challenges of working with AlphaESS.

CharlesGillanders, 2023-11-04:
I've pushed a beta update to through HACS that uses @matt-oneill 's temporary workaround of changing the base URL for the API.  I'm going to work on incorporating the changes identified by @MorneSaunders360 and @ckarrie

ckarrie, 2023-11-04:
Nice workaround! Thank you!

matt-oneill, 2023-11-04:
Nice one Charles. I had a look at the other apis earlier and a lot of them have changed the path so not a simple blanket change to the login endpoint.
Maybe it’s time to re visit the new api

strickersuk, 2023-11-04:
Thanks folks, it’s working for me.

CharlesGillanders, 2023-11-04:
> Nice one Charles. I had a look at the other apis earlier and a lot of them have changed the path so not a simple blanket change to the login endpoint. Maybe it’s time to re visit the new api

I've found several equivalents, most of them in fact but the missing one for me is the Discharge numbers - I used to grab them from system statistics but I cannot see any mention of discharge in any of the results I'm getting back in the web interface now.....

braindeadmalc, 2023-11-04:
I'm sorry to be a pain but what do I need to do to get the fix applied? I'm not an expert at the internals of HA, just an enthusiastic amateur.

weissjuergen, 2023-11-04:
> I'm sorry to be a pain but what do I need to do to get the fix applied? I'm not an expert at the internals of HA, just an enthusiastic amateur.

Same for me, I have the same question.

CharlesGillanders, 2023-11-04:
If you deployed using HACS then should just be a matter of finding the AlphaESS repo in HACS and click on update information in the  three dots menu.  Navigate back to the main list of HACS integrations and you should see an update button for AlphaESS.

braindeadmalc, 2023-11-04:
> If you deployed using HACS then should just be a matter of finding the AlphaESS repo in HACS and click on update information in the three dots menu. Navigate back to the main list of HACS integrations and you should see an update button for AlphaESS.

Thanks, you put me on the right track, sort of. I went into HACS, clicked on the Alpha Integration then from the  Menu selected Redownload with the "show Betas" option, then selected the latest Beta you released.
An HA restart and all seems good.

Thanks for your help and the integration. much appreciated.

@weissjuergen , hope this helps

weissjuergen, 2023-11-04:
> @weissjuergen , hope this helps

Yes, it did, I can see my entities again. Thank you Charles !

Rasenkardinal, 2023-11-04:
Amazing! Thank you so much! Ist there any oportunity to support your work with money? Would be a pleasure for me.

CharlesGillanders, 2023-11-04:
> Amazing! Thank you so much! Ist there any oportunity to support your work with money? Would be a pleasure for me. 

No need for any cash - I do this because I have my own solar system to manage.  This fix is only temporary, so don't be alarmed if it doesn't last for long - it's dependent on Alpha being slow to update another URL but I think we all know that's unlikely to last forever.

TheSimTek, 2023-11-04:
Thank you Charles.  However, updating to the latest beta version did not work for me here in Australia.  Will wait until you release a fix.

wallus, 2023-11-05:
I downloaded the beta version (v0.3.4beta) and it is working correctly again. Thanks Charles!

foylema, 2023-11-05:
Just updated to the BETA and it's all working again. Thanks for fixing this so quickly.

ceeessbee, 2023-11-05:
+1, BETA looks good. To all the experts who know what they're doing with this, big thumbs up, and to you Charles for the time and effort to bring this to us all. HA is great but nothing without people to bring us the goodies!

weissjuergen, 2023-11-06:
All was working fine since the temp fix. A few min ago the Connection ist broken again. Can you check your systems and see this issue is wide spread or just me.

Poshy163, 2023-11-06:
Appears to be working, check your systems connection to the cloud, seems to be a issue on your end.

tasranson, 2023-11-07:
Unfortunately I still can't get it to Configure the integration,  It gives Unknown Error after entering UserName and Password and the logs show the following:

Logger: custom_components.alphaess.config_flow
Source: custom_components/alphaess/config_flow.py:66
Integration: Alpha ESS ([documentation](https://github.com/CharlesGillanders/homeassistant-alphaESS))
First occurred: 14:21:23 (3 occurrences)
Last logged: 14:24:50

Unexpected exception
Traceback (most recent call last):
  File "/config/custom_components/alphaess/config_flow.py", line 66, in async_step_user
    info = await validate_input(self.hass, user_input)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/alphaess/config_flow.py", line 44, in validate_input
    if not await client.authenticate(data["username"],data["password"]):
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/alphaess/alphaess.py", line 60, in authenticate
    if "AccessToken" in json_response["data"]:
                        ~~~~~~~~~~~~~^^^^^^^^
KeyError: 'data'

I expect I still have remants of the old version on my system, but I can not seem to find out how to clean it out and start again.  I have removed the integration previously and added the repository back in today, but setting up the integration gives the same error.   I must be doing something wrong, but have not idea what.   Any hints on how to get the new version sorted in this scenario?

CharlesGillanders, 2023-11-07:
> scenario?

It's maybe worth trying to completely remove the component and the associated library manually and then trying again.

I think the library can be deleted by removing the entire folder /usr/local/lib/python3.11/site-packages/alphaess

The component can be deleted by removing the sub-folder custom_components/alphaess/ under your home assistant configuration folder.

ElevenFan, 2023-11-07:
Our next version of Web API will add anti-crawling function. Please move to OpenAPI, otherwise you will update with us every time.Thanks。Open api we are slowly optimizing the request limit。

ckarrie, 2023-11-07:
Challenge accepted

CharlesGillanders, 2023-11-07:
I have a working version of the component using the AlphaESS OpenAPI running in test.  Before I consider pushing it out as an updated release I want to give everyone a heads up that the Open API will **require** you to register with the Alpha ESS developer portal and obtain an AppID and an AppSecret for yourself and to also bind your own inverter to that AppID in the portal.

If anyone hasn't already registered with the AlphaESS OpenAPI portal then this component will stop working when I push the updated release.  

I will put a notification on the main page of the GitHub repository as well but for everyone following this thread - please register with [https://open.alphaess.com/](https://open.alphaess.com/) as soon as you are able.

thospe, 2023-11-07:
done, looking forward to a more reliable and stable connection.

It looks like they (AlphaESS) are having other software problems as well in their monitoring. Currently my wallbox usage isn’t reported properly anymore - and haha, they told me to check the correct cabling.

Rasenkardinal, 2023-11-07:
That really gets things moving. The registration is complete. Ready when you are!

And again: Thanks a lot!

braindeadmalc, 2023-11-07:
Registration complete but how the heck do you bind your system? When I try and add mine it asks for the system S/N which I guess is the one beginning AL rather than the actual Inverter S/N? I have both. But then it wants  CheckCode which is where I get unstuck. Anyone any idea what this is pls? I assume after entering that I may get the Verification Code.

strickersuk, 2023-11-07:
It’s on the side of the unit.

braindeadmalc, 2023-11-07:
> It’s on the side of the unit.

Well I'll be!
Never thought to look at the hardware itself. Doh!

Thanks for that

ceeessbee, 2023-11-08:
Registered. Thanks for the ongoimng work

ZeeDK, 2023-11-11:
Looking very much forward to the fix. Realized (for the last ~week it has been down) that this is the most important integration in my HA setup :D

Didn't try the "beta", but will try out the fix as soon as HACS gets the update :)

CharlesGillanders, 2023-11-14:
I have published release 0.4.0 of the HomeAssistant custom component - this is the switch to using the official OpenAPI.

There are some details on how to switch an existing deployment to use the new API in the Github README [https://github.com/CharlesGillanders/homeassistant-alphaESS/blob/main/README.md](https://github.com/CharlesGillanders/homeassistant-alphaESS/blob/main/README.md)

jasonwins, 2023-11-14:
Downloaded and it works for me. 
Huge thanks Charles

umrath, 2023-11-14:
Works like a charm.
One minor issue: The dialog to add the credentials does not have any kind of description for me. All I saw were 2 unnamed input fields.

kinektek, 2023-11-14:
Working like a dream, thank you for all the efforts!

strickersuk, 2023-11-14:
Thanks Charles, all working.

matt-oneill, 2023-11-14:
Thanks @CharlesGillanders great work.
Same for me  - but its working again hopefully we won't have to keep chasing api changes
[image]

braindeadmalc, 2023-11-14:
Thanks for this update
I think I've found a problem when calling the Alpha ESS: Set Battery Charge
service, It just returns an unknown error when I call it.

DavidqStokes, 2023-11-14:
I haven't yet tried to call the battery services (I use an automation later in the evening)

CharlesGillanders, 2023-11-14:
> Thanks for this update I think I've found a problem when calling the Alpha ESS: Set Battery Charge service, It just returns an unknown error when I call it.
> […](#)

Would you mind opening a new issue to track that separately please?

braindeadmalc, 2023-11-14:
Issue opened as requested.

DarkSlice1, 2023-11-14:
im using the new API (outside of home assistant), the set charge is a PUT call now
https://github.com/DarkSlice1/AlphaEss_Monitor/blob/dumpZappi/src/main/scala/api/alpha/alpha.scala#L108

ZeeDK, 2023-11-14:
AWESOME to have this integration running again!
As matt-oneill mentioned I also didn't have any labels in the "credentials prompt".
[image]

But the most important thing is that it works.
I went from having 21 to 23 entities after adding the new "integration entry", don't know what the new ones are but everything works as before :)

Thanks for the great work Charles!

Rasenkardinal, 2023-11-14:
I can confirm that the new version of the integration works as expected. 
Well done Charles!

ceeessbee, 2023-11-14:
Thank you Charles, all working on the new version.

CharlesGillanders, 2023-11-14:
> But the most important thing is that it works. I went from having 21 to 23 entities after adding the new "integration entry", don't know what the new ones are but everything works as before :)
> 

The two new entities are additional PPV3, and PPV4 real time PV power generation figures - I don't know which AlphaESS inverters support four strings of PV (my install only has two) but if you are lucky enough to have them the new API displays them.

CharlesGillanders, 2023-11-14:
> I haven't yet tried to call the battery services (I use an automation later in the evening)

There was an error with changing the charge/discharge settings - please make sure to update to 0.4.1 which fixes this.

closed 2023-11-14 by CharlesGillanders