CharlesGillanders/homeassistant-alphaESS #78: Sensor for PV Power Forcasteing

ernstmosch, 2023-11-10:


ernstmosch, 2023-11-10:
Will see if i can do some first steps on my own.

closed 2023-11-10 by ernstmosch

DavidqStokes, 2023-11-10:
I have added the existing Forecast.Solar (https://www.home-assistant.io/integrations/forecast_solar) integration to HA and use the data from that to calculate whether to charge or discharge or do neither with my battery overnight overnight. I then use the AlphaESS integration to update AlphaESS charge/discharge setting around 22.45 hrs every night. The Forecast.Solar forecast isn't perfect (it can get fooled by low cloud or fog coming in from the North Sea - I'm 15km from the coast), but it's reliable enough that I don't look at it most evenings.
I reckon it's helped to reduce my overall grid imports an appreciable amount (we now self consume ~90 - 95% of our PV, compared to around 30% before we installed the battery. Around 95+% of our grid use is now on cheaper overnight tariffs)