CharlesGillanders/homeassistant-alphaESS #80: Version 4.0 - Calling Set Battery Charge service returns error

braindeadmalc, 2023-11-14:
When I call the Alpha ESS: Set Battery Charge service it doesn't set the battery parameters and it returns Unknown Error in my automation.

If I can provide more info, please let me know

CharlesGillanders, 2023-11-14:
> After Update to Alpha ESS v.4.0 logon failed. Error Message " Unknown error". Can you pleased fixed?

Did you follow the instructions [https://github.com/CharlesGillanders/homeassistant-alphaESS#alphaess-openapi](https://github.com/CharlesGillanders/homeassistant-alphaESS#alphaess-openapi) to change the authentication to use your OpenAPI developer AppID and AppSecret?

milkmanrox, 2023-11-14:
I too am getting the error when it comes to using the service to set charge.

Attached is the debug logs.
[home-assistant_alphaess_2023-11-14T20-11-13.301Z.log](https://github.com/CharlesGillanders/homeassistant-alphaESS/files/13355679/home-assistant_alphaess_2023-11-14T20-11-13.301Z.log)

CharlesGillanders, 2023-11-14:
Apologies all, stupid mistake on my part - just released 0.4.1 which fixes this.

braindeadmalc, 2023-11-14:
Everyone makes mistakes.
Thanks for you work on this. It is appreciated.

DavidqStokes, 2023-11-14:
Thanks as always for your work on getting this new version out (and to everyone who helped with diagnosing / designing that) and for getting this latest fix out so quickly.

Your hard work and timeliness are always appreciated.

milkmanrox, 2023-11-14:
Can confirm that 0.4.1 has fixed the issue, thanks once again.

DavidqStokes, 2023-11-15:
Also for me - my overnight battery charge was set as usual.

closed 2023-11-16 by CharlesGillanders

J-89436, 2023-11-16:
> Calling Set Battery Charge service returns error (Issue #80)
>
>
>
> Closed #80
> <https://github.com/CharlesGillanders/homeassistant-alphaESS/issues/80>  as
> completed.
>
> -
> <https://github.com/CharlesGillanders/homeassistant-alphaESS/issues/80#event
> YYUJAVCNFSM6AAAAAA7K76M46VHI2DSMVQWIX3LMV45UABCJFZXG5LFIV3GK3TUJZXXI2LGNFRWC
> 5DJN5XDWMJQHE4DENJWGEZTEMQ> .
> <https://github.com/notifications/beacon/AO3AL57RW6UK2MVAOGGWUVTYEYYUJA5CNFS
> M6AAAAAA7K76M46WGG33NNVSW45C7OR4XAZNWJFZXG5LFIV3GK3TUJZXXI2LGNFRWC5DJN5XKUY3
> ithub.com
> <mailto:CharlesGillanders/homeassistant-alphaESS/issue/80/issue_event/109825

It work fine. Many thank for your support.

 

Best regards

 

Joerg Voelschow