CharlesGillanders/homeassistant-alphaESS #82: PV Power Forecast

ernstmosch, 2023-11-16:
Hi Charles,

do you think we can get the Forecast into Home-Assistant?

[image]

I tried some research but my network sniffing is to bad.

Thanks you in advance, Thomas/Michael

CharlesGillanders, 2023-11-16:
No - that's unfortunately not included in the API that AlphaESS provide.  On the other hand I know that other people have been successful using other non AlphaESS 3rd party solar forecast integrations.

For example see [https://www.home-assistant.io/integrations/forecast_solar/](https://www.home-assistant.io/integrations/forecast_solar/)  as well as [https://powerforum.co.za/topic/19441-solar-forecasting-in-home-assistant/](https://powerforum.co.za/topic/19441-solar-forecasting-in-home-assistant/)

closed 2023-11-16 by CharlesGillanders

DavidqStokes, 2023-11-16:
I'm using the Forecast.Solar integration - which also works with the HA Energy dashboard. 
I use the Forecast.Solar forecast to determine how much to charge my AlphaESS battery overnight, based on the current state of charge (SOC) and the forecast generation from my (separate) SolarEdge PV system.
Alternatively, if the next day has plenty of PV power forecast and the SOC is okay, I either allow the battery to discharge overnight, to neither charge/discharge overnight.
I find the Forecast.Solar forecast to be reliable enough to not worry about setting this manually.
As long as you have the details of your solar panels (location, orientation, peak generation) it should work with any connected PV system.