CharlesGillanders/homeassistant-alphaESS #84: Integrations stops working

dan-s-github, 2023-11-18:
I installed new Integration on Friday afternoon and noticed that the Energy dashboard didn't show any values on Saturday.

After rebooting Home Assistant values came back.

Same thing happened today and I restarted Home Assistant again, waiting for Energy to show data

Startup log looks mostly fine

```
2023-11-19 08:00:10.909 WARNING (Recorder) [homeassistant.components.sensor.recorder] Entity sensor.al2002118050331_solar_to_battery from integration alphaess has state class total_increasing, but its state is negative. Triggered by state -0.3 with last_updated set to 2023-11-18T18:54:59.999999+00:00. Please create a bug report at https://github.com/CharlesGillanders/homeassistant-alphaESS/issues
2023-11-19 08:14:10.626 DEBUG (MainThread) [custom_components.alphaess] Finished fetching alphaess data in 3.099 seconds (success: True)
```

checking home-assistant.log.1 I found these issues

```
--
2023-11-18 17:33:23.268 ERROR (MainThread) [alphaess.alphaess] 502, message='Bad Gateway', url=URL('https://openapi.alphaess.com/api/getEssList')
2023-11-18 17:33:23.268 ERROR (MainThread) [alphaess.alphaess] 502, message='Bad Gateway', url=URL('https://openapi.alphaess.com/api/getEssList')
2023-11-18 17:33:23.269 ERROR (MainThread) [custom_components.alphaess] Error fetching alphaess data: 502, message='Bad Gateway', url=URL('https://openapi.alphaess.com/api/getEssList')
--
2023-11-19 00:00:51.156 ERROR (MainThread) [custom_components.alphaess] Unexpected error fetching alphaess data: 'NoneType' object has no attribute 'get'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 290, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/alphaess/coordinator.py", line 45, in _async_update_data
    _pv =  _onedateenergy.get("epv")
           ^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'get'
2023-11-19 00:01:53.919 ERROR (MainThread) [custom_components.alphaess] Unexpected error fetching alphaess data: 'NoneType' object has no attribute 'get'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 290, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/alphaess/coordinator.py", line 45, in _async_update_data
    _pv =  _onedateenergy.get("epv")
           ^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'get'
2023-11-19 00:02:56.932 ERROR (MainThread) [custom_components.alphaess] Unexpected error fetching alphaess data: 'NoneType' object has no attribute 'get'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 290, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/alphaess/coordinator.py", line 45, in _async_update_data
    _pv =  _onedateenergy.get("epv")
           ^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'get'
2023-11-19 00:03:59.887 ERROR (MainThread) [custom_components.alphaess] Unexpected error fetching alphaess data: 'NoneType' object has no attribute 'get'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 290, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/alphaess/coordinator.py", line 45, in _async_update_data
    _pv =  _onedateenergy.get("epv")
           ^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'get'
--
2023-11-19 03:05:10.388 WARNING (Recorder) [homeassistant.components.sensor.recorder] Entity sensor.al2002118050331_solar_to_battery from integration alphaess has state class total_increasing, but its state is negative. Triggered by state -0.1 with last_updated set to 2023-11-18T14:04:01.891520+00:00. Please create a bug report at https://github.com/CharlesGillanders/homeassistant-alphaESS/issues
```

What I also noticed that solar production was not reset at midnight the day before (#75) affects also other totals but not all, somehow grid_to_battery was not reset yesterday but was today while others didn't reset at all.

[image]

Service seem to work without issues

milkmanrox, 2023-11-19:
I was about to open an issue about the same thing. Everything with the new API works except the Energy tab reporting.

I am in GMT+10, and I think what I'm seeing is that the values start working by themselves by 10am.

dan-s-github, 2023-11-19:
Interesting observation and I had to check

I am in GMT+13 and can confirm that my values also started to show up at 1pm

dan-s-github, 2023-11-19:
three sensors were reset at midnight
[image]
and five at 1pm
[image]
one sensor did not some other jump
[image]

will check if this will happen again tomorrow

Poshy163, 2023-11-19:
Dupe of #85, Seems to be a upstream issue, has been reported [here](https://github.com/alphaess-developer/alphacloud_open_api/issues/44)

dan-s-github, 2023-11-20:
last nights log entry around midnight
```
2023-11-20 00:00:07.646 ERROR (MainThread) [custom_components.alphaess] Unexpected error fetching alphaess data: 'NoneType' object has no attribute 'get'
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 290, in _async_refresh
self.data = await self._async_update_data()
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/config/custom_components/alphaess/coordinator.py", line 45, in _async_update_data
_pv = _onedateenergy.get("epv")
^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'get'
2023-11-20 00:01:10.619 ERROR (MainThread) [custom_components.alphaess] Unexpected error fetching alphaess data: 'NoneType' object has no attribute 'get'
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 290, in _async_refresh
self.data = await self._async_update_data()
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/config/custom_components/alphaess/coordinator.py", line 45, in _async_update_data
_pv = _onedateenergy.get("epv")
^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'get'
```
and two other entries after 1pm
```
2023-11-20 13:45:10.539 WARNING (Recorder) [homeassistant.components.sensor.recorder] Entity sensor.al2002118050331_solar_to_load from integration alphaess has state class total_increasing, but its state is not strictly increasing. Triggered by state 12.6 (12.8) with last_updated set to 2023-11-20T00:40:30.680668+00:00. Please create a bug report at https://github.com/CharlesGillanders/homeassistant-alphaESS/issues
2023-11-20 13:45:10.539 WARNING (Recorder) [homeassistant.components.sensor.recorder] Entity sensor.al2002118050331_total_load from integration alphaess has state class total_increasing, but its state is not strictly increasing. Triggered by state 12.6 (12.8) with last_updated set to 2023-11-20T00:40:30.680994+00:00. Please create a bug report at https://github.com/CharlesGillanders/homeassistant-alphaESS/issues
``` 
History panel views
[image]
[image]

CharlesGillanders, 2023-11-29:
Duplicat of Issue 85

closed 2023-11-29 by CharlesGillanders