CharlesGillanders/homeassistant-alphaESS #85: Sensors not reseting at local midnight - Instead at UTC, and not returning to Zero - since new version

gregpetersau, 2023-11-19:
Hi.
Since the new version for the open API my daily cumulative sensors like Charge, Discharge etc are no longer reseting at midnight - instead they are reseting at UTC Midnight, which for me is 11am. 

Secondly, they are not reseting to Zero but some random amount.

Any suggestions?

BTW thanks for the amzing work at geting the new API up and running.

[image]

Poshy163, 2023-11-19:
Yep, i have the same issue, based me being in +10:30 timzone, it seems to reset at the same time, this seems to be an upstream issue tho, so may need to lodge it here (https://github.com/alphaess-developer/alphacloud_open_api)

[image]

Poshy163, 2023-11-19:
Pushed upstream [here](https://github.com/alphaess-developer/alphacloud_open_api/issues/44)

Poshy163, 2023-11-22:
Apologies, digging into it now, appears to be a issue with the requests being pushed with the UTC timezone, will work on a fix soon and make a pull soon

dan-s-github, 2023-11-22:
could this error
```
2023-11-22 00:03:14.647 ERROR (MainThread) [custom_components.alphaess] Unexpected error fetching alphaess data: 'NoneType' object has no attribute 'get'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 290, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/alphaess/coordinator.py", line 45, in _async_update_data
    _pv =  _onedateenergy.get("epv")
           ^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'get'
```
have an impact on not resetting data at midnight?
This pops up daily in my HA logs.

gregpetersau, 2023-11-22:
I also get that error daily at midnight

CharlesGillanders, 2023-11-22:
> Apologies, digging into it now, appears to be a issue with the requests being pushed with the UTC timezone, will work on a fix soon and make a pull soon

I am looking at the date.today() call in my OpenAPI library and I'm presuming that's the source of the issue - that date call is not timezone aware.  I see there are alternative libraries that might be better - if you have any proposed code I'd be happy to pull it.

Poshy163, 2023-11-23:
Made a pull request [here](https://github.com/CharlesGillanders/alphaess-openAPI/pull/1)

This should work? time class is handled locally and then interpreted, rather than grabbing the UTC date

CharlesGillanders, 2023-11-23:
Thanks to @Poshy163 - I have updated the underlying library and pushed version 0.4.3 of the home assistant integration which should fix this issue.  I run my own Home Assistant in UTC so it will need someone in a different timezone to update and confirm if the problem is resolved.

gregpetersau, 2023-11-23:
Testing and will report back - its been terrible weather here so hopefully i can validate using something like grid to load vs the charge. Assuming i will also not see a reset to zero at UTC Midnight in about 1 hr time (11am local)

Poshy163, 2023-11-23:
Maybe give it a day or so, it should reset locally. The pull to the API was done past midnight locally for me, but from my understanding it may be fixed

gregpetersau, 2023-11-25:
1 day in doesnt look fixed. 

Still reseting values at 11am locat (midnight UTC). THe value that it resets too is the correct value based on todays production etc.
[image]

Note in the above a few (Solar to load, Solar to battery) are resetting correctly, but i just checked turns out those modules were reseting correctly all along.... weirid.

[image]

Poshy163, 2023-11-25:
Yeah, noticed that as well, will have a crack at going though it tonight. But it seems to be an issue on the intergration side, not the API? Anyways, will do some debugging and make another pull soon

dan-s-github, 2023-11-25:
I actually mentioned that some sensors are reseting correctly and some don't in #84.

Also I still get the error mentioned in [previous comment](https://github.com/CharlesGillanders/homeassistant-alphaESS/issues/85#issuecomment-1822241433) and would be interesting if users running in UTC also get this?

Poshy163, 2023-11-26:
Yeah, its more about pinning a few specific calls being made, and how its interpreted (from the API). 

Doing some local testing at the moment, will make a pull when i think ive solved it.

Poshy163, 2023-11-26:
Just an update, appears when it is past midnight (for any time + of UTC), the 'OneDateEnergy' Function appears to return
`'OneDateEnergy': None` 
and it holds all the statistics related to it, until midnight UTC is passed where it can re-poll, will double check the API in the morning and see whats going on

Going over the postman API pulls, it appears that using the query date is being pulled from a 
`pm.collectionVariables.set('CurrentDate', moment.utc().format(("YYYY-MM-DD")));`
so, no doubt something like this is implemented within the code of the API to get the date, to hence make the request with the past date to the API. 

Will continue to find out where the issue could be, should be a pull soon

Poshy163, 2023-11-26:
Pull here with more details 
[Temp fix to the wrong date being passed through into getOneDateEnergyBySn() ](https://github.com/CharlesGillanders/alphaess-openAPI/pull/2)

gregpetersau, 2023-11-28:
> Pull here with more details [Temp fix to the wrong date being passed through into getOneDateEnergyBySn() ](https://github.com/CharlesGillanders/alphaess-openAPI/pull/2)

Happy to test this but since it's sitting in pull on the source API page how do I get it into mine to try. Sorry not a programmer just a tech savvy old dude

Poshy163, 2023-11-28:
Not much you can do as the API is external (no direct script you can find on HA using visual studio plugin),

I have an idea of a more permanent solution, which I'll test past midnight ACST and update the pull, and notify Charles

CharlesGillanders, 2023-11-29:
Has anyone seen any improvement after moving to release 0.4.4 (from 28th November)?

gregpetersau, 2023-11-29:
Unfortunately it seems to have made no difference.

Poshy163, 2023-11-29:
Yep, same. This seemed to be a solution when tested the API with the test application, so might be something on the home assistant end of it

CharlesGillanders, 2023-11-29:
> Yep, same. This seemed to be a solution when tested the API with the test application, so might be something on the home assistant end of it 

I spotted this old home assistant post discussing time zone awareness https://www.home-assistant.io/blog/2015/05/09/utc-time-zone-awareness/  does anyone want to try setting their local time zone in the yaml config as shown?

bmannz, 2023-12-04:
Hi Charles
I have installed the new version and enabled debug and it's output the following at 8:25am local time
`Entity sensor.al20022210x_solar_to_battery from integration alphaess has state class total_increasing, but its state is negative. Triggered by state -0.4 with last_updated set to 2023-12-04T19:21:04.859358+00:00. Please report it to the custom integration author.`

What other information would you like us to provide to assist you with this?  We are GMT+13

CharlesGillanders, 2023-12-04:
> Hi Charles
>
> I have installed the new version and enabled debug and it's output the following at 8:25am local time
>
> `Entity sensor.al20022210x_solar_to_battery from integration alphaess has state [...]

You should see some alphaess.alphaess debug statements in your logs - it's those I need to see, specifically any that mention date.

bmannz, 2023-12-04:
O)k great,  Here are logs.  Current local time is shown in the logs.
`2023-12-05 09:01:43.598 DEBUG (MainThread) [alphaess.alphaess] [{'sysSn': 'AL2002221x', 'popv': 5.0, 'minv': 'SMILE5-INV', 'poinv': 5.0, 'cobat': 10.1, 'mbat': 'SMILE-BAT-10.1P', 'surplusCobat': 9.1, 'usCapacity': 90.0, 'emsStatus': 'Normal', 'SumData': {'epvtoday': 17.7, 'epvtotal': 14706.7, 'eload': 31.02, 'eoutput': 0.76, 'einput': 14.08, 'echarge': 10.8, 'edischarge': 9.9, 'todayIncome': 7.39, 'totalIncome': 4802.04, 'eselfConsumption': 0.72528, 'eselfSufficiency': 0.43136, 'treeNum': 2517.9697, 'carbonNum': 12647.762, 'moneyType': 'NZD'}, 'OneDateEnergy': {'sysSn': 'AL2002221x', 'theDate': '2023-12-05', 'eCharge': 7.0, 'epv': 1.9, 'eOutput': 0.0, 'eInput': 14.41, 'eGridCharge': 6.9, 'eDischarge': 1.0, 'eChargingPile': 0.0}, 'LastPower': {'ppv': 2028.0, 'ppvDetail': {'ppv1': 1018.0, 'ppv2': 1010.0, 'ppv3': 0.0, 'ppv4': 0.0, 'pmeterDc': 0.0}, 'soc': 80.8, 'pev': 0, 'pevDetail': {'ev1Power': 0, 'ev2Power': 0, 'ev3Power': 0, 'ev4Power': 0}, 'prealL1': 473.0, 'prealL2': 0.0, 'prealL3': 0.0, 'pgrid': 0.0, 'pgridDetail': {'pmeterL1': 0.0, 'pmeterL2': 0.0, 'pmeterL3': 0.0}, 'pbat': -1367.0, 'pload': 661.0}, 'ChargeConfig': {'gridCharge': 1, 'timeChaf1': '05:30', 'timeChae1': '07:00', 'timeChaf2': '00:00', 'timeChae2': '00:00', 'batHighCap': 100.0}, 'DisChargeConfig': {'ctrDis': 1, 'timeDisf1': '07:00', 'timeDise1': '17:00', 'timeDisf2': '17:00', 'timeDise2': '06:00', 'batUseCap': 15.0}}]
2023-12-05 09:01:43.599 DEBUG (MainThread) [custom_components.alphaess] Finished fetching alphaess data in 3.974 seconds (success: True)
2023-12-05 09:02:43.624 DEBUG (MainThread) [alphaess.alphaess] Trying to call https://openapi.alphaess.com/api/getEssList
2023-12-05 09:02:44.599 DEBUG (MainThread) [alphaess.alphaess] Trying to call https://openapi.alphaess.com/api/getSumDataForCustomer?sysSn=AL2002221x
2023-12-05 09:02:45.416 DEBUG (MainThread) [alphaess.alphaess] DATE: 2023-12-05
2023-12-05 09:02:45.416 DEBUG (MainThread) [alphaess.alphaess] DATE: 2023-12-05
2023-12-05 09:02:45.417 DEBUG (MainThread) [alphaess.alphaess] Trying to call https://openapi.alphaess.com/api/getOneDateEnergyBySn?sysSn=AL2002221x&queryDate=2023-12-05
2023-12-05 09:02:46.040 DEBUG (MainThread) [alphaess.alphaess] Trying to call https://openapi.alphaess.com/api/getLastPowerData?sysSn=AL2002221x
2023-12-05 09:02:46.655 DEBUG (MainThread) [alphaess.alphaess] Trying to call https://openapi.alphaess.com/api/getChargeConfigInfo?sysSn=AL2002221x
2023-12-05 09:02:47.272 DEBUG (MainThread) [alphaess.alphaess] Trying to call https://openapi.alphaess.com/api/getDisChargeConfigInfo?sysSn=AL2002221x
2023-12-05 09:02:47.941 DEBUG (MainThread) [alphaess.alphaess] [{'sysSn': 'AL2002221x', 'popv': 5.0, 'minv': 'SMILE5-INV', 'poinv': 5.0, 'cobat': 10.1, 'mbat': 'SMILE-BAT-10.1P', 'surplusCobat': 9.1, 'usCapacity': 90.0, 'emsStatus': 'Normal', 'SumData': {'epvtoday': 17.7, 'epvtotal': 14706.7, 'eload': 31.02, 'eoutput': 0.76, 'einput': 14.08, 'echarge': 10.8, 'edischarge': 9.9, 'todayIncome': 7.39, 'totalIncome': 4802.04, 'eselfConsumption': 0.72528, 'eselfSufficiency': 0.43136, 'treeNum': 2517.9697, 'carbonNum': 12647.762, 'moneyType': 'NZD'}, 'OneDateEnergy': {'sysSn': 'AL2002221x', 'theDate': '2023-12-05', 'eCharge': 7.0, 'epv': 1.9, 'eOutput': 0.0, 'eInput': 14.41, 'eGridCharge': 6.9, 'eDischarge': 1.0, 'eChargingPile': 0.0}, 'LastPower': {'ppv': 2028.0, 'ppvDetail': {'ppv1': 1018.0, 'ppv2': 1010.0, 'ppv3': 0.0, 'ppv4': 0.0, 'pmeterDc': 0.0}, 'soc': 80.8, 'pev': 0, 'pevDetail': {'ev1Power': 0, 'ev2Power': 0, 'ev3Power': 0, 'ev4Power': 0}, 'prealL1': 473.0, 'prealL2': 0.0, 'prealL3': 0.0, 'pgrid': 0.0, 'pgridDetail': {'pmeterL1': 0.0, 'pmeterL2': 0.0, 'pmeterL3': 0.0}, 'pbat': -1367.0, 'pload': 661.0}, 'ChargeConfig': {'gridCharge': 1, 'timeChaf1': '05:30', 'timeChae1': '07:00', 'timeChaf2': '00:00', 'timeChae2': '00:00', 'batHighCap': 100.0}, 'DisChargeConfig': {'ctrDis': 1, 'timeDisf1': '07:00', 'timeDise1': '17:00', 'timeDisf2': '17:00', 'timeDise2': '06:00', 'batUseCap': 15.0}}]`

Let me know if you need more.

bmannz, 2023-12-04:
How frustrating.
No problem Charles.  I will leave debug running and capture the logs around both midnight local time and midnight UTC.
I will send them through when I have them.

gregpetersau, 2023-12-04:
I will be able to pull some logs today, but i wont be able to catch them right on midnight as ill be at work. If the logs let me rewind far enough back i will extract the appropriate ones.

My logs are similar to bmannz - showing the correct local time
5th Dec, 2023, 7:14AM

2023-12-05 07:14:50.585 DEBUG (MainThread) [alphaess.alphaess] Trying to call https://openapi.alphaess.com/api/getEssList
2023-12-05 07:14:52.427 DEBUG (MainThread) [alphaess.alphaess] Trying to call https://openapi.alphaess.com/api/getSumDataForCustomer?sysSn=AL2002120050775
2023-12-05 07:14:53.655 DEBUG (MainThread) [alphaess.alphaess] DATE: 2023-12-05
2023-12-05 07:14:53.655 DEBUG (MainThread) [alphaess.alphaess] DATE: 2023-12-05
2023-12-05 07:14:53.655 DEBUG (MainThread) [alphaess.alphaess] Trying to call https://openapi.alphaess.com/api/getOneDateEnergyBySn?sysSn=AL2002120050775&queryDate=2023-12-05
2023-12-05 07:14:54.569 DEBUG (MainThread) [alphaess.alphaess] Trying to call https://openapi.alphaess.com/api/getLastPowerData?sysSn=AL2002120050775
2023-12-05 07:14:55.011 DEBUG (MainThread) [alphaess.alphaess] Trying to call https://openapi.alphaess.com/api/getChargeConfigInfo?sysSn=AL2002120050775
2023-12-05 07:14:55.471 DEBUG (MainThread) [alphaess.alphaess] Trying to call https://openapi.alphaess.com/api/getDisChargeConfigInfo?sysSn=AL2002120050775
2023-12-05 07:14:55.911 DEBUG (MainThread) [alphaess.alphaess] [{'sysSn': 'AL2002120050775', 'popv': 6.6, 'minv': 'SMILE5-INV', 'poinv': 5.0, 'cobat': 10.3, 'mbat': 'SMILE-BAT-10.3P', 'surplusCobat': 9.3, 'usCapacity': 90.0, 'emsStatus': 'Normal', 'SumData': {'epvtoday': 27.4, 'epvtotal': 20934.2, 'eload': 22.45, 'eoutput': 5.03, 'einput': 0.08, 'echarge': 8.3, 'edischarge': 5.7, 'todayIncome': 5.27, 'totalIncome': 3564.48, 'eselfConsumption': 0.7381, 'eselfSufficiency': 0.82631, 'treeNum': 3584.1951, 'carbonNum': 18003.412, 'moneyType': 'AUD'}, 'OneDateEnergy': {'sysSn': 'AL2002120050775', 'theDate': '2023-12-05', 'eCharge': 0.3, 'epv': 0.5, 'eOutput': 0.0, 'eInput': 0.0, 'eGridCharge': 0.0, 'eDischarge': 2.1, 'eChargingPile': 0.0}, 'LastPower': {'ppv': 1812.0, 'ppvDetail': {'ppv1': 703.0, 'ppv2': 1109.0, 'ppv3': 0.0, 'ppv4': 0.0, 'pmeterDc': 0.0}, 'soc': 55.2, 'pev': 0, 'pevDetail': {'ev1Power': 0, 'ev2Power': 0, 'ev3Power': 0, 'ev4Power': 0}, 'prealL1': 203.0, 'prealL2': 0.0, 'prealL3': 0.0, 'pgrid': 0.0, 'pgridDetail': {'pmeterL1': 0.0, 'pmeterL2': 0.0, 'pmeterL3': 0.0}, 'pbat': -1510.0, 'pload': 302.0}, 'ChargeConfig': {'gridCharge': 0, 'timeChaf1': '15:00', 'timeChae1': '17:00', 'timeChaf2': '05:00', 'timeChae2': '06:00', 'batHighCap': 100.0}, 'DisChargeConfig': {'ctrDis': 0, 'timeDisf1': '00:00', 'timeDise1': '00:00', 'timeDisf2': '00:00', 'timeDise2': '00:00', 'batUseCap': 13.0}}]
2023-12-05 07:15:55.640 DEBUG (MainThread) [alphaess.alphaess] Trying to call https://openapi.alphaess.com/api/getEssList
2023-12-05 07:15:56.089 DEBUG (MainThread) [alphaess.alphaess] Trying to call https://openapi.alphaess.com/api/getSumDataForCustomer?sysSn=AL2002120050775
2023-12-05 07:15:56.710 DEBUG (MainThread) [alphaess.alphaess] DATE: 2023-12-05
2023-12-05 07:15:56.711 DEBUG (MainThread) [alphaess.alphaess] DATE: 2023-12-05
2023-12-05 07:15:56.711 DEBUG (MainThread) [alphaess.alphaess] Trying to call https://openapi.alphaess.com/api/getOneDateEnergyBySn?sysSn=AL2002120050775&queryDate=2023-12-05
2023-12-05 07:15:57.164 DEBUG (MainThread) [alphaess.alphaess] Trying to call https://openapi.alphaess.com/api/getLastPowerData?sysSn=AL2002120050775
2023-12-05 07:15:57.600 DEBUG (MainThread) [alphaess.alphaess] Trying to call https://openapi.alphaess.com/api/getChargeConfigInfo?sysSn=AL2002120050775
2023-12-05 07:15:58.038 DEBUG (MainThread) [alphaess.alphaess] Trying to call https://openapi.alphaess.com/api/getDisChargeConfigInfo?sysSn=AL2002120050775
2023-12-05 07:15:58.478 DEBUG (MainThread) [alphaess.alphaess] [{'sysSn': 'AL2002120050775', 'popv': 6.6, 'minv': 'SMILE5-INV', 'poinv': 5.0, 'cobat': 10.3, 'mbat': 'SMILE-BAT-10.3P', 'surplusCobat': 9.3, 'usCapacity': 90.0, 'emsStatus': 'Normal', 'SumData': {'epvtoday': 27.4, 'epvtotal': 20934.2, 'eload': 22.45, 'eoutput': 5.03, 'einput': 0.08, 'echarge': 8.3, 'edischarge': 5.7, 'todayIncome': 5.27, 'totalIncome': 3564.48, 'eselfConsumption': 0.7381, 'eselfSufficiency': 0.82631, 'treeNum': 3584.1951, 'carbonNum': 18003.412, 'moneyType': 'AUD'}, 'OneDateEnergy': {'sysSn': 'AL2002120050775', 'theDate': '2023-12-05', 'eCharge': 0.5, 'epv': 0.5, 'eOutput': 0.0, 'eInput': 0.0, 'eGridCharge': 0.0, 'eDischarge': 2.1, 'eChargingPile': 0.0}, 'LastPower': {'ppv': 1895.0, 'ppvDetail': {'ppv1': 733.0, 'ppv2': 1162.0, 'ppv3': 0.0, 'ppv4': 0.0, 'pmeterDc': 0.0}, 'soc': 56.0, 'pev': 0, 'pevDetail': {'ev1Power': 0, 'ev2Power': 0, 'ev3Power': 0, 'ev4Power': 0}, 'prealL1': 241.0, 'prealL2': 0.0, 'prealL3': 0.0, 'pgrid': 0.0, 'pgridDetail': {'pmeterL1': 0.0, 'pmeterL2': 0.0, 'pmeterL3': 0.0}, 'pbat': -1537.0, 'pload': 358.0}, 'ChargeConfig': {'gridCharge': 0, 'timeChaf1': '15:00', 'timeChae1': '17:00', 'timeChaf2': '05:00', 'timeChae2': '06:00', 'batHighCap': 100.0}, 'DisChargeConfig': {'ctrDis': 0, 'timeDisf1': '00:00', 'timeDise1': '00:00', 'timeDisf2': '00:00', 'timeDise2': '00:00', 'batUseCap': 13.0}}]

CharlesGillanders, 2023-12-04:
> How frustrating.
> 
> No problem Charles.  I will leave debug running and capture the logs around both midnight local time and midnight UTC.
> 
> I will send them through when I have them.

Thank you!

Poshy163, 2023-12-05:
Yep, this might just come to a case of dumping a bunch of logging calls in different parts of that specific call and seeing what comes of it.

gregpetersau, 2023-12-05:
Here is a log at 11am local (immediately before and after) which is UTC Midnight 

You can see the daily values reset, i have put in bold the edischarge which reflects what is graphed.

2023-12-05 10:59:47.641 DEBUG (MainThread) [alphaess.alphaess] Trying to call https://openapi.alphaess.com/api/getEssList
2023-12-05 10:59:48.307 DEBUG (MainThread) [alphaess.alphaess] Trying to call https://openapi.alphaess.com/api/getSumDataForCustomer?sysSn=AL2002120050775
2023-12-05 10:59:48.898 DEBUG (MainThread) [alphaess.alphaess] DATE: 2023-12-05
2023-12-05 10:59:48.899 DEBUG (MainThread) [alphaess.alphaess] DATE: 2023-12-05
2023-12-05 10:59:48.899 DEBUG (MainThread) [alphaess.alphaess] Trying to call https://openapi.alphaess.com/api/getOneDateEnergyBySn?sysSn=AL2002120050775&queryDate=2023-12-05
2023-12-05 10:59:49.342 DEBUG (MainThread) [alphaess.alphaess] Trying to call https://openapi.alphaess.com/api/getLastPowerData?sysSn=AL2002120050775
2023-12-05 10:59:49.770 DEBUG (MainThread) [alphaess.alphaess] Trying to call https://openapi.alphaess.com/api/getChargeConfigInfo?sysSn=AL2002120050775
2023-12-05 10:59:50.200 DEBUG (MainThread) [alphaess.alphaess] Trying to call https://openapi.alphaess.com/api/getDisChargeConfigInfo?sysSn=AL2002120050775
2023-12-05 10:59:50.641 DEBUG (MainThread) [alphaess.alphaess] [{'sysSn': 'AL2002120050775', 'popv': 6.6, 'minv': 'SMILE5-INV', 'poinv': 5.0, 'cobat': 10.3, 'mbat': 'SMILE-BAT-10.3P', 'surplusCobat': 9.3, 'usCapacity': 90.0, 'emsStatus': 'Normal', 'SumData': {'epvtoday': 27.4, 'epvtotal': 20934.2, 'eload': 22.45, 'eoutput': 5.03, 'einput': 0.08, 'echarge': 8.3, **'edischarge': 5.7**, 'todayIncome': 5.27, 'totalIncome': 3564.48, 'eselfConsumption': 0.7381, 'eselfSufficiency': 0.82631, 'treeNum': 3584.1951, 'carbonNum': 18003.412, 'moneyType': 'AUD'}, 'OneDateEnergy': {'sysSn': 'AL2002120050775', 'theDate': '2023-12-05', 'eCharge': 5.0, 'epv': 15.4, 'eOutput': 5.45, 'eInput': 0.07, 'eGridCharge': 0.0, 'eDischarge': 2.2, 'eChargingPile': 0.0}, 'LastPower': {'ppv': 4589.0, 'ppvDetail': {'ppv1': 2294.0, 'ppv2': 2295.0, 'ppv3': 0.0, 'ppv4': 0.0, 'pmeterDc': 0.0}, 'soc': 100.0, 'pev': 0, 'pevDetail': {'ev1Power': 0, 'ev2Power': 0, 'ev3Power': 0, 'ev4Power': 0}, 'prealL1': 4516.0, 'prealL2': 0.0, 'prealL3': 0.0, 'pgrid': -3286.0, 'pgridDetail': {'pmeterL1': -3286.0, 'pmeterL2': 0.0, 'pmeterL3': 0.0}, 'pbat': 16.0, 'pload': 1319.0}, 'ChargeConfig': {'gridCharge': 0, 'timeChaf1': '15:00', 'timeChae1': '17:00', 'timeChaf2': '05:00', 'timeChae2': '06:00', 'batHighCap': 100.0}, 'DisChargeConfig': {'ctrDis': 0, 'timeDisf1': '00:00', 'timeDise1': '00:00', 'timeDisf2': '00:00', 'timeDise2': '00:00', 'batUseCap': 13.0}}]

2023-12-05 11:00:50.640 DEBUG (MainThread) [alphaess.alphaess] Trying to call https://openapi.alphaess.com/api/getEssList
2023-12-05 11:00:51.080 DEBUG (MainThread) [alphaess.alphaess] Trying to call https://openapi.alphaess.com/api/getSumDataForCustomer?sysSn=AL2002120050775
2023-12-05 11:00:51.835 DEBUG (MainThread) [alphaess.alphaess] DATE: 2023-12-05
2023-12-05 11:00:51.836 DEBUG (MainThread) [alphaess.alphaess] DATE: 2023-12-05
2023-12-05 11:00:51.836 DEBUG (MainThread) [alphaess.alphaess] Trying to call https://openapi.alphaess.com/api/getOneDateEnergyBySn?sysSn=AL2002120050775&queryDate=2023-12-05
2023-12-05 11:00:52.272 DEBUG (MainThread) [alphaess.alphaess] Trying to call https://openapi.alphaess.com/api/getLastPowerData?sysSn=AL2002120050775
2023-12-05 11:00:52.702 DEBUG (MainThread) [alphaess.alphaess] Trying to call https://openapi.alphaess.com/api/getChargeConfigInfo?sysSn=AL2002120050775
2023-12-05 11:00:53.133 DEBUG (MainThread) [alphaess.alphaess] Trying to call https://openapi.alphaess.com/api/getDisChargeConfigInfo?sysSn=AL2002120050775
2023-12-05 11:00:53.572 DEBUG (MainThread) [alphaess.alphaess] [{'sysSn': 'AL2002120050775', 'popv': 6.6, 'minv': 'SMILE5-INV', 'poinv': 5.0, 'cobat': 10.3, 'mbat': 'SMILE-BAT-10.3P', 'surplusCobat': 9.3, 'usCapacity': 90.0, 'emsStatus': 'Normal', 'SumData': {'epvtoday': 15.4, 'epvtotal': 20934.2, 'eload': 10.02, 'eoutput': 5.45, 'einput': 0.07, 'echarge': 5.0, '**edischarge': 2.2,** 'todayIncome': 2.57, 'totalIncome': 3564.48, 'eselfConsumption': 0.7381, 'eselfSufficiency': 0.82631, 'treeNum': 3584.1951, 'carbonNum': 18003.412, 'moneyType': 'AUD'}, 'OneDateEnergy': {'sysSn': 'AL2002120050775', 'theDate': '2023-12-05', 'eCharge': 5.0, 'epv': 15.4, 'eOutput': 5.45, 'eInput': 0.07, 'eGridCharge': 0.0, 'eDischarge': 2.2, 'eChargingPile': 0.0}, 'LastPower': {'ppv': 4589.0, 'ppvDetail': {'ppv1': 2294.0, 'ppv2': 2295.0, 'ppv3': 0.0, 'ppv4': 0.0, 'pmeterDc': 0.0}, 'soc': 100.0, 'pev': 0, 'pevDetail': {'ev1Power': 0, 'ev2Power': 0, 'ev3Power': 0, 'ev4Power': 0}, 'prealL1': 4516.0, 'prealL2': 0.0, 'prealL3': 0.0, 'pgrid': -3286.0, 'pgridDetail': {'pmeterL1': -3286.0, 'pmeterL2': 0.0, 'pmeterL3': 0.0}, 'pbat': 16.0, 'pload': 1319.0}, 'ChargeConfig': {'gridCharge': 0, 'timeChaf1': '15:00', 'timeChae1': '17:00', 'timeChaf2': '05:00', 'timeChae2': '06:00', 'batHighCap': 100.0}, 'DisChargeConfig': {'ctrDis': 0, 'timeDisf1': '00:00', 'timeDise1': '00:00', 'timeDisf2': '00:00', 'timeDise2': '00:00', 'batUseCap': 13.0}}]

Poshy163, 2023-12-05:
Yep, so I can see a synced date called between the 2, I'm in the ACST Timezone (you appear to be on the AEST), so I'll play around with the values tonight and see if we the date desync, and where it's occouring within the call

bmannz, 2023-12-05:
Here is a log covering nearly 24hrs so it covers midnight UTC and midnight local time. Hopefully it gives you some helpful information.
[home-assistant_2023-12-05T17-21-34.857Z.zip](https://github.com/CharlesGillanders/homeassistant-alphaESS/files/13565090/home-assistant_2023-12-05T17-21-34.857Z.zip)

Poshy163, 2023-12-05:
Super strange, it appears that your local timezone date is being parsed into the command/query correctly, as seen by 
`2023-12-05 23:59:48.600 ... 'epvtoday': 18.3,`

Then on the following "day"

` 2023-12-06 00:01:57.381 ... 'epvtoday': 18.3, `

This condition is no included as one of the postman scripts (of which ill make a pull for now) and looking in the alphaess docs for the openAPI, the `getSumDataForCustomer` call doesn't include any date parameters? therefore must only be returning the UTC date

Poshy163, 2023-12-06:
Just to update yall on how its going, we may have to go back to webscraping to get the info we need.....

[image]

[image]

Poshy163, 2023-12-06:
By the sounds of it, he seems pretty stern on not implementing a querydate call, or any other solution for `getSumDataForCustomer`

We may have to-re implement the old method to get all the stats returned from it

gregpetersau, 2023-12-06:
It could be a language barrier thing and not understanding that this is a problem for all users not in UTC.
I read the response as not interested in being able to recall any historical date, not related to the incorrect today data being transferred

Poshy163, 2023-12-06:
Yeah, ill try and say it a few different way, but it seems that he is just using google translate, so might have to go through someone else to get this resolved

Poshy163, 2023-12-06:
He is really not understanding...
@CharlesGillanders we may need to suggest it another way to get anything meaningful to happen

[image]

dan-s-github, 2023-12-06:
other option would be to ignore the 'SumData' from the API and use sthe values from 'OneDateEnergy', only missing value is eload.
epvtoday > epv
eload !!! no entry in OneDateEnergy 
einput > eInput
echarge > eCharge
edischarge > eDischarge
eoutput > eOutput

I am in UTC+13 here log from UTC reset
```
2023-12-06 12:59:17.603 
'SumData': {'epvtoday': 7.3**, 'epvtotal': 516214.9, 'eload': 9.7, 'eoutput': 0.0, 'einput': 2.4, 'echarge': 2.3, 'edischarge': 7.9, 'todayIncome': 1.97, 'totalIncome': 123609.13, 'eselfConsumption': 0.8789, 'eselfSufficiency': 0.36855, 'treeNum': 88382.4037, 'carbonNum': 443944.814, 'moneyType': 'NZD'}, 
'OneDateEnergy': {'sysSn': 'AL2002118050331', 'theDate': '2023-12-06', 'eCharge': 12.0, 'epv': 15.7,** 'eOutput': 3.6, 'eInput': 6.0, 'eGridCharge': 3.7, 'eDischarge': 1.8, 'eChargingPile': 0.0},

2023-12-06 13:00:24.267 
'SumData': {'epvtoday': 15.7, 'epvtotal': 516214.9, 'eload': 18.1, 'eoutput': 3.6, 'einput': 6.0, 'echarge': 12.0, 'edischarge': 1.8, 'todayIncome': 4.65, 'totalIncome': 123609.13, 'eselfConsumption': 0.8789, 'eselfSufficiency': 0.36855, 'treeNum': 88382.4037, 'carbonNum': 443944.814, 'moneyType': 'NZD'}, 
'OneDateEnergy': {'sysSn': 'AL2002118050331', 'theDate': '2023-12-06', 'eCharge': 12.0, 'epv': 15.7, 'eOutput': 3.6, 'eInput': 6.0, 'eGridCharge': 3.7, 'eDischarge': 1.8, 'eChargingPile': 0.0},

```

Poshy163, 2023-12-06:
Just an update and a rough translation from what he has said 
> Is this time UTC time or your local time?

> I roughly know what the problem is.

> This statistical data is refreshed every 5Min. The first data is uploaded at 12:05. If the data does not meet the 2 data, the first statistical value data will be directly defaulted. When your time reaches around 12:10~12:13, the second data will be refreshed. Data, the statistical values ​​​​will be updated at this time.

I have sent more data from later in the day (about 6am ish), so i appear to be getting closer 

[image]

Poshy163, 2023-12-06:
Alright, a meaningful update this time. 

It appears that we are meant to be using getOneDateEnergyBySn to return the data, instead of getSumDataForCustomer, which the sum data, is meant to be the total (hence why the response includes tree num, eselfconsumption, total income 

So ill adjust the separated data-points in a pull request, which might fuck a few things up. `eload` will have to be calculated, as its not calculated by the response from `getOneDateEnergyBySn`

`epvtoday` will turn into `epv`
`eoutput` will turn into `eOutput`
`eoutput` will turn into `eCharge`
`eoutput` will turn into `eOutput`
`einput` will turn into `eInput`
`edischarge` will turn into `eDischarge

As seen here, i will work on a pull now for the changes

[image]

[image]

Poshy163, 2023-12-06:
@CharlesGillanders @dan-s-github @gregpetersau.

i am working on a fix for all of this on a fork, i have added some temp code which im testing locally, and will make a pull request when i can verify it works. you all are free to also have a look and see if it works for you.

this is the pull here [Updated location of many commands in the UTC bug
](https://github.com/Poshy163/homeassistant-alphaESS/commit/6e77cbadddc434b61956a65252d1a774763a51a7)

Also had a good convo with him about new things down the pipeline, of which i can implement some asap

dan-s-github, 2023-12-07:
Hi @Poshy163 thanks for working on a fix, made comments on one of the commits in your repo

Also I would argue that there is a bug in the SumData API from AlphaESS as `epvtoday` is not actually resting to 0 but to currently accumulated value of `epv`.

Regarding `eload` we should ask if this could be added to the `getOneDateEnergyBySn` API response

Poshy163, 2023-12-07:
Yep, fixed that in a pull immediately after releasing it.

Will make a pull into the repo asap

gregpetersau, 2023-12-07:
fix looking good! :) ill monitor

SiL-X, 2023-12-09:
Everything is working fine here since 0.4.6.
States resets at midnight.
Nice work! :)

gregpetersau, 2023-12-10:
All seems to be working except the sensor Total Load which i see in the code comments was expected as its not in the `getOneDateEnergyBySn` 
Assume we have little influence over that?

Poshy163, 2023-12-10:
yeah, according to eleven.fan (head of dev), 
> There is no Load here. Many people like to calculate the Load themselves.

[image]

bmannz, 2023-12-10:
Working perfect for me too.  I don't use Total load for anything important.  Only use grid_to_load and battery_to_load.
Thanks to you guys for fixing it so quickly.

closed 2023-12-12 by CharlesGillanders