CharlesGillanders/homeassistant-alphaESS #86: Error Failed Setup

grahambelli, 2023-11-19:
After upgrade the HomeAssistant component to  version 0.4.2  and removing old Alpha service  and setting up new service i get repeated Failed Setup error

2023-11-19 11:45:47.281 ERROR (MainThread) [alphaess.alphaess] 'NoneType' object is not iterable
2023-11-19 11:45:47.281 ERROR (MainThread) [custom_components.alphaess] Unexpected error fetching alphaess data: 'NoneType' object is not iterable
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 290, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/alphaess/coordinator.py", line 30, in _async_update_data
    jsondata: json = await self.api.getdata()
                     ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/alphaess/alphaess.py", line 240, in getdata
    for unit in units:
TypeError: 'NoneType' object is not iterable
2023-11-19 11:46:00.457 ERROR (MainThread) [custom_components.alphaess] Unexpected error fetching alphaess data: 'NoneType' object has no attribute 'get'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 290, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/alphaess/coordinator.py", line 45, in _async_update_data
    _pv =  _onedateenergy.get("epv")
           ^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'get'
2023-11-19 11:46:00.460 DEBUG (MainThread) [custom_components.alphaess] Finished fetching alphaess data in 8.095 seconds (success: False)
2023-11-19 11:46:18.730 ERROR (MainThread) [custom_components.alphaess] Unexpected error fetching alphaess data: 'NoneType' object has no attribute 'get'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 290, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/alphaess/coordinator.py", line 45, in _async_update_data
    _pv =  _onedateenergy.get("epv")
           ^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'get'
2023-11-19 11:46:18.732 DEBUG (MainThread) [custom_components.alphaess] Finished fetching alphaess data in 7.948 seconds (success: False)
 
Any Guidance much appreciated
Graham

KortgatGH, 2023-11-19:
Also noticed strange behavior just now. Registered and changed to new API this afternoon. All worked well. Then I changed my Node-Red Alpha ESS node to the new API and it works well. But now it seems my Alpha ESS HA integration stopped working. I deleted and re-created the entry but it still reports "Failed setup, will retry". Is it perhaps problematic to access the API from both HA and Node-Red?

grahambelli, 2023-11-19:
> Just to confirm - you changed the authentication credentials to use your API key and secret from the openAPI developer portal and you also linked your inverter to your account in the same portal?

Correct exactly what I done

CharlesGillanders, 2023-11-19:
> > Just to confirm - you changed the authentication credentials to use your API key and secret from the openAPI developer portal and you also linked your inverter to your account in the same portal?
> 
> 
> 
> Correct exactly what I done 

Thanks for confirming.  Right now I am not at all sure what the issue might be.  How comfortable would you be running some tests for me?  I'd like to try to see if you could run the apitest python script that comes with the library that the home assistant integration relies on.  Let me know if that's feasible and I will try to send you instructions on it.

grahambelli, 2023-11-20:
I am happy to give it a go but I will need instruction to run the tests.

tbgoose, 2023-11-23:
Similar issue here after following the change-over instructions for OpenAPI. 

```
This error originated from a custom integration.

Logger: custom_components.alphaess
Source: helpers/update_coordinator.py:290
Integration: Alpha ESS (documentation, issues)
First occurred: 1:15:09 PM (6 occurrences)
Last logged: 1:19:01 PM

Unexpected error fetching alphaess data: 'NoneType' object has no attribute 'get'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 290, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/alphaess/coordinator.py", line 42, in _async_update_data
    _gridpowerdetails = _powerdata.get("pgridDetail",{})
                        ^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'get'

```
Strangly my system was working fine using the old auth method until I updated HA this morning. Following update it was broken, and after checking out github I noted the new API etc. I am in Aus so perhaps a regional rollout to changes or something.

droopycool, 2023-11-28:
Same issue here, new system set up today, getting "Failed setup, wil retry" error in HA

MonsterEnergy-wtf, 2023-11-28:
Short update: the integration seems to work fine. I guess that it takes some time until the servers at AlphaEss setup the usage of the API for a new created developer account.
For me it took about 20 hours.

Right now everything works as expected.So I assume this is not related to the implementation rather than the endpoint at AlphaESS.

CharlesGillanders, 2023-11-28:
> Short update: the integration seems to work fine. I guess that it takes some time until the servers at AlphaEss setup the usage of the API for a new created developer account.
>
> For me it took about [...]

Great news, thanks for confirming

droopycool, 2023-11-28:
Oh, thanks for the info! I will give it some time then!

tbgoose, 2023-11-29:
> For me it took about 20 hours.

Mine has been not working for far longer than 20 hours. There is something else happening here. Would be very interested in helping do some testing if it helps. I really miss my data in HA as I relied on it to know battery SoC and run appliances etc when producing solar.

CharlesGillanders, 2023-11-29:
> I am happy to give it a go but I will need instruction to run the tests.

@grahambelli Thanks for volunteering!  Here's instructions, let me know if you get stuck at any point:

1) You will need CLI access to your home assistant environment - depending on how you are running HA this might be achieved in different ways.  I am running home assistant OS so I chose to install the Terminal & SSH add-on through the HA add-on store.  That gives me a handy terminal option in the home assistant side menu.

2) Once you have CLI access, connect to your home assistant and then use the cd command to browse to the config folder - in my case that's just /root/config
`$ cd /root/config`

3) use the ls command to confirm that you are in the right folder by checking that you can see a custom_components folder, if so use cd to navigate into that folder.
`config $ ls`
`config $ cd custom_components`

4) do the same again as step 3 but this time you want to look for and navigate to the alphaess folder
`custom_components $ ls`
`custom_components $ cd alphaess`

5) Once in the alphaess folder check that you have all of the files - use the ls command and you should expect to see a file listing like this:
`alphaess $ ls`
`__init__.py    __pycache__     config_flow.py    const.py    coordinator.py    entity.py    enums.py    manifest.json    sensor.py 
   services.yaml    strings.json    translations`

6) create a new temporary folder for testing the underlying library using "mkdir templib", and then use cd to navigate into that folder.
`alphaess $ mkdir templib`
`alphaess $ cd templib`

7) Grab a copy of the underlying library from github using git clone

`templib $ git clone https://github.com/CharlesGillanders/alphaess-openAPI.git`
`Cloning into 'alphaess-openAPI'`
`...`

8) cd into the new alphaess-openAPI folder
`templib $ cd alphaess-openAPI`

9) create and activate a python virtual environment
`alphaess-openAPI $ python -m venv venv`
`alphaess-openAPI $ source venv/bin/activate`
`(venv) alphaess-openAPI $`

10) install the required libraries
`(venv) alphaess-openAPI $ pip install aiohttp`
`(venv) alphaess-openAPI $ pip install voluptuous`

11) run the apitest.py python script, when prompted paste in your AppID and AppSecret from the AlphaESS developer portal
`(venv) alphaess-openAPI $ python apitest.py`
`AppID: xxxxxxxxxxxxxx`
`AppSecret: xxxxxxxxxxxxxxxxxxx`

12) Copy all of the output and results from that apitest script and attach them as a file to this issue and I will take a look to see if I can start to figure out what is going wrong.

13) Clean up and remove everything that was done
`(venv) alphaess-openAPI $ deactivate`
`alphaess-openAPI $ cd ../..`
`alphaess $ rm -rf templib`

grahambelli, 2023-11-30:
> > Hi Charles
> > Step 9. python -m venv venv
> > results in the error below
> > bash: phython : command not found
> 
> That looks like a typo - please check that you spelt python correctly. (Unless I spelt it wrong in the instructions, if so sorry!)

Sorry my typo in the error but command was correct, so I’m getting command not found.

grahambelli, 2023-11-30:
Running on oracle VM on win10 machine

tbgoose, 2023-11-30:
I can get the test working to the point it identifies my system serial number correctly (redacted but it does match my system), but then get the following error. Apologies I can't seem to copy the output from the command from SSH&Terminal addon, so screenshot is below.

[image]

CharlesGillanders, 2023-12-01:
> Running on oracle VM on win10 machine

Assuming that the VM is relatively standard and that it's only the PATH that might be a problem here can you perhaps try using what should be the full path to python?

`/usr/bin/python -m venv venv`

If that also fails perhaps you could try to see if you can find the python executable within your VM

`find / -name python -print`

grahambelli, 2023-12-01:
> Assuming that the VM is relatively standard and that it's only the PATH that might be a problem here can you perhaps try using what should be the full path to python?
>
> `/usr/bin/python -m venv [...]

Hi Charles 

Sorry if I'm over complicating the error with my setup. here is a screen shot of what i found. 
[image]

CharlesGillanders, 2023-12-02:
> Hi Charles
>
>
>
> Sorry if I'm over complicating the error with my setup. here is a screen shot of what i found.

I **think** the problem is with how you are accessing the cli - are you using the ssh add on from the add on store or are you using something else?

CharlesGillanders, 2023-12-04:
For everyone experiencing this issue, I am not sure why the extra logging I added is not giving me more information.  It might help if I could "borrow" a set of api credentials for a  on working system temporarily to try some further debugging myself.  I believe you can easily change the app secret once I'm done.

If any of you is comfortable sharing that I suggest you email me rather than post the details here.

I'm reachable at [firstname@firstnamelastname.com] just substitute my actual name there...

tbgoose, 2023-12-04:
> I'm reachable at ***@***.*** just substitute my

Happy to share if it will help. Emailing now.

tbgoose, 2023-12-04:
Hi all. AlphaESS has [fixed a bug on the platform](https://github.com/alphaess-developer/alphacloud_open_api/issues/47) that was stopping me from removing my battery to re-bind it. After removing and then re-binding this addon is working correctly again. Thanks for your help @CharlesGillanders

closed 2023-12-12 by CharlesGillanders