CharlesGillanders/homeassistant-alphaESS #87: Entity sensor.alphaess_solar_to_load from integration alphaess has state class total_increasing, but its state is not strictly increasing

MrIcemanLE, 2023-11-24:
```
Logger: homeassistant.components.sensor.recorder
Source: components/sensor/recorder.py:288
Integration: Sensor (documentation, issues)
First occurred: 18. November 2023 um 11:05:10 (2 occurrences)
Last logged: 18. November 2023 um 14:00:10

Entity sensor.alphaess_solar_to_load from integration alphaess has state class total_increasing, but its state is not strictly increasing. Triggered by state 3.06 (3.07) with last_updated set to 2023-11-18T10:04:29.585956+00:00. Please create a bug report at https://github.com/CharlesGillanders/homeassistant-alphaESS/issues
Entity sensor.alphaess_total_load from integration alphaess has state class total_increasing, but its state is not strictly increasing. Triggered by state 18.35 (18.36) with last_updated set to 2023-11-18T12:59:46.008211+00:00. Please create a bug report at https://github.com/CharlesGillanders/homeassistant-alphaESS/issues
```

CharlesGillanders, 2023-11-29:
Related to (and possibly duplicate of) #85 Can you please make sure you upgrade to version 0.4.4 and try again?

closed 2023-12-12 by CharlesGillanders

qbushome, 2024-01-10:
Hi Charles

Just a quick message to report that I still get the same errors as reported by [MrIcemanLE](https://github.com/MrIcemanLE)

Entity sensor.al70010xxxxxxxx_solar_to_load from integration alphaess has state class total_increasing, but its state is negative. Triggered by state -0.02 with last_updated set to 2024-01-10T14:34:59.999999+00:00.

qbushome, 2024-06-16:
Hi Charles

I still have this issue.
Maybe I left out some vital information in my previous post:

Currently running version 0.4.8

`Entity sensor.alxxxxxxxxxxxxxxx_total_load from integration alphaess has state class total_increasing, but its state is not strictly increasing. Triggered by state 8.28 (8.35) with last_updated set to 2024-06-16T16:29:28.234132+00:00. Please create a bug report at https://github.com/CharlesGillanders/homeassistant-alphaESS/issues`