CharlesGillanders/homeassistant-alphaESS #90: Cant add integration into HA. Error code: Unknown error occurred

wace70, 2023-12-02:
Since today my Integration that worked the last weeks is not working anymore.
So I decided to deinstall and reinstall it with verion 0.4.4.
I see that I need now an developer account.
So I created one and added my system to this account.
I got now the AppID and Appsecret.

If I add not the alphaess integration and enter these two credentiels into the popup window, I gor an error message: Unknown error occurred

Is there a workaround for this available or is it a brand new issue?

I need help...

CharlesGillanders, 2023-12-02:
I think you are experiencing the same issue that some others are getting.  I need some further debug information to figure out what's happening.  If you were able to try following these instructions it might help.

[https://github.com/CharlesGillanders/homeassistant-alphaESS/issues/86#issuecomment-1831817244](https://github.com/CharlesGillanders/homeassistant-alphaESS/issues/86#issuecomment-1831817244)

wace70, 2023-12-02:
Hi Charles, I have attached the output as a file now.
Thanks for your work, I am appreciate that really.
[open api output1.docx](https://github.com/CharlesGillanders/homeassistant-alphaESS/files/13537360/open.api.output1.docx)

CharlesGillanders, 2023-12-02:
> Hi Charles, I have attached the output as a file now.
> 
> Thanks for your work, I am appreciate that really.
> 
> [open api output1.docx](https://github.com/CharlesGillanders/homeassistant-alphaESS/files/13537360/open.api.output1.docx)
> 
> 

That's annoying, the extra debug information I was hoping for isn't there - I'll have to look at the code again and add more.  I'll get back to you when I've got something else for you to try.

closed 2023-12-03 by CharlesGillanders